arendst/Tasmota #7240: Possible to add IR transmitter to Sonoff 4CH Pro R2?  (aka free GPIO on 4CH Pro R2?)

sofakng, 2019-12-17:
Does anybody know if there are any free GPIO pins on the Sonoff 4CH Pro R2?  I'd like to connect an IR transmitter but I don't see any free GPIO pins.

Because the device has dual processors (ESP8255 and STM32) and an RF receiver (433 MHz) I'm wondering if all of the GPIO pins are used?

closed 2019-12-17 by close-issue-app[bot]

meingraham, 2019-12-17:
The serial pins (IO1/IO3) are free during normal operation.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

sofakng, 2019-12-17:
Thanks very much for the reply and I apologize for incorrectly posting this as a GitHub issue.