arendst/Tasmota #7242: Minimize the delay of checking poweronstate after powering on

gongtao0607, 2019-12-17:
**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
The poweronstate setting is checked (applied) after a short delay after power on (~2s, wifi connection I guess). This delay should be minimized.

**Describe the solution you'd like**  
poweronstate should be checked as soon as possible after powering on. (e.g., before wifi connection) 

**Describe alternatives you've considered**  

**Additional context**  
Scenario: an (light) outlet is already controlled by a wall switch (common US wiring). Adding a tasmota socket to the outlet can add extra control of tuning the light off. When turning on from wall switch, once tasmota is powered, it should instantly turn on the light (with poweronstate=1). The current booting sequence causes a ~2s delay and can be annoying.

**(Please, remember to close the issue when the problem has been addressed)**  
Sure.

meingraham, 2019-12-17:
The proper solution is to not cut power to the Tasmota device. Use the wall switch which is now controlling the outlet as an input to the Tasmota device instead. The Tasmota device will be on a responding to commands and toggling the wall switch will toggle the light almost instantly.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

gongtao0607, 2019-12-17:
Hi Michael,

Thanks for the feedback. I am not looking for troubleshooting or setup assistance. I have my reason and I think this delay may also cause issues in some other applications. 

I have a separate setup that is exactly what you described, using a sonoff. There are 3 major issues: 1. the sonoff is too large (even without the case) to stay inside the electric box with the original wall switch; 2. Using a DIY button can work, but looks ugly; 3. electric box requires some skills, but a wifi socket does not. 

We live in an apartment and we will have revert everything we modified before moving out. My setup compromises (only) the powering on part and does not modify wiring:
If wall switch is off, I can only turn the light on by the wall switch.
If wall switch is on but tasmota is off, I can turn it on by wiggle the wall switch (or use MQTT/Tasmota button/Alexa/Google Asistant/Homekit).
The fact is, I'm controlling a bedlamp that requires exactly as above. The wall switch is at the entry, I turn on the light upon enter. I use alexa to turn it off when sleep. Alexa turns it on in the morning. And I turn it off when leaving the room.

Hope the above explains.

ascillato, 2019-12-17:
Hi, seems that you are having another issue. The boot process are few milliseconds in order to be applied Poweronstate. It is nearly not noticeable and is not dependant on wifi connections.

Please address this to the Tasmota support chat

ascillato, 2019-12-17:
Another point. Cutting down the power as a normal use is not what the manufacturer intended. You should use another approach like a shelly for example.

closed 2019-12-17 by ascillato2

meingraham, 2019-12-17:
@gongtao0607 

> Cutting down the power as a normal use is not what the manufacturer have intended

That's makes two of us pointing out the same thing. So, http://xyproblem.info/