arendst/Tasmota #7244: Sun blinder open/close position

guiumb, 2019-12-17:
**Have you looked for this feature in other issues and in the wiki?** 

**Is your feature request related to a problem? Please describe.**  
If you use Tasmota in order to control sun blinders up and down (tipically sun blinder on terrace) the buttons refer to the closeing /opening positions are swapped compared to shutters: in a shutter, close position 0 rappresents fully unrolled while in open position 100 it is rolled up.
On sun blinder close position is rolled up and full open position is unrolled down.

**Describe the solution you'd like**  
Should it be possible to have a parameter that "swaps" opening/closing positions (0/100) in order to map the open and closed state related to rolled up and unrolled down.

**Describe alternatives you've considered**  

**Additional context**  

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2019-12-17:
You can use the command shutterinvert 1 to invert your shutter1. If you need to invert other shutters just add the index.

ascillato2, 2019-12-17:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-17 by ascillato2

guiumb, 2019-12-18:
Many thanks for quickly replay!

Could I suggest adding a brief description about swap close/open position in wiki page, section Sthutter?

stefanbode, 2019-12-18:
Agree. I will add it soon. There will be a new release for the stepper motors quite soon that require also some document changes.

The invert is only mentioned in the commands. This is for sure not enough
https://tasmota.github.io/docs/#/Commands?id=blinds-shutters-and-roller-shades

guiumb, 2019-12-18:
Great !!

Many thanks for your work.

Have a nice day.