arendst/Tasmota #7247: Serial Received Option to remove New Line (CR/LF)

vzorglub, 2019-12-17:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
I receive data from a device called RFLink see: http://www.rflink.nl/blog2/
The data arrived via the serial port and is terminated with a new line character (CR/LF)
Therefore on the console we get: 
```
21:13:26 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"20;13;GlobalTronics;ID=000c;TEMP=0062;BAT=OK;
"}
```
And this is not strictly a valid JSON and gets refused by transformation services like node-red...

**Describe the solution you'd like**  
An option to remove the NewLine character from the received serial stream
And separate data received with the new line character:
I also get this in the console:
```
21:04:53 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"20;10;TriState;ID=0a8a8a;SWITCH=2;CMD=ON;
20;11;TriState;ID=0a8aaa;SWITCH=2;CMD=ON;
"}
```
This should have been 2 outputs instead of 1
**Describe alternatives you've considered**  
I have removed the NewLine character in software downsteam before processing the data
**Additional context**  
None
**(Please, remember to close the issue when the problem has been addressed)**

vzorglub, 2019-12-17:
@ascillato2 
Thank you
I was starting to wonder what I was doing wrong here.

ascillato2, 2019-12-17:
Your issue was published just in the moment the template was being updated. Now fixed.

Jason2866, 2019-12-17:
@vzorglub  use a other serialreceive modus. It gets activated if you do a serialsend.
Details see wiki in commands section

vzorglub, 2019-12-17:
I do a serial send on reboot with a rule.
The serial mode according to the RFLink website is 8N1
I have set the serial mode to 8N1 bt using the new serialconfig command:
```
21:52:42 CMD: serialconfig
21:52:42 MQT: stat/RFLinkBridge/RESULT = {"SerialConfig":"8N1"}
```
And this is the output when I receive data from my temperature sensor:
```
21:53:27 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"20;31;GlobalTronics;ID=000c;TEMP=0061;BAT=OK;
"}
```
Including the CR/LF characters
I would like to have the option to remove them in tasmota so that the MQTT payload sent is:
```
21:53:27 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"20;31;GlobalTronics;ID=000c;TEMP=0061;BAT=OK;"}
```

ascillato, 2019-12-17:
Can you test the command `serialdelimiter`?

From wikipedia:

> CR and LF are control characters, respectively coded 0x0D (13 decimal) and 0x0A (10 decimal). They are used to mark a line break in a text file. As you indicated, Windows uses two characters the CR LF sequence; Unix only uses LF and the old MacOS ( pre-OSX MacIntosh) used CR

So, can you test in the console the following? 
```
serialdelimiter 13

Or

serialdelimiter 10
```

vzorglub, 2019-12-18:
Ok,:
```
06:49:23 MQT: stat/RFLinkBridge/RESULT = {"SerialDelimiter":10}
06:52:37 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"20;12;Auriol;ID=0030;TEMP=0108;BAT=LOW;
"}
```
Didn't work

```
06:56:08 MQT: stat/RFLinkBridge/RESULT = {"SerialDelimiter":13}
06:58:27 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"20;13;GlobalTronics;ID=000c;TEMP=0041;BAT=OK;"}
06:58:28 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"
"}
```
Partially works, now CR (Character 13) splits the serial but then we get 2 payload with the second one only containing LF (Character 10)
It gets even messier with the streams close back to back:
```
07:03:07 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"20;17;TriState;ID=0a8a8a;SWITCH=2;CMD=ON;"}
07:03:07 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"
20;18;TriState;ID=0a8aaa;SWITCH=2;CMD=ON;"}
07:03:08 MQT: tele/RFLinkBridge/RESULT = {"SerialReceived":"
"}
```
Is there a possibility to do `serialdelimiter 13,10` ?

ascillato, 2019-12-18:
> Wow, undocumented command, cool!!

Sorry, no. It is an old command. See docs. 

Let's see if the 2 delimiters can be implemented

vzorglub, 2019-12-18:
> Sorry, no. It is an old command. See docs.

I checked, it's not there...

> Let's see if the 2 delimiters can be implemented

Thanks

ascillato, 2019-12-18:
https://github.com/arendst/Tasmota/wiki/Commands#serial-bridge

It is in the wiki. 

You are right that it is missing in the docs

vzorglub, 2019-12-18:
Didn't check the old wiki, I have now a Bookmark to the new docs....

meingraham, 2019-12-18:
SerialDelimiter command is now restored. It went AWOL and has been sent to the cooler for a few days to think about its infraction :wink:

vzorglub, 2019-12-18:
Thanks. The formating is a bit wonky though...
```
1..127 = set serial delimiter to decimal ASCII
```
Should be on its own line, should it not?

meingraham, 2019-12-18:
Yup. Fixed. Thx

I copied the markdown from the old wiki... and it had the error... but rendered properly. The new docs site is more strict apparently :wink:

vzorglub, 2019-12-18:
@ascillato2 
Should the awaiting feedback, and troubleshooting be removed and put back on feature request?
Thanks

ascillato2, 2019-12-18:
Ok. Feature request to do serialdelimiter 13,10

vzorglub, 2019-12-25:
Actually, this is not required as I found out that `SerialDelimiter 128` will remove both 13 and 10 and keep only 32 to 128 (As described in the docs) (Which were not available...)
I am closing this

closed 2019-12-25 by vzorglub