arendst/Tasmota #7248: irhav is not working inmy tasmota 7.0.0.6

sivaareddi, 2019-12-18:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

schrej-zz, 2019-12-18:
Are you using Standard tasmota.bin or tasmota-ir.bin? IR is not part of the standard tasmota.bin anymore.

sivaareddi, 2019-12-18:
I am using tasmota compiled using arduino IDE on NodeMCU

blakadder, 2019-12-18:
And did you define the required defines to allow you to use hvac IR protocols?

sivaareddi, 2019-12-18:
i am new to tasmota. Can you please say

> 
> 
> And did you define the required defines to allow you to use hvac IR protocols?

i am new to tasmota can  you please say what are those defines

blakadder, 2019-12-18:
Then why open an issue without understanding how it works?

Support is available in our documentation or support channels listed on https://tasmota.github.io/docs/

Closing issue.

closed 2019-12-18 by blakadder