arendst/Tasmota #7253: EXS Relay V5.1, no status in Alexa App

cheise, 2019-12-19:
### PROBLEM DESCRIPTION
After switching with Alexa App, no status displaying and a error showing "Device cannot be reached" in the app. Switching is no longer possible via the app :(
In the WebIf and over MQTT all things possible.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): EXS Relay V5.1
- [ ] Tasmota binary firmware version number used: 6.6.0
  - [ x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x ] Flashing tools used: NodeMCU Falsher
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
09:19:54 MQT: stat/Licht_Pflanze/STATUS = {"Status":{"Module":16,"FriendlyName":["Licht Pflanze","Sonoff2"],"Topic":"Licht_Pflanze","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:30","StartupUTC":"2019-12-19T08:19:24","Sleep":50,"CfgHolder":4617,"BootCount":143,"SaveCount":2503,"SaveAddress":"F6000"}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS2 = {"StatusFWR":{"Version":"6.6.0.2(sonoff)","BuildDateTime":"2019.07.18 19:07:07","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BKA_Berlin",""],"TelePeriod":300,"Resolution":"55A180C0","SetOption":["0000C009","2805C8000100060000005AFF000000000000","00000000"]}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS4 = {"StatusMEM":{"ProgramSize":505,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashMode":3,"Features":["00000407","0FDAE396","001683A0","22B617CD","01001BC0"]}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS5 = {"StatusNET":{"Hostname":"Licht_Pflanze-4510","IPAddress":"192.168.1.128","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"68:C6:3A:A1:91:9E","Webserver":2,"WifiConfig":5}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.222","MqttPort":1883,"MqttClientMask":"Licht_Pflanze","MqttClient":"Licht_Pflanze","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 19 08:19:54 2019","Local":"Thu Dec 19 09:19:54 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:39","Sunset":"16:54"}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS10 = {"StatusSNS":{"Time":"2019-12-19T09:19:54","Epoch":1576743594,"Switch1":"ON"}}
09:19:54 MQT: stat/Licht_Pflanze/STATUS11 = {"StatusSTS":{"Time":"2019-12-19T09:19:54","Epoch":1576743594,"Uptime":"0T00:00:30","UptimeSec":30,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":50,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"BKA_Berlin","BSSId":"CC:CE:1E:AD:EE:E1","Channel":11,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-19:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.
 
Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).
 
Thanks.
 
### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Jason2866, 2019-12-19:
Anyway you use a old version. Try latest.

closed 2019-12-19 by ascillato2