arendst/Tasmota #7255: Template module doesn't show settings

mkh595, 2019-12-19:
### PROBLEM DESCRIPTION
New template from Generic was created for Wemos D1. Relays work as expected but configuration is not shown. It happen for pre-compiled version 7.1.2 and same is for self-compiled 7.1.2.6 or 7.2.0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x]  Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WemosD1R2
- [x] Tasmota binary firmware version number used: 7.1.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmota PyFlasher
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
11:05:01 CMD: Backlog Template; Module; GPIO 255
11:05:01 RSL: stat/tasmota/RESULT = {"NAME":"Test","GPIO":[255,255,255,255,22,21,0,0,255,255,52,255,255],"FLAG":0,"BASE":18}
11:05:01 RSL: stat/tasmota/RESULT = {"Module":{"0":"Test"}}
11:05:01 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"22":"Relay2"},"GPIO5":{"21":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"52":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
Create new template

### EXPECTED BEHAVIOUR
Modul configuration shows pins settings like standard modules

### SCREENSHOTS
Generic module configured for two relay and Led
[image]

Based on this new template Test created
[image]

Template is activated
[image]

Configuration of module doesn't show setting of pins. It looks like it is shown only unused pins because GPIO 4,5,14 are missing
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2019-12-19:
It works as designed, see https://tasmota.github.io/docs/#/Templates?id=creating-your-template

GPIOs which are set to a specific module function will not be show in Configure Module. In Configure Module you wil only see GPIOs set to User (255)

effelle, 2019-12-19:
A template is normally created for devices where you can't swap/add/remove component on selected gpios. 
Since gpios `4`,`5` and `14` are already assigned on your template you will not see them on module page, where just the gpios marked as usable with `user (255)`  or `user(15)` are available.

mkh595, 2019-12-19:
> It works as designed, see https://tasmota.github.io/docs/#/Templates?id=creating-your-template
> 
> GPIOs which are set to a specific module function will not be show in Configure Module. In Configure Module you wil only see GPIOs set to User (255)

I read this documentation but it doesn't say what you wrote. That's why I am confused.
I wanted to create template  to easy copy pins configuration for new devices as they are same. But as I want to see configuration of pins when assembling new device, I have to stay with module Generic and create backlog with pins assignment and store it somewhere.
Anyway thanks for clarification.

effelle, 2019-12-19:
The fastest way to copy the gpios configuration is using `template` command:
```
10:14:00 CMD: template
10:14:00 MQT: tasmota/stat/RESULT = {"NAME":"test ","GPIO":[17,255,255,255,21,22,255,255,255,52,255,255,255],"FLAG":15,"BASE":18}
```
And then use the created json on another device console or paste the result on `configure Other` -->`Template` .
If you want to see all the gpios assigned, including the names, issue `gpio 255`:
```
10:15:54 CMD: gpio 255
10:15:54 MQT: tasmota/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"21":"Relay1"},"GPIO5":{"22":"Relay2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"52":"Led1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for **Tasmota software feature requests and bug reporting**. Troubleshooting and setup assistance is more effective using an interactive forum.

Thanks.

F.

meingraham, 2019-12-19:
@mkh595 

When you create a template, You assign specific components to GPIO and the other GPIO you either set as None (0) or User (255). When you activate the template, The module (0) will display only the pins that are available to be assigned as a peripheral - the pins you specified as User.

closed 2019-12-19 by ascillato2