arendst/Tasmota #7260: 50ms hook guaranteed?

stefanbode, 2019-12-19:
### PROBLEM DESCRIPTION
Hi Theo,

in my shutter implementation, I use the FUNC_EVERY_50_MSECOND hook to control the accelerate decelerate of the stepper motor. Sometimes without any reason, a step is missing and cause some "stress". In this example timestep: `time: 235,` is missing. I already tried to increase CPU to 160MHz without any visible impact. Also commenting out the output does not prevent the "hopping".

Any idea where it does come from and if there is a way to control it. I have a more or less working workaround.

```
17:15:35 SHT: time: 233, velocity 100, minstopway -1565,cur_freq 3128, max_frequency 5000, act_freq_change 62->62, min_runtime_ms 2502, act.pos 1565, next_stop 0, target: 0
17:15:35 SHT: time: 234, velocity 100, minstopway -1503,cur_freq 3066, max_frequency 5000, act_freq_change 62->63, min_runtime_ms 2452, act.pos 1502, next_stop -1, target: 0
17:15:35 SHT: time: 236, velocity 100, minstopway -1442,cur_freq 3003, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 2402, act.pos 1379, next_stop -63, target: 0
17:15:35 SHT: time: 237, velocity 100, minstopway -1378,cur_freq 2935, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 2348, act.pos 1310, next_stop -68, target: 0
17:15:35 SHT: time: 238, velocity 100, minstopway -1314,cur_freq 2867, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 2293, act.pos 1251, next_stop -63, target: 0
17:15:35 SHT: time: 239, velocity 100, minstopway -1253,cur_freq 2799, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 2239, act.pos 1196, next_stop -57, target: 0
17:15:35 SHT: Shutter 2: Real. 1196, Start: 20000, Stop: 0, dir -1, motordelay 80, rtc: 11.95 [s], freq 2731
17:15:35 MQT: hm/status/tasmota/RESULT = {"Shutter2":{"Position":95,"direction":-1}}
17:15:35 SHT: time: 240, velocity 100, minstopway -1192,cur_freq 2731, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 2184, act.pos 1141, next_stop -51, target: 0
17:15:35 SHT: time: 241, velocity 100, minstopway -1134,cur_freq 2663, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 2130, act.pos 1088, next_stop -46, target: 0
17:15:35 SHT: time: 242, velocity 100, minstopway -1077,cur_freq 2595, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 2076, act.pos 1034, next_stop -43, target: 0
17:15:35 SHT: time: 243, velocity 100, minstopway -1021,cur_freq 2527, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 2021, act.pos 984, next_stop -37, target: 0
17:15:35 SHT: time: 244, velocity 100, minstopway -967,cur_freq 2459, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 1967, act.pos 932, next_stop -35, target: 0
17:15:36 SHT: time: 245, velocity 100, minstopway -914,cur_freq 2391, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 1912, act.pos 884, next_stop -30, target: 0
17:15:36 SHT: time: 246, velocity 100, minstopway -863,cur_freq 2323, max_frequency 5000, act_freq_change 62->68, min_runtime_ms 1858, act.pos 835, next_stop -28, target: 0
```

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-19:
I guess it's the same isuue the light fade encounters and I'm afraid it is caused by SDK wifi issues.

In any case I have no real clue why we miss a tick but I can imagine something can hold up for over 50 mSecs

stefanbode, 2019-12-19:
I will check this without WiFi, make no sense but can give some insights.

s-hadinger, 2019-12-19:
Light Fade had jitter because of writing settings into Flash. You should see it with loglevel 3.

stefanbode, 2019-12-19:
This is already loglever 4 and there is nothing viewable in between. Just steps missing

closed 2019-12-19 by stefanbode

s-hadinger, 2019-12-19:
How do you detect the missing step? Is one step overshooting the 50ms or it's exactly one step ignored?

stefanbode, 2019-12-19:
It is completely ignored and therefore easy to detect.

s-hadinger, 2019-12-19:
Sorry, let me rephrase: in your logs above, how do you increment step number. Is it based on the `millis()` function? And does it mean you skip exactly 50ms or do you get a random pause?

I didn't find code that produced the log above.