arendst/Tasmota #7262: Fade feature causes flash writes

TimelessNL, 2019-12-19:
### PROBLEM DESCRIPTION
With all `flash save` options turned **OFF** and `fade` turned **ON** any RGB change is still written to Flash.
### REQUESTED INFORMATION
- ✅ Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- ✅ Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- ✅ Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- ✅ Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- ✅ Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- ✅ Device used (e.g., Sonoff Basic): Any devices with at least 1 PWM channel
- ✅ Tasmota binary firmware version number used: 7.1.2
  - ⛔️ Pre-compiled
  - ✅ Self-compiled
    - ✅ IDE / Compiler used: ArduinoIDE SDK 2.6.1
- ✅ Flashing tools used: Web OTA
- ✅ Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
00:18:37 CMD: Backlog Template; Module; GPIO 255
00:18:37 RSL: RESULT = {"NAME":"LSC RGBW LED","GPIO":[0,255,0,255,0,0,0,0,181,0,180,0,0],"FLAG":15,"BASE":18}
00:18:38 RSL: RESULT = {"Module":{"0":"LSC RGBW LED"}}
00:18:38 RSL: RESULT = {"GPIO0":{"0":"Geen"},"GPIO1":{"0":"Geen"},"GPIO2":{"0":"Geen"},"GPIO3":{"0":"Geen"},"GPIO4":{"0":"Geen"},"GPIO5":{"0":"Geen"},"GPIO9":{"0":"Geen"},"GPIO10":{"0":"Geen"},"GPIO12":{"181":"SM2135 Dat"},"GPIO13":{"0":"Geen"},"GPIO14":{"180":"SM2135 Clk"},"GPIO15":{"0":"Geen"},"GPIO16":{"0":"Geen"}}
00:18:38 RSL: STATE = {"Time":"2019-12-20T00:18:38","Uptime":"0T00:00:11","UptimeSec":11,"Vcc":3.512,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Dimmer":28,"Color":"0000001647","HSBColor":"0,0,0","Channel":[0,0,0,9,28],"CT":419,"Scheme":0,"Fade":"ON","Speed":10,"LedTable":"ON","Wifi":{"AP":1,"SSId":"HIDDEN","BSSId":"HIDDEN","Channel":9,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:06"}}
```
- ⛔️ If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
- ⛔️ Provide the output of this command: `Status 0`:
- ⛔️ Provide the output of the Console log output when you experience your issue; if applicable:

### TO REPRODUCE
1. Flash Tasmota 7.1.2
2. Command `reset 5`
3. Command `backlog poweronstate 1; savedata0; setoption0 0; restart 1;`
4. Change the RGB colors and see that there are **no new** flash writes.
5. Command `backlog fade 1; speed 10; restart 1`
6. Change the RGB colors and see that there **are new** flash writes.

### EXPECTED BEHAVIOUR
If `setoption0 0` and `savedata 0` are set no flash writes should occur.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
I prefer to keep smart-lights functional for ordinary light switches, so when they are turned off by cutting the power they should return to their pre-saved RGBW value the next time they are turned on. Basically when the light is turned on it can be controlled thought MQTT but when the power is cut it should restore to it's normal light intensity the next time it's turned on by reapplying power.

ascillato, 2019-12-20:
Already solved. Please update to the latest Tasmota in the development branch (thehackbox.org/tasmota/tasmota.bin) Thanks.

closed 2019-12-20 by ascillato2

s-hadinger, 2019-12-20:
I would rephrase to "soon to be solved" instead of "already solved".

s-hadinger, 2019-12-20:
Now solved. Thanks for reporting.

TimelessNL, 2019-12-20:
Nice work. Thanks!

PR #7266 (2019-12-20): Fix Fade would ignore ``savedata 0`` and store to flash anyways (#7262)
## Description:

**Related issue (if applicable):** fixes #7262

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).