arendst/Tasmota #7273: Scheme 4 improvements

ConsciousCode, 2019-12-20:
**Have you looked for this feature in other issues and in the docs?**  Yes

**Is your feature request related to a problem? Please describe.** 
_A clear and concise description of what the problem is._

In testing I noticed that when `scheme` was set to `4`, it often repeated hues which subjectively looks like it just stops cycling.

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

xdrv_04_light randomly walks up or down in hue, so instead I suggest a simple LCG, rescaling to hue, so the RNG doesn't repeat and "feels" random by appealing to the gambler's fallacy.

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

This could probably be implemented as a rule, but it feels important enough to include as part of the core. I'll update this description if/when I write such a rule.

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2019-12-21:
Hi, please, update to latest Tasmota from the development branch. (http://thehackbox.org/tasmota/tasmota.bin) A lot of improvements has been merged recently. Thanks.

ascillato2, 2020-01-15:
Closing this issue as it has been fixed. Thanks

closed 2020-01-15 by ascillato2