arendst/Tasmota #7276: Add default sorting option to settings

uivc, 2019-12-21:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Would be nice to be able to set the default sorting of the main device list in settings, especially sort by Name, e.g. of you got your devices named with Living Room *, Hallway *, etc. they all come in sequence.

Jason2866, 2019-12-21:
There is no device overwiew in Tasmota?!

uivc, 2019-12-21:
> 
> 
> There is no device overwiew in Tasmota?!

Here's the default overview, I have 12 devices, but they're all sorted by the way I added them. Not sorted after name, like I would prefer to.

Jason2866, 2019-12-21:
You dont have added a picture

effelle, 2019-12-21:
Please close this issue and address this to the [TasmoAdmin](https://github.com/reloxx13/TasmoAdmin/issues) GitHub. 

Thanks.

closed 2019-12-21 by blakadder

Jason2866, 2019-12-21:
This not Tasmota! This is a helper tool for Tasmota. NOT maintained here nor from the author of Tasmota. Please visit the site of the author @shantur  https://github.com/IOTReboot/TasUI/tree/master
Please close this issue because it is wrong here. Thx