arendst/Tasmota #7277: Tasmota 7.2.0 Compile ERROR

helmar74, 2019-12-21:
I tried compiling Tasmota 7.2.0 (master), the following error occured:
Arduino: 1.8.10 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:none OTA:~502KB), 2, nonos-sdk 2.2.1+113 (191105), v2 Higher Bandwidth (no features), Disabled, None, Only Sketch, 115200"

U:\02_daten\27_Programmierung\04_Arduino_ESP8266\06_Projekte_8266_ESPxx\13_SONOFF_Tasmota\02_Sonoff_Standardfirmware\tasmotaStandard\support_crash_recorder.ino: In function 'void custom_crash_callback(rst_info*, uint32_t, uint32_t)':

support_crash_recorder:29:6: error: previous declaration of 'void custom_crash_callback(rst_info*, uint32_t, uint32_t)' with 'C++' linkage

 extern "C" void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end )

      ^

support_crash_recorder:29:102: error: conflicts with new declaration with 'C' linkage

 extern "C" void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end )

                                                                                                      ^

Mehrere Bibliotheken wurden für "ESP8266httpUpdate.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266httpUpdate
Mehrere Bibliotheken wurden für "Ticker.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\Ticker
Mehrere Bibliotheken wurden für "ihx.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\C2Programmer-1.0.0
Mehrere Bibliotheken wurden für "ESP8266WiFi.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WiFi
Mehrere Bibliotheken wurden für "ESP8266mDNS.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266mDNS
Mehrere Bibliotheken wurden für "TasmotaSerial.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\TasmotaSerial-2.4.1
Mehrere Bibliotheken wurden für "Adafruit_SSD1306.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\Adafruit_SSD1306-1.3.0-gemu-1.1
Mehrere Bibliotheken wurden für "EEPROM.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\EEPROM
Mehrere Bibliotheken wurden für "Adafruit_LEDBackpack.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\Adafruit_LED_Backpack-1.1.6
Mehrere Bibliotheken wurden für "bme680.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\BME680_driver-bme680_v3.5.9
Mehrere Bibliotheken wurden für "Wire.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\Wire
Mehrere Bibliotheken wurden für "Adafruit_GFX.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\Adafruit-GFX-Library-1.5.6-gemu-1.0
Mehrere Bibliotheken wurden für "Tsl2561Util.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\Joba_Tsl2561-2.0.10
Mehrere Bibliotheken wurden für "ESP8266WebServer.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer
Mehrere Bibliotheken wurden für "Adafruit_ILI9341.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\Adafruit_ILI9341-1.2.0-Tasmota-1.0
Mehrere Bibliotheken wurden für "NewPing.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\NewPing-1.9.1
Mehrere Bibliotheken wurden für "DNSServer.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\DNSServer
Mehrere Bibliotheken wurden für "PubSubClient.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\PubSubClient-EspEasy-2.7.12
Mehrere Bibliotheken wurden für "FT6236.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\FT6236-gemu-1.0
Mehrere Bibliotheken wurden für "LiquidCrystal_I2C.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\LiquidCrystal_I2C-1.1.3
Mehrere Bibliotheken wurden für "TasmotaModbus.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\TasmotaModbus-1.2.0
Mehrere Bibliotheken wurden für "esp-knx-ip.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\esp-knx-ip-0.5.2
Mehrere Bibliotheken wurden für "ESP8266HTTPClient.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266HTTPClient
Mehrere Bibliotheken wurden für "ArduinoJson.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\ArduinoJson-5.13.4
Mehrere Bibliotheken wurden für "SPI.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\packages\esp8266\hardware\esp8266\2.6.1\libraries\SPI
Mehrere Bibliotheken wurden für "IRremoteESP8266.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\IRremoteESP8266-2.7.1
Mehrere Bibliotheken wurden für "renderer.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\esp-epaper-29-ws-20171230-gemu-1.1
Mehrere Bibliotheken wurden für "NeoPixelBus.h" gefunden
 Benutzt: D:\Users\Helmut\Desktop\arduino-1.8.10_Tasmota_261\portable\sketchbook\libraries\NeoPixelBus-2.5.0.09
exit status 1
previous declaration of 'void custom_crash_callback(rst_info*, uint32_t, uint32_t)' with 'C++' linkage

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Differenz_____
- [X] Tasmota binary firmware version number used: _____7.2.0
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: ARDUINO
- [X] Flashing tools used: OTA_____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

helmar74, 2019-12-21:
Maybe it's an error in Line 29:
`extern "C" void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end )`

arendst, 2019-12-21:
Indeed fails on Arduino IDE.

Works fine in PlarformIO so for now use that as a workaround

helmar74, 2019-12-21:
Ok, how can I use Core 2.6.1 on platformio? I think mine compiles with core 2.3.0
Sorry, but I have no experience with platformio and I don't know how to use it...

arendst, 2019-12-21:
See the docs how to set-up pio.

If you had copied the repository files to the correct location it should have compiled for 2.6.1 as that is the Tasmota default

arendst, 2019-12-21:
Or try as a quick fix the following the change in file support_crash_recorder.ino line 29:
```
extern void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end )
```
so removing the ``"C"``

arendst, 2019-12-21:
Final fix.

Add the following line to file tasmota_post.h just below line 41:
```
extern "C" void custom_crash_callback(struct rst_info * rst_info, uint32_t stack, uint32_t stack_end);
```
This will add the prototype for correct Arduino IDE use.

NB. Leave the original line as is so WITH ``"C"``!

closed 2019-12-21 by arendst

helmar74, 2019-12-21:
Can confirm, that compile with arduino IDE worked