arendst/Tasmota #7280: aqara button v1 zigbee

nguyenaitriet, 2019-12-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

I have done adding aqara button v1 to tasmota zigbee. When 1 click is fine but 2 or 3 click i receive like this.
1 click:
15:10:29 MQT: tele/zigbee001/SENSOR = {"ZigbeeReceived":{"0x368F":{"Power":false,"LinkQuality":98}}}
15:10:29 MQT: tele/zigbee001/SENSOR = {"ZigbeeReceived":{"0x368F":{"Power":true,"LinkQuality":96}}}
bouble click:
15:11:10 MQT: tele/zigbee001/SENSOR = {"ZigbeeReceived":{"0x368F":{"0006/8000":2,"LinkQuality":70}}}
3 clicks:
15:11:10 MQT: tele/zigbee001/SENSOR = {"ZigbeeReceived":{"0x368F":{"0006/8000":3,"LinkQuality":70}}}
The question is: what is 0006/8000? and how i can change it to power like 1 click
like this

15:11:10 MQT: tele/zigbee001/SENSOR = {"ZigbeeReceived":{"0x368F":{"Power":1,"LinkQuality":70}}}
15:11:10 MQT: tele/zigbee001/SENSOR = {"ZigbeeReceived":{"0x368F":{"Power":2,"LinkQuality":70}}}
15:11:10 MQT: tele/zigbee001/SENSOR = {"ZigbeeReceived":{"0x368F":{"Power":3,"LinkQuality":70}}}
thanks

ascillato, 2019-12-21:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

s-hadinger, 2019-12-21:
Thanks for reporting this issue.

`0006/8000` means cluster:

- Cluster = `0x0006`: On/Off cluster
- Attribute = `0x8000`: proprietary ZCL attribute.

I will submit a change to have `0006/8000` reported as `Power` too.

With this change, it will report either: `false`, `true`, `2`, and `3`.

Would you be kind enough to set `WebLog 3` and copy back the logs when receiving double/triple clicks (I want to know the Manuf code if any), and do `ZigbeeStatus3`

closed 2019-12-21 by arendst

nguyenaitriet, 2019-12-22:
thanks. It work like a charm.

I did WebLog 3
and this is double clicks:

03:38:58 ZIG: ZigbeeZNPReceived 4481000006008F360101007A0069CB2B00000718220A008020028F361D
03:38:58 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":34,"cmdid":"0x0A","payload":"00802002"}}
03:38:58 ZigbeeZCLRawReceived: {"ZigbeeReceived":{"0x368F":{"0006/8000":2}}}
03:38:58 RSL: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x368F":{"Power":2,"LinkQuality":122}}}

triple clicks:
03:40:39 ZIG: ZigbeeZNPReceived 4481000006008F360101005800DD963000000718230A008020038F361D
03:40:39 {"ZigbeeZCLReceived":{"groupid":0,"clusterid":6,"srcaddr":"0x0000","srcendpoint":0,"dstendpoint":0,"wasbroadcast":0,"LinkQuality":0,"securityuse":0,"seqnumber":0,"timestamp":0,"fc":"0x18","manuf":"0x0000","transact":35,"cmdid":"0x0A","payload":"00802003"}}
03:40:39 ZigbeeZCLRawReceived: {"ZigbeeReceived":{"0x368F":{"0006/8000":3}}}
03:40:39 RSL: tele/tasmota/SENSOR = {"ZigbeeReceived":{"0x368F":{"Power":3,"LinkQuality":88}}}

03:41:31 CMD: ZigbeeStatus3
03:41:31 SRC: WebConsole from 192.168.31.41
03:41:31 CMD: Group 0, Index 3, Command "ZIGBEESTATUS", Data ""
03:41:31 RSL: stat/tasmota/RESULT = {"ZigbeeStatus3":[{"Device":"0x368F","IEEEAddr":"00158D0002E27BBC","ModelId":"lumi.sensor_switch","Endpoints":{}}]}

s-hadinger, 2019-12-22:
Thanks. Manuf id is indeed zero.

PR #7284 (2019-12-21): Zigbee aqara switch, support for double/triple clicks
## Description:

Report `00006/8000` Aqara Switch proprietary attribute as `Power` attribute.

Possible values are: `false`/`true` (1 click), `2` (double click), `3` (triple click)

**Related issue (if applicable):** fixes #7280 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
   [... cut]