arendst/Tasmota #7281: Sonoff RFBridge not reporting any signal from 2 different PIR

SeLLeRoNe, 2019-12-21:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
RFBridge is not "capturing" (or at least reporting) RF signal from 2 PIR sensors.
1 of those two was working pre-update, the other one was surely working on 6.9.x release, I just noticed it is not working with 7.1.2 and 7.2.0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

7.1.2
16:09:00 CMD: status 0;
16:09:00 MQT: stat/RFBridge01/STATUS = {"Status":{"Module":25,"FriendlyName":["RF Bridge"],"Topic":"RFBridge01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
16:09:00 MQT: stat/RFBridge01/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:08:06","StartupUTC":"2019-12-21T15:00:54","Sleep":50,"CfgHolder":4617,"BootCount":1601,"SaveCount":2207,"SaveAddress":"FA000"}}
16:09:00 MQT: stat/RFBridge01/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-06T11:22:22","Boot":6,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285"}}
16:09:00 MQT: stat/RFBridge01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008229","2805C8000000060000005AFF000000000000","10000000","00000000"]}}
16:09:00 MQT: stat/RFBridge01/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:09:01 MQT: stat/RFBridge01/STATUS5 = {"StatusNET":{"Hostname":"RFBridge01","IPAddress":"","Gateway":".1","Subnetmask":"255.255.0.0","DNSServer":"","Mac":"","Webserver":2,"WifiConfig":5}}
16:09:01 MQT: stat/RFBridge01/STATUS6 = {"StatusMQT":{"MqttHost":"casa.andrea.iannucci.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B2A7E3","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:09:01 MQT: stat/RFBridge01/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 21 15:09:01 2019","Local":"Sat Dec 21 16:09:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:55"}}
16:09:01 MQT: stat/RFBridge01/STATUS10 = {"StatusSNS":{"Time":"2019-12-21T16:09:01"}}
16:09:01 MQT: stat/RFBridge01/STATUS11 = {"StatusSTS":{"Time":"2019-12-21T16:09:01","Uptime":"0T00:08:07","UptimeSec":487,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"Wifi":{"AP":1,"SSId":"","BSSId":"48:00:33:79:EA:1F","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:09"}}}

7.2.0
16:17:55 CMD: status 0;
16:17:55 MQT: stat/RFBridge01/STATUS = {"Status":{"Module":25,"FriendlyName":["RF Bridge"],"Topic":"RFBridge01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
16:17:55 MQT: stat/RFBridge01/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:39","StartupUTC":"2019-12-21T15:17:16","Sleep":50,"CfgHolder":4617,"BootCount":1602,"SaveCount":2210,"SaveAddress":"FA000"}}
16:17:55 MQT: stat/RFBridge01/STATUS2 = {"StatusFWR":{"Version":"7.2.0(tasmota)","BuildDateTime":"2019-12-21T12:54:33","Boot":6,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"304/1151"}}
16:17:55 MQT: stat/RFBridge01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008229","2805C8000000060000005AFF000000000000","10000000","00000000"]}}
16:17:55 MQT: stat/RFBridge01/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:17:55 MQT: stat/RFBridge01/STATUS5 = {"StatusNET":{"Hostname":"RFBridge01","IPAddress":"","Gateway":"","Subnetmask":"255.255.0.0","DNSServer":"","Mac":"","Webserver":2,"WifiConfig":5}}
16:17:55 MQT: stat/RFBridge01/STATUS6 = {"StatusMQT":{"MqttHost":"casa.andrea.iannucci.net","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B2A7E3","MqttUser":"","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:17:55 MQT: stat/RFBridge01/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 21 15:17:55 2019","Local":"Sat Dec 21 16:17:55 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:40","Sunset":"16:55"}}
16:17:55 MQT: stat/RFBridge01/STATUS10 = {"StatusSNS":{"Time":"2019-12-21T16:17:55"}}
16:17:56 MQT: stat/RFBridge01/STATUS11 = {"StatusSTS":{"Time":"2019-12-21T16:17:55","Uptime":"0T00:00:39","UptimeSec":39,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"","BSSId":"48:00:33:79:EA:1F","Channel":1,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:09"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Updated from 7.1.2 to 7.2.0 (please note, I already saw this problem on 7.1.2.6 dev updating from 7.1.2.4 or .3 dev)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
See RF signals in console and report them to MQTT

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-21:
Could you add the logs when the PIR fired?

SeLLeRoNe, 2019-12-21:
That's the problem, it doesn't show anything on the console on the newer version.

For the 7.1.2 this is what I got:
```
16:29:27 MQT: tele/RFBridge01/RESULT = {"Time":"2019-12-21T16:29:27","RfReceived":{"Sync":12390,"Low":410,"High":1210,"Data":"E8447E","RfKey":"None"}}
16:29:33 MQT: stat/RFBridge01/STATUS = {"Status":{"Module":25,"FriendlyName":["RF Bridge"],"Topic":"RFBridge01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":1,"PowerRetain":1}}
```

turin400, 2019-12-21:
test this:

SerialConfig 8N1

arendst, 2019-12-21:
Just to confirm that indeed you will need have to execute command ``SerialConfig 8N1`` once to get the bridge going again.

After the command execute a restart with ``restart 1`` and all should be fine again (at least that was what happened to me just now ;-)

I'll check if a better solution can be made and where it went wrong.

closed 2019-12-21 by arendst

PrathikGopal, 2019-12-21:
> Just to confirm that indeed you will need have to execute command `SerialConfig 8N1` once to get the bridge going again.
>
> After the command execute a restart with `restart 1` and all should be fine [...]

This worked for me!

SeLLeRoNe, 2019-12-21:
Hi there,

Thanks for the command, that actually worked with one of the 2 PIR.

The second (still a Sonoff but somehow a little smoller), even it recon movment (meaning that the green light goes on), still the signal doesn't seems to be "received" by the RF Bridge.

Unfortuantely, this is still not working (it was but I think on 6.9, I cannot be sure it did work since 7.X.X release because I set and forget while the other one automate my patio light, so I can "see" when it does his job.

Any idea?

Thanks
Andrea

SeLLeRoNe, 2019-12-21:
It actually started to work now.
I had it next to me so it has been triggered a few times and now for some unknown reason it started to work!

Thanks