arendst/Tasmota #7282: v7.2.0: No Web Setup or serial communication

jeroenrnl, 2019-12-21:
### PROBLEM DESCRIPTION
I've flashed a couple of new Shellys (Shelly1 v3 + Shelly2.5) with v7.2.0 but it didn't show an AP to connect to and on the serial terminal I see only this:
`
00:00:00 WIF: Connecting to AP1  in mode 11N as tasmota-7798...                
00:00:16 WIF: Connect failed with AP timeout                                   
00:00:16 WIF: Connecting to AP2  in mode 11N as tasmota-7798...                
00:00:32 WIF: Connect failed with AP timeout
00:00:34 WIF: Connecting to AP2  in mode 11N as tasmota-7798...
`
Typing on the serial terminal doesn't do anything.

I've downgraded to 7.1.2 and that works flawless

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic):Shelly1, Shelly2.5
- [X] Tasmota binary firmware version number used: 7.2.0
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: esptool.py
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  <no output, no reaction on serial console>

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
  <no output, no reaction on serial console>

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
  <no output, no reaction on serial console>

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
  <no output, no reaction on serial console>

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
esptool.py --port /dev/ttyUSB0 erase_flash
esptool.py --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 tasmota.bin

<check wifi>
<check serial console>
### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
configuration AP or possibility to set via console

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Danny02, 2019-12-21:
Hi, first time user here. Just "bricked" two sonoff minis myself. seems to be same problem. The blue LED of the mini is just blinking maybe 500ms off 500ms on.

jeroenrnl, 2019-12-21:
Mine weren't bricked, simply reflashing them with other software fixed it, at least for one, reflashing the others now, but I'm quite convinced they will be fine too.

arendst, 2019-12-21:
Verified. Will release updated files.

closed 2019-12-21 by arendst

jeroenrnl, 2019-12-21:
I reflashed one with the updated release and confirm: issue fixed. Thanks for the amazing service (40 minutes from bug report to fix!!!)