arendst/Tasmota #7285: downgrade from 7.2

theisgroup, 2019-12-21:
### PROBLEM DESCRIPTION
How to Downgrade from 7.2

### REQUESTED INFORMATION
i upgraded my device through OTA url to 7.2 and ment to just get to 7.1.2
ive install minimal.bin, but can not install minimal.bin from 7.1.2
how do i downgrade?

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-21:
Use command ``Setoption78 1`` and try again.

theisgroup, 2019-12-21:
thanks that did the trick. 
i have another issue. on another switch i tried the "reset config" button. i can't get back into that switch anymore. not sure where to start. i can open a new issue

closed 2019-12-21 by theisgroup

meingraham, 2019-12-22:
@arendst 

`SetOption78 - Disable OTA compatibility check`

Is SetOption78 still available or does the fix remove the need for it or deprecates it? If not deprecated, what is the "compatibility check" that gets disabled? Will add to the docs but need to know what it is that's being disabled.

arendst, 2019-12-23:
@meingraham See my comment in private message on Discord. We can discuss there further if you like.