arendst/Tasmota #7294: "basic" still in "Program Version" of 7.2.0

TobiasBg, 2019-12-22:
### PROBLEM DESCRIPTION

After just OTA-updating to 7.2.0-lite (with OTAUrl http://thehackbox.org/tasmota/release/tasmota-lite.bin), on a device that used 7.1.1.1-basic before that, I see `7.2.0(basic)` as the "Program Version" on http://my-sonoff/in?.
It's not really a bug, but I was expecting to see `7.2.0(lite)` after the recent renaming of basic to lite.
Maybe another string needs to be changed somewhere?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-22:
How can I resist someone so perseverant to change basic to lite ;-)

Will change asap.

arendst, 2019-12-22:
Done.

Thx for reporting.

closed 2019-12-22 by arendst

TobiasBg, 2019-12-22:
> How can I resist someone so perseverant to change basic to lite ;-)

I just had to beat that auto-close-bot ;-)

Thanks!