arendst/Tasmota #7295: Sensor MHZ19B after deep sleep does not have time to initialize

dejavy, 2019-12-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ok ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ok] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ok ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ok ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ok] Device used (e.g., Sonoff Basic): nodeMCU ESP8266
- [ ] Tasmota binary firmware version number used: Tasmota Version 7.1.2(tasmota)-2_5_2
  - [ ] Pre-compiled
  - [ ok] Self-compiled
    - [ok ] IDE / Compiler used: Arduino IDE
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  `14:01:09 CMD: Backlog Template; Module; GPIO 255
14:01:09 MQT: stat/MHZ19B/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
14:01:09 MQT: stat/MHZ19B/RESULT = {"Module":{"18":"Generic"}}
14:01:09 MQT: stat/MHZ19B/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"60":"MHZ Tx"},"GPIO2":{"0":"None"},"GPIO3":{"61":"MHZ Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6":"I2C SDA"},"GPIO13":{"0":"None"},"GPIO14":{"5":"I2C SCL"},"GPIO15":{"52":"Led1"},"GPIO16":{"182":"DeepSleep"}}`

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
`14:02:25 CMD: Status 0
14:02:25 MQT: stat/MHZ19B/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"MHZ19B","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":8,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:02:25 MQT: stat/MHZ19B/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Power on","Uptime":"0T00:39:54","StartupUTC":"2019-12-22T12:22:31","Sleep":50,"CfgHolder":4617,"BootCount":82,"SaveCount":165,"SaveAddress":"3F7000"}}
14:02:25 MQT: stat/MHZ19B/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-20T22:50:27","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)","Hardware":"ESP8266EX"}}
14:02:25 MQT: stat/MHZ19B/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MineRoute",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C8000000060000005A00000000000000","00000000","00000000"]}}
14:02:25 MQT: stat/MHZ19B/STATUS4 = {"StatusMEM":{"ProgramSize":580,"Free":3492,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":2,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
14:02:25 MQT: stat/MHZ19B/STATUS5 = {"StatusNET":{"Hostname":"MHZ19B-5479","IPAddress":"10.10.1.127","Gateway":"10.10.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.10.1.1","Mac":"BC:DD:C2:24:95:67","Webserver":2,"WifiConfig":4}}
14:02:25 MQT: stat/MHZ19B/STATUS6 = {"StatusMQT":{"MqttHost":"77.122.171.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_249567","MqttUser":"admin","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
14:02:25 MQT: stat/MHZ19B/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 22 13:02:25 2019","Local":"Sun Dec 22 14:02:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:56"}}
14:02:25 MQT: stat/MHZ19B/STATUS10 = {"StatusSNS":{"Time":"2019-12-22T14:02:25","ANALOG":{"A0":915},"BH1750":{"Illuminance":38},"MHZ19B":{"Model":"B","CarbonDioxide":4067,"Temperature":26.0},"TempUnit":"C"}}
14:02:25 MQT: stat/MHZ19B/STATUS11 = {"StatusSTS":{"Time":"2019-12-22T14:02:25","Uptime":"0T00:39:54","UptimeSec":2394,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"MineRoute","BSSId":"00:18:E7:FD:8C:B2","Channel":11,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:07"}}}`

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

dejavy, 2019-12-22:
This CO2 sensor "MHZ19B" needs some time after switching on, so that it correctly measures the readings.

`13:22:40 MQT: tele/MHZ19B/INFO3 = {"RestartReason":"Power on"}
13:22:44 MQT: tele/MHZ19B/SENSOR = {"Time":"2019-12-22T13:22:44","ANALOG":{"A0":929},"BH1750":{"Illuminance":167},"MHZ19B":{"Model":"","CarbonDioxide":0,"Temperature":0.0},"TempUnit":"C"}`

It can be seen that the sensor "BH1750" immediately has the correct reading, and "MHZ19B" takes time (about 1-2 minutes)

`13:23:04 MQT: tele/MHZ19B/SENSOR = {"Time":"2019-12-22T13:23:04","ANALOG":{"A0":929},"BH1750":{"Illuminance":168},"MHZ19B":{"Model":"B","CarbonDioxide":410,"Temperature":22.0},"TempUnit":"C"}`

`13:23:54 MQT: tele/MHZ19B/SENSOR = {"Time":"2019-12-22T13:23:54","ANALOG":{"A0":929},"BH1750":{"Illuminance":159},"MHZ19B":{"Model":"B","CarbonDioxide":2184,"Temperature":23.0},"TempUnit":"C"}`

The problem is that when using the "deep sleep" mode, after waking up, he takes only one measurement and it has a zero value. All sensors have time to turn on, and have the correct readings, but this "MHZ19B" is not.
Changes in the parameter "teleperiod" does not change the situation.

`13:19:32 MQT: tele/MHZ19B/INFO1 = {"Module":"Generic","Version":"7.1.2(tasmota)","FallbackTopic":"cmnd/DVES_249567_fb/","GroupTopic":"cmnd/tasmotas/"}
13:19:32 MQT: tele/MHZ19B/INFO2 = {"WebServerMode":"Admin","Hostname":"MHZ19B-5479"}
13:19:32 MQT: tele/MHZ19B/INFO3 = {"RestartReason":"Power on"}
13:19:37 MQT: tele/MHZ19B/SENSOR = {"Time":"2019-12-22T13:19:37","ANALOG":{"A0":930},"BH1750":{"Illuminance":4},"MHZ19B":{"Model":"","CarbonDioxide":0,"Temperature":0.0},"TempUnit":"C"}
13:19:37 DSL: Reset wrong settings wakeup: 0, slip 0
13:19:37 MQT: stat/MHZ19B/RESULT = {"DeepSleep":{"Time":"2019-12-22T13:20:00","Epoch":1577017200}}`

I understand that this can be solved somehow by the "Rules".
To postpone the survey for a while, wait until the sensor wakes up. But I am at a loss for understanding the "rules"

stefanbode, 2019-12-22:
As far as I remember the deepsleep is initiated earliest after teleperiod. Therefore setting teleperiod 120 should solve your problem. In this case the deepsleep will do a teleperiod after 2minutes and then go into deepsleep. Pleas can you check and provide logs with log level 4 if it does not solve your problen

stefanbode, 2019-12-22:
Can you comment what you mean with “teleperiod did not change the problem”? Does the device still get into deepsleep immediately or does it go into sleep after 2 minutes and the device still send 0 values. In this case i assume we need to send the initiate command to the device earlier. Looking forward to hear from you

dejavy, 2019-12-22:
I also immediately thought that the "teleperiod" parameter should solve my problem. But changing the “teleperiod” to 120 or 300 does not change the result.
He immediately sends the value of all sensors, he does not wait 120 seconds. This can be seen from the console log in time (shotshot)

[image]
[image]

dejavy, 2019-12-22:
I probably realized what the problem is.
I use the bc337 transistor to turn off all sensors during deep sleep. I turn it on when I wake up ("GPIO15": {"52": "Led1"}) - the question is this, does it give a GPIO15 signal as soon as it wakes up, or is it expecting a "teleperiod  120"?

arendst, 2019-12-22:
After a restart, albeit power on, restart or deepsleep there will always be a tele message send within some seconds after the wifi connectes. Once send deepsleep will start some seconds later.

As the goal of deepsleep is to save power this cycle is made as short as possible. I don't think the MZ19 sensor is a candidate for batterypowered devices. The sensor itself also uses a lot of power.

stefanbode, 2019-12-22:
Hi Theo, this was the reason why the original deepsleep waits for teleperiod and this was send first time e.g. 300 seconds after wakeup. Even if the device is not best for this case, we are now into trouble to configure a suitable uptime. I assume the only workaround currentY is to define one rule to disable deepsleep on boot and the. After 2minuter after boot have another rule sending the device to deepsleep.

stefanbode, 2019-12-22:
The gPio15 is controlled by the relay. If you power it down you must define that it should power on on restart. There is a config setting for this behavior. The default is to start with the saved state. This is not what you want to have.

arendst, 2019-12-22:
In that case why not re-implement the deepslseep start after the second teleperiod message? I think that's the most versatile solution. In worst case it adds 10 seconds more uptime due to minimal teleperiod.

stefanbode, 2019-12-22:
Hi Theo, yes make sense. Should be easily done through a if teleperiod >10. In this case with 10 everything is as fast as possible and with 11 and more you can control for a second teleperiod

dejavy, 2019-12-22:
I use GPIO15 because when the device sleeps "signal strength is LOW"
[image]

And when I wake up I turn it on "" GPIO15 ": {" 52 ":" Led1 "}"
The question remains - does it translate the "GPIO15 port from 0 to 1" as soon as it wakes up, and thereby connects all the sensors, or first it waits for the "teleperiod 120"?
If at once - then the MHZ19B sensor has a time of 120 seconds to start reading the correct readings.
If not immediately, but after a “teleperiod 120” time has elapsed, then the MHZ19B sensor has no chance.

stefanbode, 2019-12-22:
In this case it should work. The tasmota state is on.  and because of the internal resistors it goes to off directly after deepsleep. With a wakeup it should go on immediately. Can you check on the web interface if the relay status remains at ON after 2-3 deepsleeps cycle. Just to ensure we have no side effect. There is another interesting setting.....

stefanbode, 2019-12-22:
Setoption63 1
Is the magic key if you relay state go to OFF after some deepsleep cycles. It is very likely with the bc337

dejavy, 2019-12-22:
So.
I noticed a couple of nuances.
Added an external LED for visual observation.

`GPIO2":{"56":"Led1i"}`  - LED
`GPIO15":{"53":"Led2"}` - bc337

Parameters
`"DeepSleepTime":"120"`
`"Teleperiod":"120"`
[image]

Parameters
`"DeepSleepTime":"300"`
`"Teleperiod":"300"`
[image]

the "Teleperiod" parameter does not change the wake time

stefanbode, 2019-12-22:
I think with the `setoption63 1` you can avoid the GPIO probing and then the LED should stay ON.

dejavy, 2019-12-22:
This option did not change the situation setoption63 1.
The obvious problem is that after waking up, he immediately sends the sensor values (10-15 seconds awake) and then falls asleep.
the “teleperiod” parameter does not change the device’s activity time.

dejavy, 2019-12-22:
I already checked this code. Yes, he solved this problem! It's fine.

dejavy, 2019-12-22:
But I have a few suggestions.
After waking up, it immediately sends the data from the sensors, but they are not valid, the values ​​may be absent or zero, the sensor may not be initialized at all.

Variant of how I imagine it:

1) a parameter that sets the time in seconds before sending the first message (so that all sensors are prepared and give correct readings)
2) parameter setting the number of messages sent during the period of activity (awakening)

I will explain.
Our device is powered by a battery.
Wakes up every 3 days, connects its sensors and sensors to power (they are also powered by a battery).
It expects 3 minutes for all sensors to go into working mode (some need time, for example, it can be a full-fledged microcontroller, which takes time from 2-5 minutes to turn on).
Performs several measurements 1-10 with the interval "teleperiod" (this is necessary so that on the sarver you can calculate the average values ​​or check the validity or keep the maximum).
And only after that he again goes into a deep sleep.

stefanbode, 2019-12-22:
Isn’t it possible that you just filter the 0 values out of the result. It might be very complicated to avoid the first teleperiod with the current implementation.

dejavy, 2019-12-22:
My skill is a little more in microelectronics (sensors, modules) than in programming, although I also understand a little about this.
I do not think that it is a difficult task to add a "timer" before the first survey of sensors after switching on.
The problem is that the value of the sensors can go directly (bypassing the server, where you can filter the zero value). The sensor value will not necessarily be 0, they may be different from 0 but not true for various reasons. For example, the sensor did not warm up, or not until the end flared up completely in a powerful halogen lamp.

stefanbode, 2019-12-23:
The issue is caused by the MQTT module. This send the teleperiod 5 seconds in the future. Line 567ff in xdrv02_mqtt.ini. When changing it to something like this it will behave the same for teleperiod 10 and respect longer teleperiod time.
```
    if (Settings.tele_period == 10) {
      tele_period = Settings.tele_period -5;  // Enable TelePeriod in 5 seconds
    } else {
      tele_period = 0;
    }
```

dejavy, 2019-12-23:
Good.
We can consider adding two parameters
"Delayed start of the first message" - from 0 to 600 seconds as an option.
"Number of messages in Deep Sleep mode" - 1 - 10 messages.

I am ready to test.

arendst, 2019-12-23:
I suggest to add a moneymaker too.

I am ready to test.

We don't do that as this will break many sensor specific codes implemented over time. Your only issue is the slow MHZ19. So as stefan suggested we'll do some tweaking around teleperiod and that's it. Should be enough. Your data handler may filter out the correct information.

dejavy, 2019-12-23:
In this case, I'm not talking about a problem with the "MHZ19" sensor.
That is, it does not concern the problem at all.

I'm talking about a small extension of functionality.
1) the ability to delay the start of the first survey of all sensors
2) when using the Deep Sleep mode, the ability to send more than one message.

I am ready to participate in the refinement of this functionality.
Tell me in which files to watch the logic.

stefanbode, 2019-12-23:
Gents, agree to Theo, we have stay compatible with current functionality. Can you compile you own version with the change in mqtt above and tell us if this solves you problem. I assume so

ascillato, 2019-12-26:
@dejavy

Hi,

Any news on this?

dejavy, 2019-12-26:
Hello.
I think I need your help at the initial stage.
When I add a new parameter to the structure

`struct SYSCFG {`
`uint16_t      cfg_holder;              // 000 v6 header`
`uint16_t      cfg_size;                  // 002`
`unsigned long save_flag;           // 004`
`//...`
`//...`
`uint16_t      tele_period_aside;`
`} Settings;`

It does not save the settings for connecting to my network.
The result is a reset device, and all over again.

[image]

I have assumptions that the problem is the lack of memory space, I blocked unused parameters (this is only for the test) - but this also did not produce results.

[image]

I need your help.
I'm missing something

ascillato, 2019-12-26:
When you modify the memory map, all settings got scrambled. You need to erase all flash and then flashing your custom version with your custom settings memory map.

dejavy, 2019-12-26:
This goes without saying.

dejavy, 2019-12-26:
It is enough for me to block this line and everything again works correctly.

[image]

dejavy, 2019-12-26:
I understand that in addition to adding this variable, something else needs to be changed.
But I don’t find where to change (add) and what to change (add)
This code is very large and little documented.

ascillato, 2019-12-27:
Were you erasing all flash BEFORE flashing by serial your custom version?

dejavy, 2019-12-27:
Yes

stefanbode, 2019-12-27:
Of you add a Parameter you Lust do this before the free_ea4 paramerter. This first. That you have to decrease the number of bytes you add by your change. Here from 326 to 324 because the uint_16 uses two bytes. That’s the mechanic

arendst, 2019-12-27:
Change settings.h as follows:
```
  uint8_t       mqttlog_level;             // F01
  uint8_t       sps30_inuse_hours;         // F02

  uint8_t       free_f03[1];               // F03

  uint16_t      tele_period_aside;         // F04

  uint8_t       free_f06[230];             // F06

  uint32_t      i2c_drivers[3];            // FEC I2cDriver
  uint32_t      cfg_timestamp;             // FF8
  uint32_t      cfg_crc32;                 // FFC
} Settings;
```
and your good to go.

dejavy, 2019-12-27:
Yes, thank you.
I understood.
I already have something worthwhile coming out.

stefanbode, 2020-01-05:
There where many changes in the code to support this. Is the issue now closed?

ascillato2, 2020-01-11:
Closing as there is no feedback. Please, report back when you have time to test. Thanks.

closed 2020-01-11 by ascillato2