arendst/Tasmota #7297: "Sleep" command has no effect.

huckes, 2019-12-22:
### PROBLEM DESCRIPTION
Sending a "Sleep"-command from the console seems to have no effect on the system. 
Eg. : 
19:05:11 CMD: Sleep 50
19:05:11 RSL: RESULT = {"Sleep":{"50":{"Active":"50"}}}
19:05:21 CMD: Sleep
19:05:21 RSL: RESULT = {"Sleep":{"50":{"Active":"10"}}}

This issue occurs independly of the SleepMode. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Luminea-ZX-2880 (equal toMirabella Genio Bulb)
- [x ] Tasmota binary firmware version number used: 7.2.0 2019.12.22 11:45:04
  - [x ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
19:12:31 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"37":"PWM1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"38":"PWM2"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:14:36 CMD: STATUS 0
19:14:36 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["Luminea-Gartenleuchte"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:14:36 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:52:18","StartupUTC":"2019-12-22T17:22:18","Sleep":50,"CfgHolder":4617,"BootCount":39,"SaveCount":393,"SaveAddress":"F7000"}}
19:14:36 RSL: STATUS2 = {"StatusFWR":{"Version":"7.2.0(tasmota)","BuildDateTime":"2019.12.22 11:45:04","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"312/1151"}}
19:14:36 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Ebenhoch",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","2805C8000100060000005A00000000000000","00000401","00000000"]}}
19:14:36 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":432,"Heap":28,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000407","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
19:14:36 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-3955","IPAddress":"192.168.178.158","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"50:02:91:45:0F:73","Webserver":2,"WifiConfig":4}}
19:14:36 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 22 18:14:36 2019","Local":"Sun Dec 22 19:14:36 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:56"}}
19:14:36 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-12-22T19:14:36"}}
19:14:36 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-12-22T19:14:36","Uptime":"0T00:52:18","UptimeSec":3138,"Heap":28,"SleepMode":"Normal","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":69,"Color":"00B0","HSBColor":"0,0,0","Channel":[0,69],"CT":500,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Ebenhoch","BSSId":"E0:28:6D:C8:E2:CF","Channel":6,"RSSI":44,"Signal":-78,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
19:16:40 CMD: sleep 50
19:16:40 SRC: WebConsole from 192.168.178.34
19:16:40 CMD: Gruppe 0, Index 1, Befehl "SLEEP", Daten "50"
19:16:40 RSL: RESULT = {"Sleep":{"50":{"Active":"50"}}}
19:16:45 WIF: Prüfe Verbindung...
19:16:45 WIF: verbunden
19:16:47 CMD: sleep
19:16:47 SRC: WebConsole from 192.168.178.34
19:16:47 CMD: Gruppe 0, Index 1, Befehl "SLEEP", Daten ""
19:16:47 RSL: RESULT = {"Sleep":{"50":{"Active":"10"}}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Use console and send command "Sleep 50"
- send command "Sleep"

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The reported "sleep"-value should be equal to the command parameter.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2019-12-22:
the sleep command has an effect and lowers down CPU usage. If you want to send your device into deepsleep then please search the wiki for this functionality

huckes, 2019-12-22:
Thanks for comment. I don't want to bring the device into deepsleep. The aim is only to lower the idle power consumption by using th sleep-command to optimize the power management as described in https://tasmota.github.io/docs/#/Dynamic-Sleep?id=dynamic-sleep-cpu-main-loop-target-cpu-power-management.

Jason2866, 2019-12-22:
You have a PWM enabled. If PWM is active user setting of sleep is overriden by 10 to ensure
smooth working of light driver functions.

ascillato2, 2019-12-22:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-22 by ascillato2

huckes, 2019-12-23:
Thanks a bunch for the fast clarification.
Merry Christmas and a Happy Nuew Year,
Hans

jeroenvermeulen, 2025-02-10:
This also happens when you are using LVGL.
I am not using any touch buttons, so I did set in `user_config_override.sh`:
```
#undef USE_LVGL_MAX_SLEEP
#define USE_LVGL_MAX_SLEEP 250
```
Which allowed me to set `Sleep 250` and drastically reduce the CPU load.