arendst/Tasmota #7299: SK6812 RGBW LED Strip not working after upgrade to 7.1.2

ameennihad, 2019-12-22:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have ESP12 with SK6812 RGBW strip (RGB + WW), it's been working fine on Tasmota 6.5, after updating to version 7.1.2 the LED color are white ,WW channel is always off, around 40 of the 60 LEDs are controllable, individual control of LEDs is not possible, only brightness control is working.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): generic
- [x] Tasmota binary firmware version number used: 7.1.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: PlatformIO
- [x] Flashing tools used: PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
22:43:01 MQT: stat/ESP_2A63AC/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":1,"BASE":18}
22:43:02 MQT: stat/ESP_2A63AC/RESULT = {"Module":{"18":"Generic"}}
22:43:02 MQT: stat/ESP_2A63AC/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"7":"WS2812"},"GPIO5":{"2":"AM2301"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"29":"Relay1i"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
22:39:24 MQT: stat/ESP_2A63AC/STATUS = {"Status":{"Module":18,"FriendlyName":["Livingroom WW LED","Livingroom RGB LED"],"Topic":"ESP_2A63AC","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://esp.home.lan/index.php","RestartReason":"Software/System restart","Uptime":"2T09:10:13","StartupUTC":"2019-12-20T10:29:11","Sleep":50,"CfgHolder":4617,"BootCount":107,"SaveCount":2149,"SaveAddress":"F5000"}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-20T12:52:46","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX"}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["home.lan",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008029","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS4 = {"StatusMEM":{"ProgramSize":562,"Free":440,"Heap":23,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS5 = {"StatusNET":{"Hostname":"ESP_2A63AC-0940","IPAddress":"192.168.8.151","Gateway":"192.168.8.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.8.4","Mac":"B4:E6:2D:2A:63:AC","Webserver":2,"WifiConfig":4}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.home.lan","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2A63AC","MqttUser":"MQTT_User","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS7 = {"StatusTIM":{"UTC":"Sun Dec 22 19:39:24 2019","Local":"Sun Dec 22 22:39:24 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+03:00","Sunrise":"07:10","Sunset":"16:52"}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS10 = {"StatusSNS":{"Time":"2019-12-22T22:39:24","AM2301":{"Temperature":21.5,"Humidity":54.4},"TempUnit":"C"}}
22:39:24 MQT: stat/ESP_2A63AC/STATUS11 = {"StatusSTS":{"Time":"2019-12-22T22:39:24","Uptime":"2T09:10:13","UptimeSec":205813,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Dimmer":100,"Color":"FF0004","HSBColor":"359,100,100","Channel":[100,0,2],"Scheme":0,"Width":1,"Fade":"ON","Speed":3,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"home.lan","BSSId":"58:EF:68:BF:7D:25","Channel":6,"RSSI":64,"LinkCount":1,"Downtime":"0T00:00:06"}}}
22:39:26 MQT: tele/ESP_2A63AC/STATE = {"Time":"2019-12-22T22:39:26","Uptime":"2T09:10:15","UptimeSec":205815,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Dimmer":100,"Color":"FF0004","HSBColor":"359,100,100","Channel":[100,0,2],"Scheme":0,"Width":1,"Fade":"ON","Speed":3,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"home.lan","BSSId":"58:EF:68:BF:7D:25","Channel":6,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:06"}}
22:39:26 MQT: tele/ESP_2A63AC/SENSOR = {"Time":"2019-12-22T22:39:26","AM2301":{"Temperature":21.2,"Humidity":54.2},"TempUnit":"C"}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
In the user_config_override.h file I have the following settings for both firmware versions:
#define USE_WS2812
#define USE_WS2812_CTYPE NEO_GRBW

Pixels are 60 in both.
I tried the following in 7.1.2 but with no effect:

Added #define USE_WS2812_DMA to user_config_override.h
Changed board_build.f_cpu to 160000000L in platformio.ini
Changed #define USE_WS2812_HARDWARE to NEO_HW_SK6812 and USE_WS2812x in user_config_override.h
Changed #define USE_WS2812_CTYPE to NEO_RGBW in user_config_override.h in combination of above settings.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-23:
Upgrade to latest release 7.2.

ameennihad, 2019-12-23:
I updated to 7.2.0, the result is the same.

ameennihad, 2019-12-23:
I solved the issue by removing all USE_WS2812 define lines from **user_config_override.h** file and change the same settings in **my_user_config.h** file, my final settings are as follows:

```
#define USE_WS2812
  //#define USE_WS2812_DMA
  #define USE_WS2812_HARDWARE  NEO_HW_SK6812
  #define USE_WS2812_CTYPE     NEO_GRBW
```

ameennihad, 2019-12-24:
Although, I managed to solve the issue in one device, when I uploaded the same firmware to a 2nd device with same configuration the LEDs are showing random colors, what could cause the same firmware to behave diferently on two similar devices?

ascillato, 2019-12-24:
Seems that you have any setoption or config different OR the device is different.

For the config, try to do a reset 6 in the tasmota console, to reset everything to default and re-config it again.

For testing the device, interchange the strips.

If you need further assistance with that, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ameennihad, 2019-12-24:
Devices are the same, both used to work on 6.5 with no issues for over a year, initially they were on lower version. It's not possible to exchange the strips but I'll try the reset option later today.

ameennihad, 2019-12-24:
Sorry, the issue with 2nd device was my fault, I uploaded an old firmware to the device. I uploaded 7.2.0, the LEDs are working as expected now.

closed 2019-12-24 by ameennihad