arendst/Tasmota #7301: Total energy (today, yesterday) statistical error in three pzem-004t

boyyao, 2019-12-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [OK ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [OK ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ OK] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ OK] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [OK ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [OK ] Device used (e.g., Sonoff Basic): Sonoff Basic
- [ OK] Tasmota binary firmware version number used: 7.1.2
  - [OK ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [OK ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
12:01:13 SRC: Backlog
12:01:13 CMD: 组: 0, 索引: 1, 命令: "MODULE", 数据: ""
12:01:13 MQT: stat/tasmota/RESULT = {"Module":{"1":"Sonoff Basic"}}
12:01:14 SRC: Backlog
12:01:14 CMD: 组: 0, 索引: 1, 命令: "GPIO", 数据: "255"
12:01:14 MQT: stat/tasmota/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"98":"PZEM016 Rx"},"GPIO2":{"0":"无"},"GPIO3":{"62":"PZEM0XX Tx"},"GPIO4":{"0":"无"},"GPIO5":{"0":"无"},"GPIO9":{"0":"无"},"GPIO10":{"0":"无"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"无"},"GPIO15":{"0":"无"},"GPIO16":{"0":"无"}}

```
- [OK ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
12:02:43 CMD: Backlog Rule1; Rule2; Rule3
12:02:43 SRC: WebConsole from 192.168.3.237
12:02:43 CMD: 组: 0, 索引: 1, 命令: "BACKLOG", 数据: "Rule1; Rule2; Rule3"
12:02:43 SRC: Backlog
12:02:43 CMD: 组: 0, 索引: 1, 命令: "RULE", 数据: ""
12:02:43 MQT: stat/tasmota/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:02:43 SRC: Backlog
12:02:43 CMD: 组: 0, 索引: 2, 命令: "RULE", 数据: ""
12:02:43 MQT: stat/tasmota/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:02:43 DMP: 01 04 14 09 3D 05 45 00 00 0A 30 00 00 47 AB 00 00 01 F4 00 52 00 00 78 EE
12:02:43 SRC: Backlog
12:02:43 CMD: 组: 0, 索引: 3, 命令: "RULE", 数据: ""
12:02:43 MQT: stat/tasmota/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [OK ] Provide the output of this command: `Status 0`:
```
  12:03:19 CMD: Status 0
12:03:19 SRC: WebConsole from 192.168.3.237
12:03:19 CMD: 组: 0, 索引: 1, 命令: "STATUS", 数据: "0"
12:03:19 MQT: stat/tasmota/STATUS = {"Status":{"Module":1,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:03:19 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Hardware Watchdog","Uptime":"0T03:07:19","StartupUTC":"2019-12-23T00:56:00","Sleep":50,"CfgHolder":4617,"BootCount":39,"SaveCount":92,"SaveAddress":"F8000"}}
12:03:19 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"7.1.2(tasmota)","BuildDateTime":"2019-12-13T02:31:32","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(5ab15d1)","Hardware":"ESP8266EX"}}
12:03:20 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["傻逼姚的wifi",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
12:03:20 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":565,"Free":436,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000804","8FDAE397","003683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:03:20 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6296","IPAddress":"192.168.3.3","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"EC:FA:BC:62:B8:98","Webserver":2,"WifiConfig":4}}
12:03:20 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.3.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_62B898","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:03:20 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 23 04:03:20 2019","Local":"Mon Dec 23 12:03:20 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+08:00","Sunrise":"15:41","Sunset":"23:56"}}
12:03:20 MQT: stat/tasmota/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
12:03:20 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-23T12:03:20","ENERGY":{"TotalStartTime":"2019-12-22T06:58:59","Total":52.093,"Yesterday":27.058,"Today":25.035,"Power":[265,14,157],"ApparentPower":[322,47,180],"ReactivePower":[182,45,87],"Factor":[0.82,0.29,0.87],"Frequency":[50,50,50],"Voltage":[236,236,238],"Current":[1.363,0.198,0.757]}}}
12:03:20 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-23T12:03:20","Uptime":"0T03:07:20","UptimeSec":11240,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"傻逼姚的wifi","BSSId":"00:90:4C:18:C2:34","Channel":10,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
At first it looked very normal.

But recently, I found that the energy statistics seem to violate the normal values I actually use. So I opened the log and found out. The statistical energy is actually higher than the total energy sent back by three 004ts. And today, yesterday and total energy refresh is slow

The data are as follows:

11:45:02 DMP: 02 04 14 09 54 00 C7 00 00 00 89 00 00 14 1C 00 00 01 F4 00 1D 00 00 C5 47
11:45:03 DMP: 01 04 14 09 52 05 47 00 00 0A 4B 00 00 47 5D 00 00 01 F4 00 52 00 00 9A 8C
11:45:04 DMP: 03 04 14 09 60 03 48 00 00 02 EA 00 00 1E 31 00 00 01 F4 00 25 00 00 F5 AD
2=14 1C = 5148Wh
1= 47 5D = 18269Wh
3=1E 31 = 7729Wh
total = 31146Wh =31.146kWh

**(Please, remember to close the issue when the problem has been addressed)**

boyyao, 2019-12-23:
Problem solved 
Settings.flag3.hardware_energy_total) {  // SetOption72 - Enable hardware energy total counter as reference (#6561)

boyyao, 2019-12-24:
In the beginning, everything is normal, but after a period of time (maybe an hour or more), the total value will become the data from which to supplement, and the value is the same as that without opt72.

sheminasalam, 2019-12-25:
what is the difference between setoption 72 enabled and disabled?

pablozg, 2019-12-25:
> what is the difference between setoption 72 enabled and disabled?

The setoption72 works only in hardware with total energy counter like sdm120, ddsu666, etc.

The reported value is used to update tasmota internal counter and be in sync both counters.

If enabled, tasmota use the reported value as total energy counter.

ascillato2, 2019-12-29:
This issue seems to be the same as #7155

Closing this one in order to have just one thread for the same issue

closed 2019-12-29 by ascillato2

leranp, 2021-02-09:
Same here, the value from the pzem doesn't match the tasmota value in the console, with SetOption72 is ON .
I have also 3 pzem connecting to same board via TX RX
Any solution for this?