arendst/Tasmota #7302: Sonoff L1 LED Lights

rob196, 2019-12-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The LED's do not get the updated color after changing the color via the sliders (on Tasmota Web UI) or via Console.
Only when I toggle the Off/On does the color change on the LED's

**NOTE:** On the very odd occasion it does update the LED color (and sometimes takes > 5 seconds)

**NOTE:** I flashed back to 6.5.0.16 (only has 1 slider on web UI - Dark --> Light ), here the LED's were being updated.
 - I noticed that after updating back to latest Dev (8.0.0.1) and using any of the 3 sliders the LED's updated, however after running Reset 5 and re-configuring to the Sonoff L1 module, the LED's do not update when using the sliders. So not sure if possibly a SetOption that now may have a different default value - I have tried various SetOption values under the Lights section with no luck.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff L1
- [x] Tasmota binary firmware version number used: 8.0.0.1 (core 2_6_1/2.2.2-dev(38a443e))
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmota-PyFlasher-1.0
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
10:38:50 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
10:38:50 RSL: stat/tasmota/RESULT = {"Module":{"70":"Sonoff L1"}}
10:38:51 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"148":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"149":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
10:40:10 RSL: stat/tasmota/STATUS = {"Status":{"Module":70,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:40:10 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:12:27","StartupUTC":"2019-12-23T08:27:43","Sleep":50,"CfgHolder":4617,"BootCount":2,"SaveCount":25,"SaveAddress":"FB000"}}
10:40:10 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.0.0.1(b90a462-tasmota)","BuildDateTime":"2019-12-22T22:00:09","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"310/699"}}
10:40:10 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Hidden","Hidden1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
10:40:10 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":27,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
10:40:10 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6155","IPAddress":"192.168.1.50","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"2C:F4:32:BF:D8:0B","Webserver":2,"WifiConfig":4}}
10:40:10 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AFC80B","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
10:40:10 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 23 09:40:10 2019","Local":"Mon Dec 23 10:40:10 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:56"}}
10:40:10 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-23T10:40:10"}}
10:40:10 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-23T10:40:10","Uptime":"0T01:12:27","UptimeSec":4347,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":0,"POWER":"OFF","Dimmer":49,"Color":"7D0003","HSBColor":"358,100,49","Channel":[49,0,1],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Hidden","BSSId":"0C:9D:92:46:2B:18","Channel":1,"RSSI":100,"Signal":-45,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
10:50:48 SRC: Switch
10:50:48 CMD: Group 0, Index 2, Command "COLOR", Data "ff0100"
10:50:48 RSL: stat/tasmota/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF0100","HSBColor":"0,100,100","Channel":[100,1,0]}
10:50:48 CFG: Saved to flash at F4, Count 72, Bytes 4096
10:50:48 HTP: Main Menu
10:50:51 SRC: WebGui from 192.168.0.69
10:50:51 CMD: Group 0, Index 1, Command "HSBCOLOR", Data "227"
10:50:51 RSL: stat/tasmota/RESULT = {"POWER":"ON","Dimmer":100,"Color":"0037FF","HSBColor":"227,100,100","Channel":[0,22,100]}
10:50:51 RSL: stat/tasmota/RESULT = {"POWER":"ON","Dimmer":100,"Color":"0037FF","HSBColor":"227,100,100","Channel":[0,22,100]}
10:50:52 CFG: Saved to flash at FB, Count 73, Bytes 4096
10:50:57 HTP: Console

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

1. Ensure device is ON (i.e. LED's are on)
2. Drag the color slider to another color
 - Can also drag one of the other sliders does not change (except dragging the Dimmer slider all the way to the left which does switch the LED's off)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
LED color to be updated to "selected" color

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

I found [this](https://github.com/arendst/Tasmota/issues/4935#issuecomment-540177756) ticket/comment but it has been closed (flagged as unsupported, but since then I see there was a [merged PR](https://github.com/arendst/Tasmota/pull/6002) to support the L1)

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-12-23:
Thx for your info.

I just tested mine and it works as you would expect; start with red, move to blue and it does.

So what happens in your case I don't know. I hope you do not leave the leds ON at full power too long while on the roll because the leds do not like that.

Here's my status 0 info:
```
12:22:09 MQT: stat/l1/STATUS = {"Status":{"Module":70,"FriendlyName":["Sonoff L1"],"Topic":"l1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:22:09 MQT: stat/l1/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:41","StartupUTC":"2019-12-23T11:18:28","Sleep":50,"CfgHolder":4617,"BootCount":118,"SaveCount":852,"SaveAddress":"FA000"}}
12:22:09 MQT: stat/l1/STATUS2 = {"StatusFWR":{"Version":"8.0.0.1(tasmota)","BuildDateTime":"2019-12-22T17:03:17","Boot":6,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"305/699"}}
12:22:09 MQT: stat/l1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"domus1","LogPort":514,"SSId":["indebuurt","indebuurt1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","000000C0","00000000"]}}
12:22:09 MQT: stat/l1/STATUS4 = {"StatusMEM":{"ProgramSize":498,"Free":504,"Heap":25,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8F1AA186","842403A1","04001E4C","01000BD0","00006881"],"Drivers":"1,2,3,4,5,6,7,9,10,22,24,29,99","Sensors":"2,3,5,8,9,10,34"}}
12:22:09 MQT: stat/l1/STATUS5 = {"StatusNET":{"Hostname":"l1","IPAddress":"192.168.2.156","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"DC:4F:22:F8:03:29","Webserver":2,"WifiConfig":5}}
12:22:09 MQT: stat/l1/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F80329","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:22:09 MQT: stat/l1/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 23 11:22:09 2019","Local":"Mon Dec 23 12:22:09 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:45","Sunset":"16:16"}}
12:22:09 MQT: stat/l1/STATUS10 = {"StatusSNS":{"Time":"2019-12-23T12:22:09"}}
12:22:09 MQT: stat/l1/STATUS11 = {"StatusSTS":{"Time":"2019-12-23T12:22:09","Uptime":"0T00:03:41","UptimeSec":221,"Heap":29,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":5,"Color":"000D02","HSBColor":"131,100,5","Channel":[0,5,1],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"indebuurt","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:07"}}}
```
You might want to compare SetOption.

BTW what is the output of command ``SerialConfig``? Should be 8N1.

rob196, 2019-12-23:
Thanks @arendst - I don't leave the LED's on for long at all while rolled up, just for quick test
Will compare SetOption & update ticket

SerialConfig is 8N1

rob196, 2019-12-23:
The following SetOptions where different and I matched yours - did not help
 - 43 (0 to 255)
 - 56 (OFF to ON)
 - 57 (OFF to ON)

I also changed Sleep form 50 to 10, set LedTable to OFF - to match your setup a bit closer

After changing Sleep to 10, the slider worked, but after toggling Off/On it didn't work again
 - Changed Sleep to 5 and it is now working as expected (I am sure I tried this, obviously not though - sorry)

Thanks for the help & awesome firmware

rob196, 2019-12-24:
Ok sorry so the above doesn't seem to actually resolve the issue - I re-flashed the device & followed the above again, but cannot get it to work (tried multiple times).

Could it be some kind of delay or "hidden" error (i.e. not saving/setting color)?
 - As I noticed now that sometimes when changing color form Red to Green for example (LED's did not change), toggle OFF, toggle ON - the LED color is still Red (slider showed green)
 - So then I got the color correct after dragging the slider again and toggling the power, and then I dragged the color slider again (LED's did not change), after +-1 minute I refreshed the browser and noticed the color slider was back to the original color (i.e. it's like the new color was not sent/saved)

Not seeing anything in the console logs (Web log level is set to 4)

Not sure if this matters as well, but the only other differences to Stats results is on StatusMEM for: Features, Drivers, Sensors

arendst, 2019-12-24:
The problem with this type of controller is that it uses a second microcontroller driving the PWM channels. Communication between the microcontroller and Tasmota  on ESP8266 is via serial interface. If that comms fails you'll see the GUI color changing while the LEDs stay different.

It may be caused by a power issue (I only have 1 meter of leds) or something else. 

If you compile yourselves you might want to enable more logging in file xlgt_05_sonoff_l1.ino

rob196, 2019-12-24:
Ah ok, thanks, I will give it a try over the next couple of days and update

I have 5 meter LED's (unfortunately nothing shorter to test)

rob196, 2019-12-26:
It seems changing the Serial Logging level to "1 Error" or "0 None" gets around this issue.

arendst, 2019-12-26:
Seriallogging would interfere with the comms to microcontroller so indeed it should be set to 0

closed 2019-12-26 by rob196

Wriman, 2020-04-25:
Hi! Can you send me the homeassistant config?

rob196, 2020-04-26:
@Wriman I have actually not set them up in HA yet, just did a quick test though and got it working

 - If you need to add the light see [here](https://community.home-assistant.io/t/led-strip-rgbw-magic-home-with-tasmota-and-mqtt/144922) -Dont think you need this if you have auto discovery enabled (SetOption19)
 - The card I used to test was this [custom light card](https://community.home-assistant.io/t/light-entity-card/96146)