arendst/Tasmota #7305: Problem with Shutter Position

Eligio-Greco, 2019-12-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Error of Shutter Position.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): ESP12-F (ESP8266 con 4 Mb)
- [X] Tasmota binary firmware version number used: 7.2
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [X] IDE / Compiler used: Arduino 1.8.10, Tasmota Libraries (Lib), Tasmota Platform (platform.txt)
- [X] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
SetOption80 1
SetOption1 1
SetOption13 1

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
sonoff_215/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff_215_Up","Sonoff_215_Dw"],"Topic":"sonoff_215","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
ShutterPosition1 0: sonoff_215/tele/RESULT = {"Shutter1":{"Position":0,"direction":0}}
ShutterPosition1 10: sonoff_215/tele/RESULT = {"Shutter1":{"Position":10,"direction":0}}
ShutterPosition1 20: sonoff_215/tele/RESULT = {"Shutter1":{"Position":20,"direction":0}}
ShutterPosition1 30: sonoff_215/tele/RESULT = {"Shutter1":{"Position":30,"direction":0}}

ShutterPosition1 32: sonoff_215/tele/RESULT = {"Shutter1":{"Position":2,"direction":0}}
ShutterPosition1 50: sonoff_215/tele/RESULT = {"Shutter1":{"Position":20,"direction":0}}

ShutterPosition1 52: sonoff_215/tele/RESULT = {"Shutter1":{"Position":2,"direction":0}}
ShutterPosition1 70: sonoff_215/tele/RESULT = {"Shutter1":{"Position":20,"direction":0}}

ShutterPosition1 72: sonoff_215/tele/RESULT = {"Shutter1":{"Position":2,"direction":0}}
ShutterPosition1 90: sonoff_215/tele/RESULT = {"Shutter1":{"Position":20,"direction":0}}

ShutterPosition1 92: sonoff_215/tele/RESULT = {"Shutter1":{"Position":2,"direction":0}}
ShutterPosition1 100: sonoff_215/tele/RESULT = {"Shutter1":{"Position":10,"direction":0}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
With Tasmota 7.1.2 (or 6.7.1) Shutter Position is Ok, 
With Tasmota 7.2 Shutter Position it's NOT Ok, 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

skawt-tee, 2019-12-24:
I'm experiencing the same behavior. The reported position during open/close movement jumps around to odd values.

Upgraded from stephenbode's codebase (6.?) back to 7.2 Tasmota, fresh reconfig from scratch (reset 6 used).

Open/Close works as expected, but reported position is rarely correct.

ascillato2, 2019-12-26:
Closing in favour of #7331 so as to not having open duplicated issues. Thanks

closed 2019-12-26 by ascillato2