arendst/Tasmota #7306: Restore Configuration on first boot.

BASM, 2019-12-23:
**Have you looked for this feature in other issues and in the docs?**  

yes

**Is your feature request related to a problem? Please describe.**  

I can't fast deploy/restore configuration if reset. Resetting the configuration has become very easy.

**Describe the solution you'd like**  

Add "Restore Configuration" on first boot menu.

**Describe alternatives you've considered**  

I don't know. Do recovery in two steps? Not user friendly.

**Additional context**  

It became extremely important because of the default reset after 3 power cycles. 
You may reset  besauce of poor AC/DC. For example, I added 4 relay to Sonoff 4ch,
switch on 8 relays and got reset configuration after 2-5 minuts.

**(Please, remember to close the issue when the problem has been addressed)**  

Of course.

ascillato, 2019-12-23:
You can disable the boot loop protection if you need with `setoption36` and `setoption65`.

The restore configuration is already there in the webUI menu of Tasmota as BACKUP and as RESTORE.

Besides, you can send by mqtt, serial, http or in the console, a `backlog` with all the config you need.

Also, you can put your settings as Default, using _user_config_override.h_ as explained in the docs.

If you need further assistance with any of the supported options, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://tasmota.github.io/docs/#/) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-23 by ascillato2

BASM, 2019-12-24:
Restore Configuration exist
[image]

Restore Configuration don't exist
[image]

BASM, 2019-12-24:
It have by main menu, but not on initial setup. Please reopen issue.

ascillato, 2019-12-24:
Ah, ok, you want the **_restore configuration button_** on the **_AP menu_**.

From the AP menu you can access all Tasmota menus but using their actual address. For example, typing `http://192.168.4.1/rs?` on the browser will take you to the **restore config menu**. Also you can access the console etc etc.

Anyway, I like your request. Make totally sense to have that button. Will make a PR to add that button later on.

Thanks for sharing and explaining your ideas. Very appreciated. :+1:

ascillato, 2019-12-24:
The PR #7314 has been submitted for revision.

ascillato, 2019-12-26:
The PR #7329 has been submitted for revision.

PR #7329 (2019-12-26): AP MODE: Add Restore Config Button
## Description:

Added a **Restore Configuration Button** when Tasmota is on AP (Access Point) Mode for easy restore of a previously saved configuration, making the initial setup faster for a reseted device.

**Related issue (if applicable):** fixes #7306

BEFORE and AFTER this PR:

[image]

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your [... cut]