arendst/Tasmota #7317: RS485 suport for modbus over MAX485

Black-Blade, 2019-12-25:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

Hello everybody
Since my Smart Meter only has RS485 (A / B) and can therefore only be addressed via a MAX485, I had to install a new constuctor in the "TasmotaModbus" class

Here is the code
   TasmotaModbus.cpp

#include "TasmotaModbus.h"
TasmotaModbus::TasmotaModbus(int receive_pin, int transmit_pin) : TasmotaSerial(receive_pin, transmit_pin, 1)
{
  mb_address = 0;
  _direction_open= false;
}

TasmotaModbus::TasmotaModbus(int receive_pin, int transmit_pin,int direction_pin) : TasmotaSerial(receive_pin, transmit_pin, 1)
{
  mb_address = 0;
  _direction_open= true;
  _direction_pin =direction_pin;
  if (_direction_pin < 99) {
    pinMode(_direction_pin, OUTPUT);
    digitalWrite(_direction_pin,RS485RESIVE);
  }
}

uint16_t CalculateCRC(uint8_t *frame, uint8_t num)
{
  uint16_t crc = 0xFFFF;

  for (uint8_t i = 0; i < num; i++) {
    crc ^= frame[i];
    for (uint8_t j = 8; j; j--) {
      if ((crc & 0x0001) != 0) {        // If the LSB is set
        crc >>= 1;                      // Shift right and XOR 0xA001
        crc ^= 0xA001;
      } else {                          // Else LSB is not set
        crc >>= 1;                      // Just shift right
      }
    }
  }
  return crc;
}

int TasmotaModbus::Begin(long speed, int stop_bits)
{
  int result = 0;

  if (begin(speed, stop_bits)) {
    result = 1;
    if (hardwareSerial()) { result = 2; }
  }
  return result;
}

void TasmotaModbus::Send(uint8_t device_address, uint8_t function_code, uint16_t start_address, uint16_t register_count)
{
  uint8_t frame[8];

  mb_address = device_address;  // Save address for receipt check

  frame[0] = mb_address;        // 0xFE default device address or dedicated like 0x01
  frame[1] = function_code;
  frame[2] = (uint8_t)(start_address >> 8);   // MSB
  frame[3] = (uint8_t)(start_address);        // LSB
  frame[4] = (uint8_t)(register_count >> 8);  // MSB
  frame[5] = (uint8_t)(register_count);       // LSB
  uint16_t crc = CalculateCRC(frame, 6);
  frame[6] = (uint8_t)(crc);
  frame[7] = (uint8_t)(crc >> 8);

  flush();
  if(_direction_open) digitalWrite(_direction_pin,RS485TRANSMIT);
  write(frame, sizeof(frame));
  if(_direction_open) digitalWrite(_direction_pin,RS485RESIVE);

}

bool TasmotaModbus::ReceiveReady()
{
  return (available() > 4);
}

uint8_t TasmotaModbus::ReceiveBuffer(uint8_t *buffer, uint8_t register_count)
{
  mb_len = 0;
  uint32_t last = millis();
  while ((available() > 0) && (mb_len < (register_count *2) + 5) && (millis() - last < 10)) {
    uint8_t data = (uint8_t)read();
    if (!mb_len) {               // Skip leading data as provided by hardware serial
      if (mb_address == data) {
        buffer[mb_len++] = data;
      }
    } else {
      buffer[mb_len++] = data;
      if (3 == mb_len) {
        if (buffer[1] & 0x80) {  // 01 84 02 f2 f1
          return buffer[2];      // 1 = Illegal Function,
                                 // 2 = Illegal Data Address,
                                 // 3 = Illegal Data Value,
                                 // 4 = Slave Error
                                 // 5 = Acknowledge but not finished (no error)
                                 // 6 = Slave Busy
                                 // 8 = Memory Parity error
                                 // 10 = Gateway Path Unavailable
                                 // 11 = Gateway Target device failed to respond
        }
      }
    }
    last = millis();
  }

  if (mb_len < 7) { return 7; }  // 7 = Not enough data

/*
  if (mb_len != buffer[2] + 5) {
    buffer[2] = mb_len - 5;      // As it's wrong anyway let's store actual number received in here (5 will be added by client)
    return 3;                    // 3 = Unexpected result
  }
*/

  uint16_t crc = (buffer[mb_len -1] << 8) | buffer[mb_len -2];
  if (CalculateCRC(buffer, mb_len -2) != crc) {
    return 9;                    // 9 = crc error
  }

  return 0;                      // 0 = No error
}

uint8_t TasmotaModbus::Receive16BitRegister(uint16_t *value)
{
  //  0  1  2  3  4  5  6
  // 01 04 02 43 21 HH LL
  // Id Cc Sz Regis Crc--

  uint8_t buffer[7];

  uint8_t error = ReceiveBuffer(buffer, 1);  // 1 x 16bit register
  if (!error) {
    *value = (buffer[3] << 8) | buffer[4];
  }

  return error;
}

uint8_t TasmotaModbus::Receive32BitRegister(float *value)
{
  //  0  1  2  3  4  5  6  7  8
  // 01 04 04 87 65 43 21 HH LL
  // Id Cc Sz Register--- Crc--

  uint8_t buffer[9];

  *value = NAN;

  uint8_t error = ReceiveBuffer(buffer, 2);  // 1 x 32bit register
  if (!error) {
    ((uint8_t*)value)[3] = buffer[3];
    ((uint8_t*)value)[2] = buffer[4];
    ((uint8_t*)value)[1] = buffer[5];
    ((uint8_t*)value)[0] = buffer[6];
  }

  return error;
}

and    TasmotaModbus.h

#ifndef TasmotaModbus_h
#define TasmotaModbus_h

#define  RS485TRANSMIT  HIGH
#define  RS485RESIVE    LOW

#include <Arduino.h>
#include <TasmotaSerial.h>

#define TM_MODBUS_BAUDRATE           9600   // Default baudrate

class TasmotaModbus : public TasmotaSerial {
  public:
    TasmotaModbus(int receive_pin, int transmit_pin);
    TasmotaModbus(int receive_pin, int transmit_pin, int direction_pin);
    
    virtual ~TasmotaModbus() {}

    int Begin(long speed = TM_MODBUS_BAUDRATE, int stop_bits = 1);

    void Send(uint8_t device_address, uint8_t function_code, uint16_t start_address, uint16_t register_count);

    bool ReceiveReady();

    /* Return codes:
     * 0 = No error
     * 1 = Illegal Function,
     * 2 = Illegal Data Address,
     * 3 = Illegal Data Value,
     * 4 = Slave Error
     * 5 = Acknowledge but not finished (no error)
     * 6 = Slave Busy
     * 7 = Not enough minimal data received
     * 8 = Memory Parity error
     * 9 = Crc error
     * 10 = Gateway Path Unavailable
     * 11 = Gateway Target device failed to respond
     */
    uint8_t ReceiveBuffer(uint8_t *buffer, uint8_t register_count);
    uint8_t Receive16BitRegister(uint16_t *value);
    uint8_t Receive32BitRegister(float *value);

    uint8_t ReceiveCount(void) { return mb_len; }

  private:
    uint8_t mb_address;
    uint8_t mb_len;

    bool _direction_open;
    int  _direction_pin;

};

Jason2866, 2019-12-26:
Thx for sharing your findings. What is your intention for posting the code here? If you want that this is considered to be included in Tasmota a PR has to be done.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

ascillato2, 2019-12-26:
Thanks, please, do a PR. Thanks

closed 2019-12-26 by ascillato2

iosifnicolae2, 2022-12-03:
I've implemented a solution here: https://github.com/arendst/Tasmota/pull/17247