arendst/Tasmota #7318: increase PWM resolution of H801 from 8 bit to 10 bit

GittaHuby, 2019-12-25:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
The current PWM resolution of H801 in Tasmota is 8 bit (0...255). Using the lowest value for barely glowing a LED-Channel (feature as nigth light) results in still to much brightness for using LED as a night light (my LED strip is 6m long). 

**Describe the solution you'd like**  
Increase the PWM resolution for the H801 (Tasmota module 20) to 10 bit (0...1023) would be fine. The lowest possible brightness of LED strip then is dark enough (I already tested it with 10 bit PWM resolution, stand alone, without Tasmota).

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-12-25:
If you use `LedTable 1` I.e. Gamma correction, you are already at 10 bits resolution. You may need to lower the pwm frequency to benefit from the full resolution.

GittaHuby, 2019-12-25:
@s-hadinger : Thank you, I'll check this out.

closed 2019-12-25 by GittaHuby