arendst/Tasmota #7325: White Blend Mode (White not 100%)

DpunktS, 2019-12-25:
### PROBLEM DESCRIPTION
Hallo, wenn ich White Blend Mode aktiviere mit RGBWWTable 255,255,255,255,0 ist der weiße Kanal immer noch dunkler wie die anderen Kanäle. Laut Messgerät hat der Kanal 1,3A bei 100% aber wenn ich White Blend Mode aktiviere hat er bei 100% nur noch 0,4A. Mit den Sk6812 LEDs geht es, nur mit den normalen LED strips nicht.
Tasmota 8.1.0/7.1.2/6.7.1

Ich Danke schon mal im voraus.

Hello, when I activate White Blend Mode with RGBWWTable 255,255,255,255,0 the white channel is still darker than the other channels. According to the measuring device, the channel has 1.3A at 100% but when I activate White Blend Mode it only has 0.4A at 100%. With the Sk6812 LEDs it works, but not with the normal LED strips.
Tasmota 8.1.0 / 7.1.2 / 6.7.1

I thank you in advance.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): MagicHome RGBW
- [x] Tasmota binary firmware version number used: 8.1.0
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [X] IDE / Compiler used: 1.8.10
- [X] Flashing tools used: CP2102
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
{"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:25:35 MQT: stat/tasmota/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:25:35 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:33:35","StartupUTC":"2019-12-25T18:52:00","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":38,"SaveAddress":"F6000"}}
20:25:35 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019.12.25 19:34:37","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(5ab15d1)","Hardware":"ESP8266EX","CR":"321/699"}}
20:25:35 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
20:25:35 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":644,"Free":356,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","87DAE397","003E83A0","22B617CD","01001BC0","00007981"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
20:25:35 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"xxxxxxxxx","IPAddress":"xxxxxxx","Gateway":"xxxxxxxx","Subnetmask":"xxxxx","DNSServer":"xxxxxxx","Mac":"xxxxxx","Webserver":2,"WifiConfig":4}}
20:25:35 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxxxx","MqttPort":xxxx,"MqttClientMask":"xxxxxxx","MqttClient":"xxxxxx","xxxxx":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:25:35 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 25 19:25:35 2019","Local":"Wed Dec 25 20:25:35 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:42","Sunset":"16:57"}}
20:25:35 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-25T20:25:35"}}
20:25:35 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-25T20:25:35","Uptime":"0T00:33:35","UptimeSec":2015,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":28,"MqttCount":1,"POWER":"OFF","Dimmer":100,"Color":"FFFFFF00","HSBColor":"0,0,100","Channel":[100,100,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"xxxxxxx","BSSId":"xxxxxxxxx","Channel":1,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
always

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2019-12-25:
Your template can't work, it has no GPIO assigned. Can you put your actual template please?

DpunktS, 2019-12-25:
[image]

s-hadinger, 2019-12-25:
Ok. Can you confirm you're seeing a visual difference between `Color FFFFFF00` and `Color 000000FF` ?

DpunktS, 2019-12-25:
yes clearly # ffffff00 without white blend (very bright) and with white blend (ca. 50%) and # 000000ff (very bright).

PR #7338 (2019-12-26): Fix White blend mode #7325
## Description:

Fixes wrong range for White channel when White-blend mode is activated.
Also added low-level log for PWM channels, this should ease further diagnosis.

**Related issue (if applicable):** fixes #7325 

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [x] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github. [... cut]
closed 2019-12-26 by arendst