arendst/Tasmota #7331: falsche shutter Position beim steuern über den Befehl "shutterposition1 *"

ihavenonick, 2019-12-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R2, Shelly 2, Shelly 2.5
- [x] Tasmota binary firmware version number used: 8.1.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSCode
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
11:19:58 MQT: stat/tasmota/RESULT = {"NAME":"Sonoff Dual R2","GPIO":[18,255,0,255,0,22,17,0,21,56,0,0,0],"FLAG":0,"BASE":39}
11:19:58 MQT: stat/tasmota/RESULT = {"Module":{"0":"Sonoff Dual R2"}}
11:19:58 MQT: stat/tasmota/RESULT = {"GPIO0":{"18":"Button2"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"17":"Button1"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
```
- [x] Provide the output of this command: `Status 0`:
```
11:20:55 MQT: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota","Tasmota2"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[3,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:20:55 MQT: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:28:23","StartupUTC":"2019-12-26T09:52:32","Sleep":100,"CfgHolder":4617,"BootCount":5,"SaveCount":35,"SaveAddress":"F9000"}}
11:20:55 MQT: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(tasmota)","BuildDateTime":"2019.12.26 10:16:38","Boot":6,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"375/699"}}
11:20:55 MQT: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FRITZ!","FRITZ!"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000000","00000000"]}}
11:20:55 MQT: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":478,"Free":524,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000407","8F082392","040003A0","20021706","010000C0","00000900"],"Drivers":"1,2,3,4,9,10,20,27,29","Sensors":"3"}}
11:20:55 MQT: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-6838","IPAddress":"192.168.100.57","Gateway":"192.168.100.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.1","Mac":"60:01:94:9E:7A:B6","Webserver":2,"WifiConfig":4}}
11:20:55 MQT: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.10","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9E7AB6","MqttUser":"Jan","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:20:55 MQT: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Thu Dec 26 10:20:55 2019","Local":"Thu Dec 26 11:20:55 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:42","Sunset":"16:58"}}
11:20:55 MQT: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-26T11:20:55","Shutter1":{"Position":19,"direction":0}}}
11:20:55 MQT: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-26T11:20:55","Uptime":"0T00:28:23","UptimeSec":1703,"Vcc":3.484,"Heap":28,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":9,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"FRITZ!","BSSId":"5C:49:79:89:9F:0F","Channel":8,"RSSI":34,"Signal":-83,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
11:25:00 CMD: ShutterPosition1 50
11:25:00 SRC: WebConsole from 192.168.100.16
11:25:00 CMD: Gruppe 0, Index 1, Befehl "SHUTTERPOSITION", Daten "50"
11:25:00 SHT: Pos. in: payload 50 (2), payload 50, idx 1, src 9
11:25:00 SHT: lastsource 9:, real 28000, target 14000, payload 50
11:25:00 SHT: Start in dir 0
11:25:00 SRC: Shutter
11:25:00 SHT: Switched relay: 2 by Shutter
11:25:00 MQT: stat/tasmota/RESULT = {"POWER2":"ON"}
11:25:00 MQT: stat/tasmota/POWER2 = ON
11:25:00 MQT: stat/tasmota/RESULT = {"ShutterPosition1":50}
11:25:00 SHT: Shutter 1: Real. 27600, Start: 28000, Stop: 14000, dir -1, motordelay 0, rtc: 0.20 [s], freq 1000
11:25:00 MQT: tele/tasmota/RESULT = {"Shutter1":{"Position":8,"direction":-1}}
11:25:01 CFG: in Flash gespeichert am F5, zählen 39, Bytes 4096
11:25:01 SHT: Shutter 1: Real. 25500, Start: 28000, Stop: 14000, dir -1, motordelay 0, rtc: 1.25 [s], freq 1000
11:25:01 MQT: tele/tasmota/RESULT = {"Shutter1":{"Position":1,"direction":-1}}
11:25:02 SHT: Shutter 1: Real. 23500, Start: 28000, Stop: 14000, dir -1, motordelay 0, rtc: 2.25 [s], freq 1000
11:25:02 MQT: tele/tasmota/RESULT = {"Shutter1":{"Position":13,"direction":-1}}
11:25:03 SHT: Shutter 1: Real. 21500, Start: 28000, Stop: 14000, dir -1, motordelay 0, rtc: 3.25 [s], freq 1000
11:25:03 MQT: tele/tasmota/RESULT = {"Shutter1":{"Position":6,"direction":-1}}
11:25:04 SHT: Shutter 1: Real. 19500, Start: 28000, Stop: 14000, dir -1, motordelay 0, rtc: 4.25 [s], freq 1000
11:25:04 MQT: tele/tasmota/RESULT = {"Shutter1":{"Position":19,"direction":-1}}
11:25:05 SHT: Shutter 1: Real. 17500, Start: 28000, Stop: 14000, dir -1, motordelay 0, rtc: 5.25 [s], freq 1000
11:25:05 MQT: tele/tasmota/RESULT = {"Shutter1":{"Position":12,"direction":-1}}
11:25:06 SHT: Shutter 1: Real. 15500, Start: 28000, Stop: 14000, dir -1, motordelay 0, rtc: 6.25 [s], freq 1000
11:25:06 MQT: tele/tasmota/RESULT = {"Shutter1":{"Position":5,"direction":-1}}
11:25:07 SRC: Shutter
11:25:07 SHT: Switched relay: 2 by Shutter
11:25:07 MQT: stat/tasmota/RESULT = {"POWER2":"OFF"}
11:25:07 MQT: stat/tasmota/POWER2 = OFF
11:25:07 SHT: Shutter 1: Real. 13800, Start: 28000, Stop: 14000, dir -1, motordelay 0, rtc: 7.1 [s], freq 1000
11:25:07 MQT: stat/tasmota/SHUTTER1 = 19
11:25:07 MQT: tele/tasmota/RESULT = {"Shutter1":{"Position":19,"direction":0}}
11:25:08 CFG: in Flash gespeichert am F4, zählen 40, Bytes 4096

```
### TO REPRODUCE
_Steps to reproduce the behavior:_
This command "ShutterPosition 1 50" should the position under "position" with "50" displayed too, but ist "19". The shutter is real in position 50 (time not real position without calibration). 
If i do the next command "ShutterPositon1 100", normal should the position 100 to, but ist 10 and the shutter ist real full open.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
in older Tasmota, the position is the same like the command it provide

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
after command "shutterposition1 50" and "shutterposition 100"
[image]

the datpoints in IOBroker with the same wrong state
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Rayzilt, 2019-12-26:
I've the same problem as described in this ticket, the "Position" is incorretly reported. Previously running 7.1.2 was runnig great, upgraded to 8.1.0 introduced this bug. I haven't run 7.2.0 but found some tickets with the same problem.

Some console logging:
```
16:15:20 CMD: ShutterRelay1 1
16:15:20 SHT: Shutter 1 (Relay:1): Init. Pos: 5200 [10 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 52000, Opentime 26.0 [s], Closetime 26.0 [s], CoeffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 1, shuttermode 0,motordelay 0
16:15:20 MQT: guestroomawning/stat/RESULT = {"ShutterRelay1":1}
16:15:35 CMD: ShutterSetClose1
16:15:35 MQT: guestroomawning/stat/RESULT = {"ShutterSetClose1":"Configuration reset"}
16:15:51 CMD: ShutterPosition1 0
16:15:51 SHT: Start in dir 0
16:15:51 MQT: guestroomawning/tele/STATE = {"Time":"2019-12-26T16:15:51","Uptime":"0T00:23:27","UptimeSec":1407,"Heap":30,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"Rayzilt-MGMT","BSSId":"F2:9F:C2:F1:BA:F6","Channel":1,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:07"}}
16:15:51 MQT: guestroomawning/stat/RESULT = {"POWER1":"ON"}
16:15:51 MQT: guestroomawning/stat/POWER1 = ON (retained)
16:15:51 MQT: guestroomawning/stat/RESULT = {"ShutterPosition1":0}
16:15:51 SHT: Shutter 1: Real. 700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 0.35 [s], freq 1000
16:15:51 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":99,"direction":1}}
16:15:52 SHT: Shutter 1: Real. 2700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 1.35 [s], freq 1000
16:15:52 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":95,"direction":1}}
16:15:53 SHT: Shutter 1: Real. 4700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 2.35 [s], freq 1000
16:15:53 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":91,"direction":1}}
16:15:54 SHT: Shutter 1: Real. 6700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 3.35 [s], freq 1000
16:15:54 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":88,"direction":1}}
16:15:55 SHT: Shutter 1: Real. 8700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 4.35 [s], freq 1000
16:15:55 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":84,"direction":1}}
16:15:56 SHT: Shutter 1: Real. 10700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 5.35 [s], freq 1000
16:15:56 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":80,"direction":1}}
16:15:57 SHT: Shutter 1: Real. 12700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 6.35 [s], freq 1000
16:15:57 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":76,"direction":1}}
16:15:58 SHT: Shutter 1: Real. 14700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 7.35 [s], freq 1000
16:15:58 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":72,"direction":1}}
16:15:59 SHT: Shutter 1: Real. 16700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 8.35 [s], freq 1000
16:15:59 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":98,"direction":1}}
16:16:00 SHT: Shutter 1: Real. 18600, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 9.30 [s], freq 1000
16:16:00 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":95,"direction":1}}
16:16:01 SHT: Shutter 1: Real. 20700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 10.35 [s], freq 1000
16:16:01 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":91,"direction":1}}
16:16:02 SHT: Shutter 1: Real. 22700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 11.35 [s], freq 1000
16:16:02 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":87,"direction":1}}
16:16:03 SHT: Shutter 1: Real. 24700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 12.35 [s], freq 1000
16:16:03 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":83,"direction":1}}
16:16:04 SHT: Shutter 1: Real. 26700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 13.35 [s], freq 1000
16:16:04 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":99,"direction":1}}
16:16:05 SHT: Shutter 1: Real. 28700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 14.35 [s], freq 1000
16:16:05 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":95,"direction":1}}
16:16:06 SHT: Shutter 1: Real. 30700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 15.35 [s], freq 1000
16:16:06 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":91,"direction":1}}
16:16:07 SHT: Shutter 1: Real. 32700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 16.35 [s], freq 1000
16:16:07 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":88,"direction":1}}
16:16:08 SHT: Shutter 1: Real. 34700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 17.35 [s], freq 1000
16:16:08 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":84,"direction":1}}
16:16:09 SHT: Shutter 1: Real. 36700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 18.35 [s], freq 1000
16:16:09 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":100,"direction":1}}
16:16:10 SHT: Shutter 1: Real. 38700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 19.35 [s], freq 1000
16:16:10 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":96,"direction":1}}
16:16:11 SHT: Shutter 1: Real. 40700, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 20.35 [s], freq 1000
16:16:11 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":92,"direction":1}}
[... cut ...] me":"0T00:00:07"}}
16:16:17 MQT: guestroomawning/stat/RESULT = {"POWER1":"OFF"}
16:16:17 MQT: guestroomawning/stat/POWER1 = OFF (retained)
16:16:17 SHT: Shutter 1: Real. 52000, Start: 0, Stop: 52000, dir 1, motordelay 0, rtc: 26.0 [s], freq 1000
16:16:17 MQT: guestroomawning/stat/SHUTTER1 = 90 (retained)
16:16:17 MQT: guestroomawning/tele/RESULT = {"Shutter1":{"Position":90,"direction":0}}
```

Tickets: https://github.com/arendst/Tasmota/issues/7305 https://github.com/arendst/Tasmota/issues/7304

kenschbacher9, 2019-12-26:
I´ve encountered the same problem today - all my shutters running 7.2 were affected.
As temporary workaround I have just downgraded to 7.1.2.6 and the position is correct again. Downgrading has shown no problem till now.

TrDA-hab, 2019-12-27:
I've the same problem as described in this ticket, the "Position" is incorretly reported. Tasmota 8.1.0.1

TrDA-hab, 2019-12-27:
I've the same problem ... the "Position" is incorretly reported. Tasmota 8.1.0.1
[image]

he-so, 2019-12-27:
I can also verify that this happens on Shutter1. But when creating a second Shutter for the same relays, the Position value is updated correctly. This might be a workaround.

he-so, 2019-12-28:
I downgraded to 7.2.0, reset all configuration and found: The position value is not updated at all when moving the shutter either with ShutterPosition or ShutterOpen/Close commands. Also creating new shutters does not solve that issue.

TrDA-hab, 2019-12-28:
@stefanbode

In version 8.1 - Shutters do not work correctly. Please see when you will have free time. Thanks.

stefanbode, 2019-12-28:
I had a bug with the shutterinvert.  I think this was already solved. Can you check if you shutters are accidentally inverted?

stefanbode, 2019-12-28:
I read to all of the comments now. Does not look like an invert problem. The last log where the position is reported as 90% look good so far. Do you have a log with the error. I’m a bit disconnected now

he-so, 2019-12-28:
Today I did some test with different tasmota versions. Here my findings:
8.1.1?(yesterday’s dev branch): not working.
Release 8.1: not working, position goes zig-zag
Release 7.2: infinite loop, shutter never stops, position stays zero
Release 7.1.2: works fine.

TrDA-hab, 2019-12-28:
I will test it now.

ihavenonick, 2019-12-28:
it works perfekt now
[image]

[image]

closed 2019-12-28 by ihavenonick

TrDA-hab, 2019-12-28:
@stefanbode

Test - **default settings**.

00:00:58 CMD: ShutterFrequency
00:00:58 RSL: stat/tasmota/RESULT = {"ShutterFrequency":1000}
00:01:21 CMD: ShutterCloseDuration
00:01:21 RSL: stat/tasmota/RESULT = {"ShutterCloseDuration1":"**0.0**"}
00:01:36 CMD: ShutterOpenDuration
00:01:36 RSL: stat/tasmota/RESULT = {"ShutterOpenDuration1":"**0.0**"}
00:01:50 CMD: ShutterMotorDelay
00:01:50 RSL: stat/tasmota/RESULT = {"ShutterMotorDelay1":"**0.00**"}

TrDA-hab, 2019-12-28:
@stefanbode

In the new Wiki, I see this.
Does this ability to manage different PWM at the same time?

[image]

jokerigno, 2019-12-28:
Sorry but I didn't understand which version should I use to solve this issue

Rayzilt, 2019-12-28:
@jokerigno Grab the current development branch or wait for a new version release of Tasmota

jokerigno, 2019-12-28:
OK thank you :) I Think I'll wait to a new stable relase before breaking all the cover around the house. Thank you :)

TrDA-hab, 2019-12-28:
@stefanbode

Thanks for the very quick good help. Shutter#1 operation errors - fixed. All work is good. Tomorrow I will test the operation with two shutters (Shutter#1 + Shutter#2). 

Thank you very much, you saved my chair.

stefanbode, 2019-12-28:
The max frequency can be defined for every shutter independently. Anyhow only one shutter can move at a time because the frequency is a global setting effecting all pwm currently. But I ave ideas how to overcome this.

stefanbode, 2019-12-29:
> @stefanbode
> 
> In the new Wiki, I see this.
> Does this ability to manage different PWM at the same time?
> 
> [image]

Yes, the idea is to define openduration and frequency first. Than you can change the motodelay for a smooth ramp up and ramp down. If this is all fine you can set the closeduration and the shutter will change to the required frequency but maintain the acceleration limits of the ramp. This is typical for doors that must open quick and close slow. With 24V up to 9kHZ PWM was possible with a smooth ramp. This is essential.

LaurentG91, 2020-01-30:
Just to confirm: in 8.1.0 recompiled to support shutter.
The intermediate position sent, through RESULT message, never go over 29 instead of going from 0 to 100.. I hope this help to solve the PB
See logs
[16:24:41] cmnd/Volet_01/ShutterOpen1 
[16:24:41] stat/Volet_01/RESULT {"POWER1":"ON"}
[16:24:41] stat/Volet_01/POWER1 ON
[16:24:41] stat/Volet_01/RESULT {"ShutterOpen1":100}
[16:24:42] tele/Volet_01/RESULT {"Shutter1":{"Position":4,"direction":1}}
[16:24:43] tele/Volet_01/RESULT {"Shutter1":{"Position":11,"direction":1}}
[16:24:44] tele/Volet_01/RESULT {"Shutter1":{"Position":18,"direction":1}}
[16:24:45] tele/Volet_01/RESULT {"Shutter1":{"Position":25,"direction":1}}
[16:24:46] tele/Volet_01/RESULT {"Shutter1":{"Position":3,"direction":1}}
[16:24:47] tele/Volet_01/RESULT {"Shutter1":{"Position":10,"direction":1}}
[16:24:48] tele/Volet_01/RESULT {"Shutter1":{"Position":17,"direction":1}}
[16:24:49] tele/Volet_01/RESULT {"Shutter1":{"Position":4,"direction":1}}
[16:24:50] tele/Volet_01/RESULT {"Shutter1":{"Position":11,"direction":1}}
[16:24:51] tele/Volet_01/RESULT {"Shutter1":{"Position":18,"direction":1}}
[16:24:52] tele/Volet_01/RESULT {"Shutter1":{"Position":5,"direction":1}}
[16:24:53] tele/Volet_01/RESULT {"Shutter1":{"Position":13,"direction":1}}
[16:24:54] tele/Volet_01/RESULT {"Shutter1":{"Position":0,"direction":1}}
[16:24:55] tele/Volet_01/RESULT {"Shutter1":{"Position":7,"direction":1}}
[16:24:55] stat/Volet_01/RESULT {"POWER1":"OFF"}
[16:24:55] stat/Volet_01/POWER1 OFF
[16:24:55] stat/Volet_01/SHUTTER1 100
[16:24:55] tele/Volet_01/RESULT {"Shutter1":{"Position":100,"direction":0}}

MarksLiveMusic, 2020-03-21:
I have the same problem: positioning ("ShutterPosition") is wrongly reported every time. I can't figure out whais is happening. Any kind of help will be appreciated. Thank you all!

ihavenonick, 2020-03-22:
[image]
everything is ok

stefanbode, 2020-03-22:
Mark, can you send some logs or better open a new issue. It should work since weeks. I assume another issue.

MarksLiveMusic, 2020-03-22:
Hi all and thank you for answer.

I have the same issue of @LaurentG91 , i think it's better continuing this issue, it's the same.

I've set only `ShutterOpenDuration` and `ShutterCloseDuration` and leave other options by default. Here in the log from "all open" to "full closed".

`14:10:55 SHT: Wait for Motorstop..
14:10:55 MQT: stat/T121-STUDIO/RESULT = {"POWER1":"OFF"}
14:10:55 MQT: stat/T121-STUDIO/POWER1 = OFF (salvato)
14:10:55 MQT: stat/T121-STUDIO/RESULT = {"POWER1":"OFF"}
14:10:55 MQT: stat/T121-STUDIO/POWER1 = OFF (salvato)
14:10:56 SHT: Wait for Motorstop..
14:10:56 MQT: stat/T121-STUDIO/RESULT = {"POWER2":"ON"}
14:10:56 MQT: stat/T121-STUDIO/POWER2 = ON (salvato)
14:10:56 SHT: Shutter 1: Real. 34477, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 0.05 [s], freq 1035
14:10:56 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":9,"direction":-1}}
14:10:57 SHT: Shutter 1: Real. 32726, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 0.90 [s], freq 1035
14:10:57 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":4,"direction":-1}}
14:10:58 SHT: Shutter 1: Real. 30666, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 1.90 [s], freq 1035
14:10:58 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":18,"direction":-1}}
14:10:59 SHT: Shutter 1: Real. 28709, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 2.85 [s], freq 1035
14:10:59 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":12,"direction":-1}}
14:11:00 SHT: Shutter 1: Real. 26649, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 3.85 [s], freq 1035
14:11:00 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":7,"direction":-1}}
14:11:01 SHT: Shutter 1: Real. 24589, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 4.85 [s], freq 1035
14:11:01 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":1,"direction":-1}}
14:11:02 SHT: Shutter 1: Real. 22529, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 5.85 [s], freq 1035
14:11:02 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":15,"direction":-1}}
14:11:03 SHT: Shutter 1: Real. 20469, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 6.85 [s], freq 1035
14:11:03 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":9,"direction":-1}}
14:11:04 SHT: Shutter 1: Real. 18409, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 7.85 [s], freq 1035
14:11:04 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":3,"direction":-1}}
14:11:05 SHT: Shutter 1: Real. 16246, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 8.90 [s], freq 1035
14:11:05 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":16,"direction":-1}}
14:11:06 SHT: Shutter 1: Real. 14289, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 9.85 [s], freq 1035
14:11:06 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":11,"direction":-1}}
14:11:07 SHT: Shutter 1: Real. 12229, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 10.85 [s], freq 1035
14:11:07 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":5,"direction":-1}}
14:11:08 SHT: Shutter 1: Real. 10066, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 11.90 [s], freq 1035
14:11:08 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":29,"direction":-1}}
14:11:09 SHT: Shutter 1: Real. 8109, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 12.85 [s], freq 1035
14:11:09 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":23,"direction":-1}}
14:11:10 SHT: Shutter 1: Real. 6049, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 13.85 [s], freq 1035
14:11:10 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":17,"direction":-1}}
14:11:11 SHT: Shutter 1: Real. 3989, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 14.85 [s], freq 1035
14:11:11 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":11,"direction":-1}}
14:11:12 SHT: Shutter 1: Real. 1929, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 15.85 [s], freq 1035
14:11:12 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":5,"direction":-1}}
14:11:13 MQT: stat/T121-STUDIO/RESULT = {"POWER2":"OFF"}
14:11:13 MQT: stat/T121-STUDIO/POWER2 = OFF (salvato)
14:11:13 SHT: Shutter 1: Real. -28, Start: 34580, Stop: 0, dir -1, motordelay 0, rtc: 16.8 [s], freq 1035
14:11:13 MQT: stat/T121-STUDIO/SHUTTER1 = 0 (salvato)
14:11:13 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":0,"direction":0}}`

Here's the log from "full closed" to "full open".

`14:15:32 SHT: Wait for Motorstop..
14:15:33 MQT: stat/T121-STUDIO/RESULT = {"POWER1":"ON"}
14:15:33 MQT: stat/T121-STUDIO/POWER1 = ON (salvato)
14:15:33 SHT: Shutter 1: Real. 72, Start: -28, Stop: 34600, dir 1, motordelay 0, rtc: 0.05 [s], freq 1000
14:15:33 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":0,"direction":1}}
14:15:33 SHT: Shutter 1: Real. 1072, Start: -28, Stop: 34600, dir 1, motordelay 0, rtc: 0.55 [s], freq 1000
14:15:33 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":3,"direction":1}}
14:15:34 SHT: Shutter 1: Real. 3072, Start: -28, Stop: 34600, dir 1, motordelay 0, rtc: 1.55 [s], freq 1000
14:15:34 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":8,"direction":1}}
14:15:35 SHT: Shutter 1: Real. 5072, Start: -28, Stop: 34600, dir 1, motordelay 0, rtc: 2.55 [s], freq 1000
14:15:35 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":14,"direction":1}}
14:15:36 SHT: Shutter 1: Real. 7072, Start: -28, Stop: 34600, dir 1, motordelay 0, rtc: 3.55 [s], freq 1000
14:15:36 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":20,"direction":1}}
[... cut ...]
14:15:49 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":5,"direction":1}}
14:15:50 MQT: stat/T121-STUDIO/RESULT = {"POWER1":"OFF"}
14:15:50 MQT: stat/T121-STUDIO/POWER1 = OFF (salvato)
14:15:50 SHT: Shutter 1: Real. 34672, Start: -28, Stop: 34600, dir 1, motordelay 0, rtc: 17.4 [s], freq 1000
14:15:50 MQT: stat/T121-STUDIO/SHUTTER1 = 100 (salvato)
14:15:50 MQT: tele/T121-STUDIO/RESULT = {"Shutter1":{"Position":100,"direction":0}}`

You can see the `ShutterPosition` giving random values instead a real position (%).

Rayzilt, 2020-03-22:
Did you guys use the 8.1.0 release or cloned the latest development branch? As this bug should be resolved. Now that 8.2.0 is out, can you try that version?

MarksLiveMusic, 2020-03-22:
Updated to 8.2.0, problem solved. Thank you!