arendst/Tasmota #7332: Sonoff RF-Bridge erkennt keine Geräte mehr

erniecs, 2019-12-26:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Aejoh, 2019-12-26:
Same problem. I'm back to 7.1.2 so it still works.

ascillato, 2019-12-26:
GOOGLE TRANSLATE OF PROBLEM DESCRIPTION:

I have updated my Sonoff RF-Bridge with Tasmota 8.1.0 since the update the bridge no longer detects values from my smoke detectors and my remote control. Everything worked smoothly with firmware 6.7.1. If a smoke detector triggers or I press a button on the remote control, there is only "CMD:" but there are no values behind it.

rjduraocosta, 2019-12-26:
Same problem here also.

arendst, 2019-12-26:
Provide output of commands ``status 0`` and ``serialconfig``

Aejoh, 2019-12-27:
After a serial programming it works for me. (8.1.0)

Jason2866, 2019-12-27:
With lastest commit https://github.com/arendst/Tasmota/commit/5682675ac9faa14c6a80c80f3754015191d88900it is fixed

rjduraocosta, 2019-12-27:
> With lastest commit [5682675](https://github.com/arendst/Tasmota/commit/5682675ac9faa14c6a80c80f3754015191d88900it) is fixed

Tested just now and I can confirm that it is fixed.

Thanks a lot for your work.

erniecs, 2019-12-27:
> Nach einer seriellen Programmierung klappt es bei mir. (8.1.0)

Kann ich bestätigen.
Danach von 8.1.0 auf 8.1.0.1 über OTA upgedatet und es läuft auch

closed 2019-12-27 by erniecs

arendst, 2019-12-27:
So it was just caused by missed serial config data upgrade.

If users had execute serialconfig 8N1 and seriallog 0 once the issue was solved too