arendst/Tasmota #7341: RGB Bulb won't turn on after installing Tasmota

FaserF, 2019-12-27:
### PROBLEM DESCRIPTION
I am unable to control any relay on my RGB Bulb. The bulb won't turn on, no matter what GPIO Pins I assign a relay. I am running the latest dev version.
My bulb seems to be very similar to the FCMILA Bulb (unfortunately there is no brand written on it). I was Not able to get any color at all working. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4) (unfortunately my issue went under after some other people wrote new problems so nobody has really taken a look into my issue)
- [x] Device used (e.g., Sonoff Basic): TUYA FCMILA RGB Bulb
- [ ] Tasmota binary firmware version number used: 8.1.0.1
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tuya-convert
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
00:00:23 CMD: Backlog Template; Module; GPIO 255
00:00:24 RSL: stat/tasmota/RESULT = {"NAME":"RGB Bulb","GPIO":[0,0,0,0,40,0,0,0,38,39,37,0,0],"FLAG":0,"BASE":18}
00:00:24 RSL: stat/tasmota/RESULT = {"Module":{"0":"RGB Bulb"}}
00:00:24 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"40":"PWM4"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"38":"PWM2"},"GPIO13":{"39":"PWM3"},"GPIO14":{"37":"PWM1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
15:38:25 CMD: Status 0
15:38:25 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["bulb-1"],"Topic":"tasmota","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:38:25 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:25","StartupUTC":"2019-12-27T14:37:00","Sleep":50,"CfgHolder":4617,"BootCount":42,"SaveCount":270,"SaveAddress":"FB000"}}
15:38:25 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(24aff9b-tasmota)","BuildDateTime":"2019-12-27T13:00:09","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"353/699"}}
15:38:25 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SmartLife","Fabian-Mi9T"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
15:38:25 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":26,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:38:25 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-3748","IPAddress":"192.168.178.89","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.2","Mac":"B4:E6:2D:07:8E:A4","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
15:38:25 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_078EA4","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:38:25 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 27 14:38:25 2019","Local":"Fri Dec 27 15:38:25 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:59"}}
15:38:25 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-27T15:38:25"}}
15:38:25 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-27T15:38:25","Uptime":"0T00:01:25","UptimeSec":85,"Heap":26,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":100,"Color":"0088FF00","HSBColor":"208,100,100","Channel":[0,53,100,0],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SmartLife","BSSId":"98:9B:CB:A5:20:6F","Channel":6,"RSSI":68,"Signal":-66,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
That I can turn on and off the bulb and change the color.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

blakadder, 2019-12-28:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. GitHub issues are intended for bug reports only.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
Read [Documentation](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-28 by blakadder

FaserF, 2019-12-28:
As written above the chat was not useful for me:

"unfortunately my issue went under after some other people wrote new problems so nobody has really taken a look into my issue"

I think a chat where everybody can post and no quoting is given doesn't really make that much sense to help people :/

blakadder, 2019-12-28:
And you're asking as to magically discover which bulb you're using without providing any concrete information... funny how communication works

FaserF, 2019-12-28:
Sorry, this wasn't meant in a negative way. I am really happy for all the work and especially all the help you are providing to us, including me who doesn't have that much knowledge with boards etc.
This time I got great support from you and meimgraham

Maybe for future issues, could you tell me what "concrete information" was missing? I tried to include as many information as I got. Should I have opened the bulb and send a Foto of it?

I hope to get it working with tuyaMCU