arendst/Tasmota #7343: Stepper Motor Shutter Control from KNX

misc2000, 2019-12-27:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**  

I did follow issue #7006 to control shutters with stepper motors.
I was able to config and control 2 shutter with the current version 7.1.2.6 with shutters enabled.

Because I have also an KNX installation my next challenge is now to control this 2 shutters from the KNX side (KNX switches)
Problems: 
The shutters are not selectable in the KNX setup page pulldown menu, only the relays.

KNX have 2 binary group address (up/down) and (stop) and for me it looks like this is not matching to the 2 relay setup for each shutter of the current implementation.

I also do think about to do something with RULES but do not find any solution.
Is there any way with rules to match that I did not see so far?
Did someone successfully configure some stepper shutters with KNX, because I see the hint to KNX at the ShutterInvert command?

My Setup Config:
D1: Relay2: A4988#1 PIN: DIR
D2: PWM1: A4988#1 PIN: STP
D3: Relay1i: A4988#1 PIN: EN
D4: Counter1: D2
D5: Relay4: A4988#2 PIN: DIR
D6: PWM2: A4988#2 PIN: STP
D7: Relay3i: A4988#2 PIN: EN
D8: Counter2: D6

Console setup:
SetOption80 1
***set for ShutterMode 1
Backlog PulseTime1 0; PulseTime2 0
Backlog PulseTime3 0; PulseTime4 0
Interlock OFF
ShutterRelay1 1
ShutterRelay2 3
Backlog ShutterOpenDuration1 15; ShutterCloseDuration1 15; shuttermotordelay1 0.25
Backlog ShutterOpenDuration2 15; ShutterCloseDuration2 15; shuttermotordelay2 0.25
shutterfrequency 1500 

Log of one shutter operation as example:
16:10:25 RSL: stat/tasmota/RESULT = {"POWER2":"OFF"}
16:10:25 RSL: stat/tasmota/POWER2 = OFF
16:10:25 SHT: Shutter 1: Real. 30000, Start: 1500, Stop: 30000, dir 1, motordelay 5, rtc: 14.7 [s], freq 0
16:10:25 RSL: stat/tasmota/SHUTTER1 = 0
16:10:25 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":0}}
16:10:31 SHT: Wait for Motorstop..
16:10:31 RSL: stat/tasmota/RESULT = {"POWER1":"OFF"}
16:10:31 RSL: stat/tasmota/POWER1 = OFF
16:10:31 RSL: stat/tasmota/RESULT = {"POWER2":"ON"}
16:10:31 RSL: stat/tasmota/POWER2 = ON
16:10:31 RSL: stat/tasmota/RESULT = {"POWER1":"ON"}
16:10:31 RSL: stat/tasmota/POWER1 = ON
16:10:31 RSL: stat/tasmota/RESULT = {"ShutterPosition1":100}
16:10:32 SHT: Shutter 1: Real. 28539, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 0.90 [s], freq 1500
16:10:32 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":5,"direction":-1}}
16:10:33 SHT: Shutter 1: Real. 26543, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 1.90 [s], freq 1500
16:10:33 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":12,"direction":-1}}
16:10:34 SHT: Shutter 1: Real. 24539, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 2.95 [s], freq 1500
16:10:34 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":19,"direction":-1}}
16:10:35 SHT: Shutter 1: Real. 22543, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 3.95 [s], freq 1500
16:10:35 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":25,"direction":-1}}
16:10:36 SHT: Shutter 1: Real. 20547, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 4.95 [s], freq 1500
16:10:36 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":32,"direction":-1}}
16:10:37 SHT: Shutter 1: Real. 18544, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 5.95 [s], freq 1500
16:10:37 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":39,"direction":-1}}
16:10:38 SHT: Shutter 1: Real. 16546, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 6.95 [s], freq 1500
16:10:38 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":45,"direction":-1}}
16:10:39 SHT: Shutter 1: Real. 14540, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 7.95 [s], freq 1500
16:10:39 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":52,"direction":-1}}
16:10:40 SHT: Shutter 1: Real. 12542, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 8.95 [s], freq 1500
16:10:40 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":59,"direction":-1}}
16:10:41 SHT: Shutter 1: Real. 10531, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 9.95 [s], freq 1500
16:10:41 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":65,"direction":-1}}
16:10:42 SHT: Shutter 1: Real. 8532, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 10.95 [s], freq 1500
16:10:42 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":72,"direction":-1}}
16:10:43 SHT: Shutter 1: Real. 6530, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 11.95 [s], freq 1500
16:10:43 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":79,"direction":-1}}
16:10:44 SHT: Shutter 1: Real. 4536, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 12.95 [s], freq 1500
16:10:44 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":85,"direction":-1}}
16:10:45 SHT: Shutter 1: Real. 2540, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 13.95 [s], freq 1500
16:10:45 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":92,"direction":-1}}
16:10:46 SHT: Shutter 1: Real. 622, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 14.90 [s], freq 1500
16:10:46 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":98,"direction":-1}}
16:10:47 RSL: stat/tasmota/RESULT = {"POWER1":"OFF"}
16:10:47 RSL: stat/tasmota/POWER1 = OFF
16:10:47 RSL: stat/tasmota/RESULT = {"POWER2":"OFF"}
16:10:47 RSL: stat/tasmota/POWER2 = OFF
16:10:47 SHT: Shutter 1: Real. 0, Start: 30000, Stop: 0, dir -1, motordelay 5, rtc: 15.6 [s], freq 0
16:10:47 RSL: stat/tasmota/SHUTTER1 = 100
16:10:47 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":100,"direction":0}}

Thanks
MISC / Michael

ascillato, 2019-12-27:
Shutters are very new in main Tasmota. I didn't have the time to make the KNX driver to support it. As shutters have their own implementation on KNX, I need to add the support in Tasmota. Added to the TO DO list for next year ;)

stefanbode, 2020-01-07:
I saw some KNX commits regarding shutters in the last days. Can you please recheck and update the issue here. Thx.

ascillato, 2020-01-07:
I didn't add anything yet. So far there is no support. I didn't have the time to work on it

ascillato, 2020-11-26:
Hi,

Acording to https://www.knx.org/wAssets/docs/downloads/Certification/Interworking-Datapoint-types/03_07_02-Datapoint-Types-v02.01.02-AS.pdf there are several ways to control a Shutter in KNX.

Which DPTs do you want to use?

A fast way is to use rules (KNXTX and KNXRX slots)

ufkel, 2020-11-26:
Hi Adrian,

glad to hear that you have found some time to improve the Tasmota KNX. Here is my recommendation: Take a real KNX Shutter Actuator as an example. For example, here is the one I use in my projects:
https://www.mdt.de/download/MDT_TM_RF_Shutter_Actuator.pdf <https://www.mdt.de/download/MDT_TM_RF_Shutter_Actuator.pdf>

Page 11-16 list all the relevant DPTs in a nice overview.

Regards, Udo

ascillato, 2020-11-26:
Thank you very much for this info. That was exactly what I needed. And also is what I was afraid of. It uses several DPTs, so it is quite a work to implement that. So far I don't have any shutters, I'm planning to have some but not any time soon. Until then, I don't think I'm going to work on this. Sorry. I have to be sincere, I just work for free only on things I need, problems I have or on small things others need but I like. Also I like to share that work, because it has no cost to me and might be useful for others, but I don't work on something will make me spend several hours of my limited free time and I have no gain on it, sorry.

Adding the label **Feature Request** to see if there is someone interested to work on this.

ascillato2, 2020-12-02:
Adding the label "requested feature (hold over)" for future reference. Please, ask to reopen if you want to work on this. Thanks.

closed 2020-12-02 by ascillato2