arendst/Tasmota #7345: 2gang dimmer flash to tasmota

Bajger-M, 2019-12-27:
Hi, I'd like to flash my dimmer on the tasmota. Could anyone help me? I can't figure out where to connect vcc, gnd, rx, tx ... and in what way. the older version was 4 holes but here are more

**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Bajger-M, 2019-12-27:
[image]

Jason2866, 2019-12-27:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-28 by blakadder

DanyaFilatov, 2020-01-27:
Hi,
I have the same device.
It can be flashed using Tuya-Convert.
Pinout is presented on the image.
[image]

It looks pretty similar to the QS-WiFi-D01 150W Dimmer, excepting that here we have 2 dimming channels. Counters for buttons are on IO13 and IO5. Both channels shares the same UART TX, so it must use different command sequences.

Unfortunately I lost the original firmware during my experiments and can not sniff UART sequences anymore. It would be fantastic if somebody can send me this firmware.

DanyaFilatov, 2020-01-27:
Do not forget that IO0 and both RE1 and RE2 pins shall be connected to GND in order to flash the chip using UART.