arendst/Tasmota #7346: APDS 9960 command not recognized

thepotoo, 2019-12-27:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Following a compile with the `#define USE_APDS9960`, the command `Sensor27` returns an error.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): NodeMCU v3
- [x] Tasmota binary firmware version number used: 8.1.0.1
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: VSCode/Platformio and gitpod just in case I was doing something wrong
- [x] Flashing tools used: esptool
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
19:22:12 CMD: Backlog Template; Module; GPIO 255
19:22:12 RSL: stat/tasmota/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
19:22:12 RSL: stat/tasmota/RESULT = {"Module":{"0":"Generic"}}
19:22:13 RSL: stat/tasmota/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"5":"I2C SCL"},"GPIO2":{"56":"Led1i"},"GPIO3":{"6":"I2C SDA"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
No rules

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
19:22:42 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
19:22:42 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T00:04:37","StartupUTC":"2019-12-27T18:18:05","Sleep":50,"CfgHolder":4617,"BootCount":12,"SaveCount":24,"SaveAddress":"FA000"}}
19:22:42 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(display)","BuildDateTime":"2019-12-27T13:16:03","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"303/699"}}
19:22:42 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wireless",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
19:22:42 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":556,"Free":444,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FAAA19F","04049FA1","62B617C1","010008C0","00004080"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6,7,8,9,10,14,15,17,18,20,22,26,27,34"}}
19:22:42 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7808","IPAddress":"192.168.1.145","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"CC:50:E3:37:BE:80","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
19:22:42 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_37BE80","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
19:22:42 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 27 18:22:42 2019","Local":"Fri Dec 27 19:22:42 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:59"}}
19:22:42 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-27T19:22:42"}}
19:22:42 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-27T19:22:42","Uptime":"0T00:04:37","UptimeSec":277,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"wireless","BSSId":"18:E8:29:91:F9:E1","Channel":3,"RSSI":98,"Signal":-51,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
19:24:22 CMD: i2cscan
19:24:22 SRC: WebConsole from 192.168.1.40
19:24:22 CMD: Group 0, Index 1, Command "I2CSCAN", Data ""
19:24:22 RSL: stat/tasmota/RESULT = {"I2CScan":"Device(s) found at 0x39"}
19:24:28 CMD: sensor27
19:24:28 SRC: WebConsole from 192.168.1.40
19:24:28 CMD: Group 0, Index 27, Command "SENSOR", Data ""
19:24:28 RSL: stat/tasmota/RESULT = {"Command":"Error"}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
In my user config override, I have only my wifi ssid, password, and `#define USE_APDS9960`

I connected the APDS 9960 to the I2C Tx and Rx pins, as well as +3 and GND, then used the web interface to configure Tx and Rx as I2C SCL and SDA.

I then run `i2cscan` and Tasmota detects the sensor at 0x39.

So far, so good (I think?)

But, when I enter `Sensor27` to enable the sensor, I get an error, as if I have compiled the wrong firmware or my user config isn't recognized.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I would expect sensor27 to return something (anything) other than error, based on the docs.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
N/A

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I am pretty sure that my user config file is correct, since this is not my first time building tasmota, and I see a warning in the compiler `#warning **** Turned off conflicting drivers SHT and VEML6070 ****`

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-27:
Status 4 doesnt show driver 21. No driver support for APDS 9960. 
See https://github.com/arendst/Tasmota/blob/development/I2CDEVICES.md
It seems you did something wrong doing the compile setup

thepotoo, 2019-12-27:
@Jason2866 Thanks, but I'm still a bit confused:
```
22:09:13 CMD: I2cDriver
22:09:13 RSL: stat/tasmota/RESULT = {"I2CDriver":"3,4,5,6,8,9,10,11,15,20,21"}
```
So it looks like it's enabled at compile time.

I then did `I2cDriver21 1` and the ESP rebooted.  But then status 4 still gives me:
```
22:10:07 CMD: status 4
22:10:07 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":556,"Free":444,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FAAA19F","04049FA1","62B617C1","010008C0","00004080"],"Drivers":"1,2,4,5,8,9,10,13,16","Sensors":"1,2,5,6,7,8,9,10,14,15,17,18,20,22,26,27,34"}}
```

Any other clues here?

arendst, 2019-12-27:
I2cdriver numbers are NOT sensor driver numbers. I2c devices can be drivers AND sensors so you will see different numbering for I2cDriver and sensors.

If sensor 21 is APDS9960 then it is not enabled in the compile as you do not see sensor 21 in the sensors list. You will need to enabled it with the correct define first and recompile.

arendst, 2019-12-27:
You might need to disable other sensors with same I2c address using comand I2cdriverx 0

thepotoo, 2019-12-27:
Here is another build, this time using the tasmota-sensor environment as a base.  Again, my user_config_override file contains only my wifi info and `#define USE_APDS9960`
```
23:09:46 CMD: I2cDriver21 1
23:09:46 RSL: stat/tasmota/RESULT = {"I2CDriver":"7,8,9,10,11,13,14,15,17,18,20,21,29,36,41"}
```
The esp rebooted and I then entered:
```
23:11:36 CMD: i2cscan
23:11:36 RSL: stat/tasmota/RESULT = {"I2CScan":"Device(s) found at 0x39"}
23:11:39 CMD: Sensor27
23:11:39 RSL: stat/tasmota/RESULT = {"Command":"Error"}
23:11:43 CMD: Sensor27 1
23:11:43 RSL: stat/tasmota/RESULT = {"Command":"Error"}
```
I see only 3 possible conflicts, and I disabled them all with these commands:
```
I2cDriver2 0
I2cDriver12 0
I2cDriver16 0
```
If it helps, here is the status 0 info:

```
23:15:33 CMD: status 0
23:15:33 RSL: stat/tasmota/STATUS = {"Status":{"Module":0,"FriendlyName":["Tasmota"],"Topic":"tasmota","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:15:33 RSL: stat/tasmota/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:54","StartupUTC":"2019-12-27T22:13:39","Sleep":50,"CfgHolder":4617,"BootCount":31,"SaveCount":52,"SaveAddress":"F6000"}}
23:15:33 RSL: stat/tasmota/STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(sensors)","BuildDateTime":"2019-12-27T17:03:14","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"303/699"}}
23:15:33 RSL: stat/tasmota/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wireless",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:15:33 RSL: stat/tasmota/STATUS4 = {"StatusMEM":{"ProgramSize":588,"Free":412,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE387","0437A3A5","F7F79FCD","01DA9BC0","0016FCC7"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,14,16,17,18,19,20,21,22,24,26,29,30","Sensors":"1,2,3,4,5,6,7,8,9,10,12,13,14,15,17,18,19,20,21,22,26,27,28,34,37,39,40,42,43,51,52,55,56,58"}}
23:15:33 RSL: stat/tasmota/STATUS5 = {"StatusNET":{"Hostname":"tasmota-7808","IPAddress":"192.168.1.145","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"CC:50:E3:37:BE:80","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:15:33 RSL: stat/tasmota/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_37BE80","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:15:33 RSL: stat/tasmota/STATUS7 = {"StatusTIM":{"UTC":"Fri Dec 27 22:15:33 2019","Local":"Fri Dec 27 23:15:33 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"16:59"}}
23:15:33 RSL: stat/tasmota/STATUS10 = {"StatusSNS":{"Time":"2019-12-27T23:15:33"}}
23:15:33 RSL: stat/tasmota/STATUS11 = {"StatusSTS":{"Time":"2019-12-27T23:15:33","Uptime":"0T00:01:54","UptimeSec":114,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"wireless","BSSId":"18:E8:29:91:F9:E1","Channel":3,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

Looking through other issues, I see a lot of people have successfully gotten this sensor working.  I tried another chip and another ESP (just in case), but that is showing the same behavior.

Is there anything else I should check here?

arendst, 2019-12-28:
It works for me too...
```
00:00:00 CFG: Loaded from flash at F6, Count 468
00:00:00 APP: Set Serial to 8N1 115200 bit/s
00:00:00 SRC: Restart
00:00:00 Project tasmota Wemos2 Version 8.1.0.1(tasmota)-2_6_1
00:00:00 I2C: APDS9960 found at 0x39
00:00:00 WIF: Attempting connection...
00:00:01 WIF: Network (re)scan started...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Network 0, AP1, SSId indebuurt, Channel 11, BSSId 18:E8:29:CA:17:C1, RSSI -35, Encryption 1
00:00:04 WIF: Connecting to AP1 indebuurt in mode 11N as wemos2...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on wemos2 with IP address 192.168.2.191
12:16:54 NTP: Drift 0, (UTC) Sat Dec 28 11:16:54 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
12:16:54 QPC: Reset
12:16:55 MQT: Attempting connection...
12:16:55 MQT: Connected
12:16:55 MQT: tele/wemos2/LWT = Online (retained)
12:16:55 MQT: cmnd/wemos2/POWER = 
12:16:55 MQT: Subscribe to cmnd/wemos2/#
12:16:55 MQT: Subscribe to cmnd/tasmotas/#
12:16:55 MQT: Subscribe to cmnd/DVES_15568C_fb/#
12:16:55 MQT: tele/wemos2/INFO1 = {"Module":"Generic","Version":"8.1.0.1(tasmota)","FallbackTopic":"cmnd/DVES_15568C_fb/","GroupTopic":"cmnd/tasmotas/"}
12:16:55 MQT: tele/wemos2/INFO2 = {"WebServerMode":"Admin","Hostname":"wemos2","IPAddress":"192.168.2.191"}
12:16:55 MQT: tele/wemos2/INFO3 = {"RestartReason":"Software/System restart"}
12:16:55 HTP: Main Menu
12:16:56 APP: Boot Count 168
12:16:57 CFG: Saved to flash at F5, Count 469, Bytes 4096
12:16:58 HTP: Console
12:16:59 MQT: tele/wemos2/STATE = {"Time":"2019-12-28T12:16:59","Uptime":"0T00:00:13","UptimeSec":13,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"indebuurt","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-37,"LinkCount":1,"Downtime":"0T00:00:07"}}
12:17:20 NONE
12:17:20 MQT: tele/wemos2/SENSOR = {"Time":"2019-12-28T12:17:20","APDS9960":{"None":1}}
12:17:25 NONE
12:17:25 MQT: tele/wemos2/SENSOR = {"Time":"2019-12-28T12:17:25","APDS9960":{"None":1}}
12:17:27 NONE
12:17:27 MQT: tele/wemos2/SENSOR = {"Time":"2019-12-28T12:17:27","APDS9960":{"None":1}}
12:17:28 LEFT
12:17:28 MQT: tele/wemos2/SENSOR = {"Time":"2019-12-28T12:17:28","APDS9960":{"Left":1}}
12:18:18 CMD: status 0
12:18:18 SRC: Serial
12:18:18 CMD: Group 0, Index 1, Command "STATUS", Data "0"
12:18:18 MQT: stat/wemos2/STATUS = {"Status":{"Module":18,"FriendlyName":["Wemos2"],"Topic":"wemos2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
12:18:18 MQT: stat/wemos2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://domus1:80/api/arduino/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:32","StartupUTC":"2019-12-28T11:16:46","Sleep":50,"CfgHolder":4617,"BootCount":168,"SaveCount":469,"SaveAddress":"F5000"}}
12:18:18 MQT: stat/wemos2/STATUS2 = {"StatusFWR":{"Version":"8.1.0.1(tasmota)","BuildDateTime":"2019-12-28T12:13:35","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"301/699"}}
12:18:18 MQT: stat/wemos2/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"domus1","LogPort":514,"SSId":["indebuurt","indebuurt1"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","000002C0","00000000"]}}
12:18:18 MQT: stat/wemos2/STATUS4 = {"StatusMEM":{"ProgramSize":499,"Free":504,"Heap":29,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1625C2","FlashMode":3,"Features":["00000809","8F1AA186","042403A1","44001E4C","01000BD0","00006881"],"Drivers":"1,2,3,4,5,6,7,9,10,22,24,29","Sensors":"2,3,5,8,9,10,27,34"}}
12:18:18 MQT: stat/wemos2/STATUS5 = {"StatusNET":{"Hostname":"wemos2","IPAddress":"192.168.2.191","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.27","Mac":"B4:E6:2D:15:56:8C","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
12:18:18 MQT: stat/wemos2/STATUS6 = {"StatusMQT":{"MqttHost":"domus1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_15568C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:18:18 MQT: stat/wemos2/STATUS7 = {"StatusTIM":{"UTC":"Sat Dec 28 11:18:18 2019","Local":"Sat Dec 28 12:18:18 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"08:46","Sunset":"16:20"}}
12:18:18 MQT: stat/wemos2/STATUS10 = {"StatusSNS":{"Time":"2019-12-28T12:18:18"}}
12:18:18 MQT: stat/wemos2/STATUS11 = {"StatusSTS":{"Time":"2019-12-28T12:18:18","Uptime":"0T00:01:32","UptimeSec":92,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"indebuurt","BSSId":"18:E8:29:CA:17:C1","Channel":11,"RSSI":100,"Signal":-38,"LinkCount":1,"Downtime":"0T00:00:07"}}}
12:19:26 CMD: sensor27
12:19:26 SRC: Serial
12:19:26 CMD: Group 0, Index 27, Command "SENSOR", Data ""
12:19:26 MQT: stat/wemos2/RESULT = {"Sensor27":"ON"}
12:20:02 CMD: i2cdriver
12:20:02 SRC: Serial
12:20:02 CMD: Group 0, Index 1, Command "I2CDRIVER", Data ""
12:20:02 MQT: stat/wemos2/RESULT = {"I2CDriver":"7,9,10,11,21"}
```
For debugging purposes you might want to try a compile with as less drivers/sensors as possible. My user_config_override.h starts off with undef ALL defines from my_user_config.h. Then I enable what I need. As you can see I have only enabled some I2C devices I really use.

Your starting point might not be the sensors version but the lite version where you have to enable I2C also. 
As you can see I have hardly enabled I2C sensors so you might want to try har

thepotoo, 2019-12-28:
I must be compiling this wrong, but I don't see the problem.

In platformio.ini, I uncommented `-DUSE_CONFIG_OVERRIDE` and `tasmota-lite`
In user_config_override.h, I added `#ifdef X #undef X #endif` for everything that looked like it could possibly matter.  The only defines were again my wifi info and 

This did not seem to make a difference.

As a sanity check, I then edited `xsns_27_apds9960.ino` to look like this:
```
#ifdef USE_I2C
#warning **** I2C support enabled ****  // Added this line as a sanity check
#ifdef USE_APDS9960
#warning **** Adding support for APDS9960 ****  // Added this line, too
```
When I compile like this, I do not see either warning (!!!) and i2cscan does not work.  It seems like tasmota_post.h is somehow overriding my defines.  Is there something else I need to change here?

arendst, 2019-12-28:
Let's not use the lite as indeed that will be overwritten by tasmota_post.

- Just use the basic tasmota which uses my_user_config.h
- enable ``-DUSE_CONFIG_OVERRIDE``
- Then in your override disable as much as possible and enable only what you need

thepotoo, 2019-12-28:
Great, that at least makes sense.  I can now build and see the warnings that I added, so it looks like I should have support for the APDS-9960

But even though it seems to be compiling correctly now, I still get this:
```
13:06:27 CMD: sensor27
13:06:27 MQT: stat/AB9722/RESULT = {"Command":"Error"}
```

thepotoo, 2019-12-28:
I get the same error whether I `#undef` everything and then enable the sensor, or just have these 4 lines in my user config:
```
#define STA_SSID1
#define STA_PASS1
#define USE_I2C
#define USE_APDS9960 
```

arendst, 2019-12-28:
As long as the hardware does not recognise eachother you will not be able to use command sensor27.

Try different gpios.

Jason2866, 2019-12-28:
Have you installed the needed pullup resistors for the i2c bus?

arendst, 2019-12-29:
Also pls change your memory model (4096) to the Tasmota supported and default memory model (1024) as provided in the platformio.ini file.

thepotoo, 2019-12-29:
Added pullup resistors on SCL and SCA and a cap across 3.3v and GND, no difference.

I added a BME280 sensor, which is working as long as I `#define USE_BMP`.  It works on any pin combination I try.

I suspect this is a bad sensor or something (although I did try two different 9960s).  This is just not worth spending any more time on.

closed 2019-12-29 by thepotoo

Basti149, 2024-09-16:
I want to connect an APDS 9960 to my Wemos mini.
For this I compiled Tasmota.bin with the following

_#ifndef USE_APDS9960
#define USE_APDS9960                // [I2cDriver21] Enable APDS9960 Proximity Sensor (I2C address 0x39). Disables SHT and VEML6070 (+4k7 code)
  #define USE_APDS9960_GESTURE        // Enable APDS9960 Gesture feature (+2k code)
  #define USE_APDS9960_PROXIMITY      // Enable APDS9960 Proximity feature (>50 code)
  #define USE_APDS9960_COLOR          // Enable APDS9960 Color feature (+0.8k code)
  #define USE_APDS9960_STARTMODE  0   // Default to enable Gesture mode
#endif_

But still nothing is displayed in the console
This is only displayed here with i2cscan.

15:46:55.893 RSL: RESULT = {"I2CScan":"Device(s) found at 0x39"}

Do I have to enter anything else? I use Tasmota 14.x.x

sfromis, 2024-09-16:
Did you verify that the driver is actually included and enabled on the device. Number 21 should be included in the output of the command `I2CDriver`.

What you have set via the command `Sensor27` will also influence what the sensor driver will produce, should match `USE_APDS9960_STARTMODE`, but if you upgrade from another build, any such defaults changes are not transferred to current config; that only happens with a full reset.

When booting with the sensor connected, and `Weblog 3`, I'm seeing this shortly after boot:
`I2C: APDS9960 found at 0x39`

Basti149, 2024-09-16:
I made a mistake and I have an APD9930 sensor, so Tasmota doesn't recognize it as 9960.

Holen Sie sich Outlook für Android<https://aka.ms/ghei36>