arendst/Tasmota #7351: Shutter broken, infinite loop

to-scho, 2019-12-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When updating from 7..1.0 to 8.1.0 shutter operation for my setup is broken. any movement of shutters results in endless relay operation. Here is my shutter setup:
ShutterCloseDuration1 30.0
ShutterCloseDuration2 0.0
ShutterCloseDuration3 0.0
ShutterCloseDuration4 0.0
ShutterInvert1 0
ShutterInvert2 0
ShutterInvert3 0
ShutterInvert4 0
ShutterMotorDelay1 0.9
ShutterMotorDelay2 0.0
ShutterMotorDelay3 0.0
ShutterMotorDelay4 0.0
ShutterOpenDuration1 33.0
ShutterOpenDuration2 0.0
ShutterOpenDuration3 0.0
ShutterOpenDuration4 0.0
ShutterPosition1 100
ShutterPosition2 0
ShutterPosition3 0
ShutterPosition4 0
ShutterRelay1 1
ShutterRelay2 0
ShutterRelay3 0
ShutterRelay4 0
ShutterSetHalfway1 50
ShutterSetHalfway2 0
ShutterSetHalfway3 0
ShutterSetHalfway4 0

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
0:10:16 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":9,"direction":-1}}
10:10:17 SHT: Shutter 1: Real. 52140, Start: 52140, Stop: 0, dir -1, motordelay 18, rtc: 1317.15 [s], freq 1100
10:10:17 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":9,"direction":-1}}
10:10:18 SHT: Shutter 1: Real. 52140, Start: 52140, Stop: 0, dir -1, motordelay 18, rtc: 1318.10 [s], freq 1100
10:10:18 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":9,"direction":-1}}
[... 380 lines cut ...]
10:15:22 SHT: Shutter 1: Real. 52140, Start: 52140, Stop: 0, dir -1, motordelay 18, rtc: 1622.10 [s], freq 1100
10:15:22 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":9,"direction":-1}}
10:15:23 SHT: Shutter 1: Real. 52140, Start: 52140, Stop: 0, dir -1, motordelay 18, rtc: 1623.10 [s], freq 1100
10:15:23 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":9,"direction":-1}}
10:15:24 SHT: Shutter 1: Real. 52140, Start: 52140, Stop: 0, dir -1, motordelay 18, rtc: 1624.10 [s], freq 1100
10:15:24 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":9,"direction":-1}}
10:15:25 SHT: Shutter 1: Real. 52140, Start: 52140, Stop: 0, dir -1, motordelay 18, rtc: 1625.10 [s], freq 1100
10:15:25 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":9,"direction":-1}}

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Any shutter movement 

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

he-so, 2019-12-28:
This seems to be related to https://github.com/arendst/Tasmota/issues/7331
Try creating Shutter2, for relays 1 and 2, maybe it works.

blakadder, 2019-12-28:
Try a clean flash of the latest binary first since there were configuration layout changes.

to-scho, 2019-12-28:
Thx, I erased flash and refreshed 8.1.0 followed by a basic setup of my module and shutterrelay1 1. Afterwards endless movement is happening again. Also setting up a second shutter by shutterrelay2 1 shows same behavior.

stefanbode, 2019-12-28:
I think I have found the error and submitted the fix. Anyhow there will be an additional change later on because with the current new version it might happen in rare cases that the position get negative. Anyhow this should remove the biggest pain until I sit again before my testing equipment.

stefanbode, 2019-12-28:
Sorry I mixed a bit the error. The latest submit fixes the position calculation. Can you please post the result of the command shutterrelay1 1
Want to check wich mode is recognized

stefanbode, 2019-12-28:
Also can you do a shutttersetclose1
This will make a reset. I already saw some problems if the internal position is outside the boundaries

to-scho, 2019-12-28:
basic setup after erase_flash:
17:46:12 SHT: Shutter 1 (Relay:1): Init. Pos: 2000 [10 %], Open Vel.: 100 Close Vel.: 100 , Max Way: 20000, Opentime 10.0 [s], Closetime 10.0 [s], CoeffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 0

to-scho, 2019-12-28:
no luck with shuttersetclose1:

17:52:02 CMD: shuttersetclose1
17:52:02 RSL: stat/tasmota/RESULT = {"ShutterSetClose1":"Configuration reset"}
17:52:14 SHT: Wait for Motorstop..
17:52:14 RSL: stat/tasmota/RESULT = {"POWER1":"OFF"}
17:52:14 RSL: stat/tasmota/POWER1 = OFF
17:52:14 RSL: stat/tasmota/RESULT = {"POWER2":"OFF"}
17:52:14 RSL: stat/tasmota/POWER2 = OFF
17:52:14 RSL: stat/tasmota/RESULT = {"POWER1":"ON"}
17:52:14 RSL: stat/tasmota/POWER1 = ON
17:52:14 RSL: stat/tasmota/RESULT = {"ShutterPosition1":100}
17:52:15 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 0.70 [s], freq 1000
17:52:15 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:16 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 1.70 [s], freq 1000
17:52:16 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:17 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 2.70 [s], freq 1000
17:52:17 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:18 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 3.70 [s], freq 1000
17:52:18 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:19 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 4.70 [s], freq 1000
17:52:19 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:20 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 5.70 [s], freq 1000
17:52:20 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:21 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 6.70 [s], freq 1000
17:52:21 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:22 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 7.70 [s], freq 1000
17:52:22 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:23 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 8.70 [s], freq 1000
17:52:23 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:24 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 9.70 [s], freq 1000
17:52:24 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:25 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 10.70 [s], freq 1000
17:52:25 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:26 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 11.70 [s], freq 1000
17:52:26 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:27 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 12.70 [s], freq 1000
17:52:27 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:28 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 13.70 [s], freq 1000
17:52:28 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:29 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 14.70 [s], freq 1000
17:52:29 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:30 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 15.70 [s], freq 1000
17:52:30 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:31 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 16.70 [s], freq 1000
17:52:31 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:32 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 17.70 [s], freq 1000
17:52:32 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:33 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 18.70 [s], freq 1000
17:52:33 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:34 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 19.70 [s], freq 1000
17:52:34 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:35 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 20.70 [s], freq 1000
17:52:35 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:36 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 21.70 [s], freq 1000
17:52:36 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:37 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 22.75 [s], freq 1000
17:52:37 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:38 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 23.75 [s], freq 1000
17:52:38 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:39 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 24.70 [s], freq 1000
17:52:39 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:40 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 25.70 [s], freq 1000
17:52:40 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:52:41 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 26.70 [s], freq 1000
17:52:41 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
[... cut ...]
17:53:26 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 71.70 [s], freq 1000
17:53:26 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:53:27 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 72.70 [s], freq 1000
17:53:27 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}
17:53:28 SHT: Shutter 1: Real. 0, Start: 0, Stop: 20000, dir 1, motordelay 0, rtc: 73.70 [s], freq 1000
17:53:28 RSL: tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":1}}

stefanbode, 2019-12-28:
Ok, you are definitely outside the boundaries. I had a big in position calculation. See submit above. If you use this version and do a shuttersetclose I have strong believes it will work. Anyhow I need to bugfix this, even if normally no a valid position

stefanbode, 2019-12-28:
Ok this log looks not good at all. I’m sure I tested this before submitted.

to-scho, 2019-12-28:
Hi Stefan, a second please. I am still on original 8.1.0 ;-)

stefanbode, 2019-12-28:
I see you have not defined interlock on and interlock 1,2
Therefore I assume your first relay is on/off and the other for up down. Correct? This is shuttermode 1

stefanbode, 2019-12-28:
I added a Lot of functionality for stepper motors. This can interfere with the shuttermode 1.but it should not. It definitely does not interfere with any other mode

stefanbode, 2019-12-28:
I checked the code and the behavior we see is strange. Only in shuttermode 3 the position is calculated by counter and not time based. Do you have defined additional PWM and Counter? It not a problem. Just for getting an idea what went wrong

to-scho, 2019-12-28:
Log above is clean 8.1.0 after erasing flash, setting up wifi and configuring GPIOs. No interlock nor PWM etc. I will go back now to my real setup having relay1 as on/off and relay2 as direction switch.

stefanbode, 2019-12-28:
I’m very limited in testing at the beach. Will do my best.

to-scho, 2019-12-28:
Back on my original setup from this post. Log below is from 8.1.0.1 (your commit above). I do not see any change in behavior.  Take your time for the fix, I downgraded my shutter to 7.1.0 and trials/logs are from a clone only. Have a nice vacation!

19:08:39 SHT: Shutter 1 (Relay:1): Init. Pos: 0 [0 %], Open Vel.: 100 Close Vel.: 110 , Max Way: 66000, Opentime 33.0 [s], Closetime 30.0 [s], CoeffCalc: c0: 300, c1 500, c2: 700, c3: 900, c4: 1000, binmask 3, is inverted 0, shuttermode 1,motordelay 18
19:13:07 CMD: shuttersetclose1
19:13:07 MQT: stat/markise-test/RESULT = {"ShutterSetClose1":"Configuration reset"}
19:13:13 SHT: Wait for Motorstop..
19:13:14 MQT: stat/markise-test/RESULT = {"POWER1":"OFF"}
19:13:14 MQT: stat/markise-test/POWER1 = OFF
19:13:14 MQT: stat/markise-test/RESULT = {"POWER2":"OFF"}
19:13:14 MQT: stat/markise-test/POWER2 = OFF
19:13:14 MQT: stat/markise-test/RESULT = {"POWER1":"ON"}
19:13:14 MQT: stat/markise-test/POWER1 = ON
19:13:14 MQT: stat/markise-test/RESULT = {"ShutterPosition1":100}
19:13:14 RUL: POWER1#STATE=1 performs "var4 0"
19:13:14 MQT: stat/markise-test/RESULT = {"Var4":"0"}
19:13:14 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 0.20 [s], freq 220
19:13:14 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:15 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 1.00 [s], freq 1000
19:13:15 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:16 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 2.10 [s], freq 1000
19:13:16 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:17 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 3.10 [s], freq 1000
19:13:17 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:18 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 4.10 [s], freq 1000
19:13:18 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:19 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 5.10 [s], freq 1000
19:13:19 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:20 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 6.10 [s], freq 1000
19:13:20 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:21 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 7.10 [s], freq 1000
19:13:21 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:22 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 8.15 [s], freq 1000
19:13:22 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:23 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 9.10 [s], freq 1000
19:13:23 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:24 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 10.10 [s], freq 1000
19:13:24 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:25 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 11.10 [s], freq 1000
19:13:25 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:26 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 12.10 [s], freq 1000
19:13:26 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:27 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 13.10 [s], freq 1000
19:13:27 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:28 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 14.10 [s], freq 1000
19:13:28 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:29 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 15.10 [s], freq 1000
19:13:29 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:30 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 16.10 [s], freq 1000
19:13:30 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:31 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 17.10 [s], freq 1000
19:13:31 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:32 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 18.10 [s], freq 1000
19:13:32 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:33 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 19.10 [s], freq 1000
19:13:33 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:34 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 20.10 [s], freq 1000
19:13:34 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:35 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 21.10 [s], freq 1000
19:13:35 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:36 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 22.10 [s], freq 1000
19:13:36 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
[... cut ...]
19:13:45 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 31.15 [s], freq 1000
19:13:45 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:46 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 32.10 [s], freq 1000
19:13:46 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}
19:13:47 SHT: Shutter 1: Real. 0, Start: 0, Stop: 66000, dir 1, motordelay 18, rtc: 33.10 [s], freq 1000
19:13:47 MQT: tele/markise-test/RESULT = {"Shutter1":{"Position":0,"direction":1}}

to-scho, 2019-12-29:
Hi Stefan, to let you sleep better ...;-)

It's a classic one in line 397:
    if ( Shutter.mode = SHT_OFF_ON__OPEN_CLOSE_STEPPER) {
    if ( Shutter.mode == SHT_OFF_ON__OPEN_CLOSE_STEPPER) {

to-scho, 2019-12-29:
BTW, is it possible to release (off) on/off relay in shutter mode 1 after any movement? Currently I am using a rule to do so but that's clumpy.

stefanbode, 2019-12-29:
Thanks for debugging. I submitted a change request. This is already implement. Relay 2 goes always of after movement.

to-scho, 2019-12-29:
It's about Relay 1 to be off, too. Both Relays should not drain power when shutter is idle.

stefanbode, 2019-12-29:
Relay 1 should not be an issue at all because this is in you configuration the power relay this will be switched off because otherwise the shutter will run infinite. Relay2 was an issue because this is only the direction. Normally there is no need to switch this off. In mode 2 for sure relays will switch off anyway.

to-scho, 2019-12-29:
Your are right, I missed the change handling Relay 2. It's switched off now without my rule which I made for initial version. I flashed your recent fix and can confirm that it solved my issue. Thanks, Tobias

closed 2019-12-29 by to-scho