arendst/Tasmota #7354: Can't update firmware to esphome or tasmota binaries pre 20 Dec 2019 (7.1.2.6)

Diramu, 2019-12-28:
I have a device with lated tasmota 8.1.0.  When I tried to "update firmware" to a  esphome binary, it failed with a "Not compatible" message.

I tried updating a similar device running tasmota 7.0.0.3 with the same esphome binary it worked.

Going back to the original device I tried to updated to this older tasmota, it also failed, but with "Upload buffer miscompare"

Then I tried several older version old tasmota-minimal.  It would successfully load any version back until  20 Dec 2019.  So it would load version 8.1.0.1,8.0.0.1,and 7.1.2.6.   Any older version would fail.

Whenever these old versions would try to upload tasmota 7.0.0.3 or esphome, they would report just the "Not compatible".

closed 2019-12-28 by close-issue-app[bot]

meingraham, 2019-12-28:
Tuya-Convert can flash any custom firmware YOU give it. If you want to flash ESPHome, then just point TC to that binary. You don't have to flash Tasmota first to then go to some other firmware. So, users who ACTUALLY want Tasmota on their devices... want the latest and greatest version. They should not be handicapped by a user who actually wants to flash something else on their device. You probably need to explore the capabilities of TC better. This boils down to the same issue. Read the documentation for TC. Just like it is always recommended that you read the Tasmota documentation. So, when you are about to upgrade... read the upgrading article. Etc.

mkormendy, 2020-01-10:
@meingraham the OP isn't talking about using Tuya-Convert to flash the custom firmware. His device is already flashed with the Tasmota firmware.

The first line is his key sentence:

> I have a device with lated tasmota 8.1.0.  When I tried to "update firmware" to a esphome binary, it failed with a "Not compatible" message.

meingraham, 2020-01-10:
Yes, I know he NOW has Tasmota on the device and wants to migrate to ESPHome. There were other posts from Diramu that you did not see asking that Tuya-Convert include an earlier version of Tasmota that allows migrating to ESPHome. My response was that Tasmota users want the latest version of Tasmota and that if users want ESPHome, then they are able to flash ESPHome directly without dependency on Tasmota versions. The fact that the OP already flashed Tasmota and no cannot flash ESPHome was already answered. Tasmota has had a memory map change making so one cannot go backward... so Tasmota is looking for a signature in the binary to determine compatibility. In the case of ESPHome, Tasmota does not find that, of course. Thus Tasmota will not flash an incompatible version to ensure that the device is not rendered inoperable. Unfortunately for this user, the ESPHome documentation is out of date.

SetOption78 will only work because there was a bug in an early version of the compatibility check. Using it is not a guarantee that the user will end up with an operational device. It is not documented for that reason. It may be deprecated in the future once time elapses to allow those early maintenance releases of Tasmota 8 are no longer in the wild. We recommend that option only in "emergency" situations. Using this to work around the ESPHome migration issue MAY work... at this time. But ESPHome users should not depend on it or expect it to work. ESPHome should update their documentation. Also, perhaps they can reach out to Theo to see if they can work out a "compatibility" mechanism... but ultimately this is not Tasmota's responsibility.

Net-net, if a user want ESPHome on their device, flash ESPHome directly from Tuya-Convert.

rudders, 2020-01-11:
Thanks @mkormendy ! Worked a treat.

martimarkov, 2020-01-19:
Wow... Some real hate towards ESPHome here, @meingraham 

Whatever the reason for ppl wanting to switch to a different firmware that is their right. If Tasmota does block `SetOption78 1` in the future to "protect the end-user" it will become the same shit firmware that everyone here is trying to escape from. 

At the end of the day, we should be allowed to do whatever we want with our devices. If for some reason let's say I want to flash an older version or something else (like my own custom firmware not based on either ESPHome or Tasamota) then I should be allowed.

meingraham, 2020-01-19:
@martimarkov 

> Net-net, if a user wants ESPHome on their device, flash ESPHome directly from Tuya-Convert.  

Hate? It does not mean what you think it means.

I posted facts as I know them from having discussed both compatibility and the intended use of SetOption78 with Theo. It is based on my dialog with Theo that we decided it was best to NOT include it in the docs. Both as a safeguard to the user and since it will likely be removed in the future.

The compatibility check in Tasmota was not added to keep users from flashing ESPHome on their devices. It was done due to continuous enhancements in Tasmota. The settings structure stored in flash was completely changed. To protect users of earlier versions of **Tasmota** from unintentionally making their devices inoperable, the compatibility check was added. It also happens to protect ESPHome users from bricking their devices since incompatible settings maps will cause the OTA upload to fail.

The device is yours, no argument there. Do with it what you please. It is entirely the user's choice as to whether they use Tasmota - which is provided at no cost.

Theo, the owner of the Tasmota project, can chose what features to include or remove. As the owner of the project he is completely allowed to do whatever he wants with HIS software project. He owes nothing to you or to any of us.

In addition to not knowing about the details and reasons for the compatibility check. you are probably not fully informed about the fact that ESPHome has recently added 12 bytes to its firmware images in order to "fake" Tasmota into finding ESPHome binaries as "compatible". I am basing this on having read the ESPHome commit. There is no guarantee by Tasmota that the "signature" itself or its location in the binary may change due to normal changes to the Tasmota code. Just be aware, and beware. In addition, Tasmota now supports uploading compressed firmware binaries. The signature for `.gz` Tasmota binaries is different... which may make the ESPHome upload inoperable. Be aware, and beware.

None of this is hate. It's just information. You may not like it, as you've made clear. But Tasmota has no responsibility to ESPHome users. My previous recommendation holds - **_if a user wants ESPHome on their device, flash ESPHome directly_**. Or jump through hoops and flash an older version of Tasmota so that you can OTA upload ESPHome. Just realize that jumping between Tasmota version may have the same unintended outcome - an inoperable device.

Regards.

martimarkov, 2020-01-19:
TBH I agree with most of the things you've said. 

I just don't want Tasmota to lose the ability to flash a different firmware OTA. So as long as SetOption78 allows the bypass of the verification check I'm happy to use Tasmota for some things and some custom or even ESPHome firmware for other things. 

BTW ignoring the whole tuya-convert, I'm more talking about flashing something after I decided to switch away for whatever reason. But hey so far Tasmota allows me to do this and is the reason I still use it. I just wanted to point out this is an important feature for me.

meingraham, 2020-01-19:
As I said, do not depend on `SetOption78`.

OTA updates are convenient, but not guaranteed to work for a variety of reasons, not always firmware feature related. When all else fails, as long as the ESP82xx is operational (and reachable), serial flashing will work to upload any compatible firmware.

And if all avenues are blocked, the saving grace is that most of these devices are pretty inexpensive.

Regards.

P.S. Glad to see you have cooled off. Just be cognizant of our  [code of conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

BartKoppers, 2020-02-25:
I was playing around with some different firmware options, after 'Tuya Converting'. 
All OTA.
After an initial flash and try of Tasmota, I wanted to try ESPHome, because some functionality I'd 
like to try. This failed.
After  **quite some searching** I came across these thread, and SetOption78 path resolved blocking the OTA flashing to ESPHome. 
That SetOption78 isn't in the current docs anymore is strange, at least to say.

meingraham, 2020-02-25:
https://github.com/arendst/Tasmota/issues/7354#issuecomment-573047413
> SetOption78 will only work because there was a bug in an early version of the compatibility check. Using it is not a guarantee that the user will end up with an operational device. It is not [...]

It has never been in the docs.

BartKoppers, 2020-02-26:
The issue is not about Docs (only).
It’s about being able to circumvent an introduced safety-feature where this feature blocks the described use-case.
See earlier comments.

(If that’s not the issue, and ESPHome is the issue: what should/could ESPHome change?)

meingraham, 2020-02-26:
From what I understood, ESPHome already added the signature bytes to "fool" Tasmota.