arendst/Tasmota #7360: Shutter position updates via mqtt are not following previous conventions

he-so, 2019-12-28:
**Have you looked for this feature in other issues and in the docs?**  
yes

**Is your feature request related to a problem? Please describe.**  
The tasmota adapter in iobroker does not recognize the mqtt messages with topic tele/RESULT.

Currently, the shutter driver sends the following after the shutter stops moving:
tele/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":0}}

Iobroker, for my understanding, expects either stat/RESULT or tele/SENSOR, since that will so far cover all relevant messages. So the shutter driver breaks this convention.

**Describe the solution you'd like**  

Therefore I'd like to request the shutter driver to send:
stat/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":0}}

https://github.com/arendst/Tasmota/blob/development/tasmota/xdrv_27_shutter.ino#L350

Change RESULT_OR_TELE to RESULT_OR_STAT

**Describe alternatives you've considered**  
One could ask the iobroker tasmota adapter to also process tele/RESULT messages, but since this is so far not usual, I'd prefer to adjust the shutter driver.

**Additional context**  
I have tested the above mentioned change, an it's working es expected.

**(Please, remember to close the issue when the problem has been addressed)**

stefanbode, 2019-12-29:
Agree, this should be in sync with all other messages. Let me double check

stefanbode, 2019-12-29:
The request was to get updates of the shutterposition and the movement every second the shutter moves. We now have two  options. Using the full teleperiod syntax and Post this every second during movement. Or second option is posting the RESULT every second as a stat message.  Normally you only get ONE result on a event and not many. Open to discuss with the community what is best in this case. No other driver has this requirement so far.

he-so, 2019-12-29:
I thought about publishing events every second. That would be really nice to show, but at least HomeKit seems not to update the GUI for shutters. So the benefit for me would be little.

Anyhow, when publishing intermediate position values, it should be tele/STATE and not RESULT. The iobroker adapter would process that.

ascillato2, 2019-12-29:
I think that the concept of telemetry is to send periodically a summary of variables to the broker. In the other hand, the stat/topic/RESULT is the instant result of an event. In the case of shutters you have an ongoing event with a serie of steps, so seems that the correct path to follow is stat/topic/RESULT for each step, due to those are just the instant result of that ongoing event.

A tele message should be used only for updating the whole status of the Tasmota device IMHO. 

So, seems that the request on this issue is aligned to that and to the rest of the drivers:

> Therefore I'd like to request the shutter driver to send:
> stat/tasmota/RESULT = {"Shutter1":{"Position":0,"direction":0}}

So, IMHO, this change should be implemented.

PR #7511 (2020-01-14): Change MQTT reporting of instantaneous shutter position from tele to …
…stat

## Description:

**Related issue (if applicable):** fixes #7360

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.6.1
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2020-01-14 by arendst