arendst/Tasmota #7365: [question] Running Tasmota on a ESP8266 with 512KB of RAM?

j4k3, 2019-12-29:
Sorry for bothering everybody, but I can't find a definitive answer to this question in the docs: Is it possible to run Tasmota on a system with only 512KB of ROM or is at least 1M a requirement?

To be more precise, I've been trying to get Tasmota to run on a Sparkfun Thing dev [https://www.sparkfun.com/products/13711](https://www.sparkfun.com/products/13711), which is an early ESP8266 dev board with only 512KB of RAM. 

I've tried the lite and minimal versions of the latest 8.1 doris as well as numerous other releases to run. 

I've installed Tasmota on a bunch of other devices, and the Thing Dev itself is working fine, other Arduino sketches run fine, including Wifi communication. I have also successfully uploaded other compiled binaries with esptool.py from the command line. When I upload the lite or minimal binaries of Tasmota, everything in the esptool.py checks out the same way. I've erased the flash before reprogramming and the process ends with the hash matching, everything fine and dandy. But after rebooting, the board doesn't seem to do anything sensible. No wireless network is established, and the serial console just outputs garbled data.

As mentioned, I've just installed the newest release on other devices including a No-name smart bulb, three Blitzwolf SHP6, and I can also get it to work on a ESP01 but, I'd like to fiddle with it on a development board. 
Am I just doing something wrong or is impossible to run Tasmota in 512k and I will have to put my smd soldering skills to the test?

closed 2019-12-29 by close-issue-app[bot]

meingraham, 2019-12-29:
The default memory map for Tasmota is 1MB and the default features compiled cause the binary to be larger than 512K. To use a device with 512K you will have to change the memory parameters for proper compilation AND you will have to disable enough features to get the binary size so it fits in 512K. However, once you flash the device via serial, you will ALWAYS have to flash any updates via serial because OTA flash requires flash to be ~2x the size of the final running binary. With 512K you will never have enough free memory to fit the two binaries concurrently.

Net-net - get yourself an ESP device with 1MB of flash memory or greater (and anything over 1MB is usually a waste).

b0rder, 2020-05-03:
For those who has 512k devices and want to use Tasmota.
As mentioned before, it is need to [build Tasmota](https://tasmota.github.io/docs/Compile-your-build/) firmware with minimal modules. If you still want WEB gui, follow instruction below (to build using platformio).
1. Create ./src/Tasmota/platformio_override.ini with the follow content:
```
[common]
build_flags               = ${core_active.build_flags}
                          -DUSE_CONFIG_OVERRIDE
board                     = esp01
board_build.ldscript      = eagle.flash.512k32.ld
```
2. Create ./src/Tasmota/tasmota/user_config_override.h with the follow content:
```
#ifndef _USER_CONFIG_OVERRIDE_H_
#define _USER_CONFIG_OVERRIDE_H_
#undef USE_EMULATION
#undef USE_EMULATION_HUE 
#undef USE_EMULATION_WEMO
#endif
```
3. Build firmware:
`pio run -e tasmota-lite
`
4. Upload compiled firmware to ESP:
esptool.py --port $DEV --baud 921600 write_flash -fm dout 0x00000 .pioenvs/tasmota-lite/firmware.bin

Instead of $DEV use your own port name.