arendst/Tasmota #7372: FR - Change MQTT Message for Longitude and Latitude

SeLLeRoNe, 2019-12-29:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

I am testing the GPS integration which is working perfectly, but I found an "issue".

In Home Assistant is present the ability to set a device_tracker ased on mqtt_json formatted (https://www.home-assistant.io/integrations/mqtt_json/)

The only problem with this is that HA is expecting "longitude", "latitude" and eventually "gps_accuracy" while Tasmota is sending "lon" and "lat" "hAcc" "vAcc".

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
Would be possible to change the naming so that the integration with HA can be smooth?

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._
Contacting HA Devs to have some option to customize the expected naming

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Staars, 2019-12-30:
Hi, Thank you for your interest.
While I see no problem in changing the naming conventions of a very new driver, I do generally not like long message names on small MCU's.

Before we eventually change the driver code, could you please check, if the following RULE already solves your problem:

`Rule1
ON GPS#lat DO Var1 %value% ENDON
ON GPS#lon DO Var2 %value% ENDON
ON GPS#hAcc DO publish tele/GPS {"latitude:“%var1%,"longitude:“%var2%,“gps_accuracy:“%value%} endon `

Maybe you have to adjust the last line a bit to taylor the payload and topicname to your needs.

SeLLeRoNe, 2019-12-30:
Hi @Staars 
Thanks, I will give this a try :)
Out of curiosity, I doubt MCU stands for Marvel Cinematic Universe, so.. what's that? ^^

I will let you know if the rule improved the integration as soon as I have 5 mins to test it

Thanks

SeLLeRoNe, 2019-12-30:
Ok, 
I have updated the rule to be in the correct format (some errors in the quotes
`Rule1 ON GPS#lat DO Var1 %value% ENDON ON GPS#lon DO Var2 %value% ENDON ON GPS#hAcc DO publish tele/Mercedes/GPS {"latitude":%var1%,"longitude":%var2%,"gps_accuracy":%value%} endon; Rule1 1`

This configuration seems to be working, but it is not definitivi because:
The message should use the topic configured instead of hard-coded
The message should follow the SENSOR retain instructions (for instance, I do retain the SENSOR message but this generated by the rule was not).

Beside those two points, I can now see the device-tracker working in HA (I don't have a fix yet but seeing the 0.0 it's a good sign) :)

I also notice the original payload have have "height" which I presume it's altitude? That naming in my opinion should be updated regardless because I think altitude it's the right term to use anyway

An example of payload (if you'll decide to update it) should be something likethis (in my opinion of  course):
`{"Time":"2019-12-30T11:32:59","GPS":{"latitude":0.0000000,"longitude":0.0000000,"altitude":0.000,"vertical_accuracy":0.000,"gps_accuracy":0.000}}`

Not even sure would be possible but I ask... would this be also able to provide "speed" at any point?

Thanks a lot for you work, it really give the ability to do much more with some gps-based ideas :)

Andrea

Staars, 2019-12-30:
I totally agree regarding altitude vs height. This will be changed (->alt or altitude).

Reading speed is technically possible, but must be added to the code. But before this is done, I would like to think about the command interface, because it might be useful to de-/select the values you want to see. 

I am pretty sure, it is not question "if" but "how" the problem can get solved.  

I will take a look later today.

arendst, 2019-12-30:
Just to chime in on the JSON message. Keep it as short as possible. Field names like ``vertical_accuracy`` always remind me on Home Assistance JSON field naming tending to be as long as possible just to tell a story instead of naming a field.

SeLLeRoNe, 2019-12-30:
Yep, that's exactly where I am trying to integrate it ^^
Maybe have a way to customize those naming so that defaults are short but can be customized for specific usages?

SeLLeRoNe, 2019-12-30:
Something additional that I figure while testing, in my opinion it might be better to have empty values (or "Unknown" when there is no fix, this because the default 0.0000000 for lon/lat  (and also for accuracy and altitude) are actual values that can simply put the GPS module in Greenwich :)

Jason2866, 2019-12-30:
@SeLLeRoNe  There is for sure a solution in Home Assistant.
As Theo already wrote he will not accept a change in long names.
Please close and ask for help in Home Assistant community to solve your task

SeLLeRoNe, 2019-12-30:
Well no there is no based on their docs, I will open an issue with them.

Anyway my last message had nothing to do with the naming and for now the rule is working just fine, at least for those initial tests.

SeLLeRoNe, 2019-12-30:
Opened an Issue at HA Github: https://github.com/home-assistant/home-assistant/issues/30296

Staars, 2019-12-30:
We will see.
For now I did a PR regarding the height<->altitude-thing, which makes sense to me. 

I still think, the most elegant way to expand the tele message could be a RULE. If you find a nice solution, please consider to add it to the WIKI.

A more brute-force-like solution would be some kind of
`#ifdef GPS_HA
  ....latitude: ....
#else 
...lat: ......
#endif //GPS_HA
` 
This would still be readable and only add 4 lines of code. I am still scratching my head about it.

SeLLeRoNe, 2019-12-30:
I am okay with the rule for now, but I would need help to have those "forwarded" messages to be retained please :)

meingraham, 2019-12-30:
@SeLLeRoNe 

You can use %topic% in rules to avoid hard coding the device topic.

For retain, sett `Publish2`

SeLLeRoNe, 2019-12-30:
So the rule I am currently using is:
Rule1 ON GPS#lat DO Var1 %value% ENDON ON GPS#lon DO Var2 %value% ENDON ON GPS#vAcc DO Var3 %value% ENDON ON GPS#hAcc DO publish2 tele/%topic%/GPS {"latitude":%var1%,"longitude":%var2%,"gps_accuracy":%value%,"vertical_accuracy":%var3%} endon;

Thanks :)

meingraham, 2019-12-30:
This issue has been resolved.

closed 2019-12-30 by meingraham