arendst/Tasmota #7373: External Switch / Sensor not working with 8.x

b0wtie, 2019-12-30:
### PROBLEM DESCRIPTION
I have Tasmota installed on a Sonoff 4CH Pro. I've wired a Sensor and a Switch to the TX/RX Ports (GPIO1 and GPIO3). They worked without a problem on Tasmota prior to 8.x. If I upgrade, they don't work anymore. There might just be a change in the Tasmota firmware but I can't find it on the internet. I am very sorry, if this is already documented anywhere.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff 4CH Pro
- [x] Tasmota binary firmware version number used: 7.1.2 (working), 8.0.0.1 and above not working
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: unneccessary
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
[17,255,255,255,23,22,18,19,21,56,20,24,0],"FLAG":0,"BASE":23}
{"Module":{"23":"Sonoff 4CH Pro"}}
{"GPIO0":{"17":"Button1"},"GPIO1":{"10":"Switch2"},"GPIO2":{"25":"Relay5"},"GPIO3":{"13":"Switch5"},"GPIO4":{"23":"Relay3"},"GPIO5":{"22":"Relay2"},"GPIO9":{"18":"Button2"},"GPIO10":{"19":"Button3"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"20":"Button4"},"GPIO15":{"24":"Relay4"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
// Not using rules
```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
06:54:01 MQT: stat/sonoff_4ch/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff","Sonoff2","Sonoff3","Sonoff4"],"Topic":"sonoff_4ch","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,3,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
06:54:01 MQT: stat/sonoff_4ch/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T12:52:25","StartupUTC":"2019-12-29T17:01:36","Sleep":50,"CfgHolder":4617,"BootCount":7,"SaveCount":22,"SaveAddress":"F5000"}}
06:54:01 MQT: stat/sonoff_4ch/STATUS2 = {"StatusFWR":{"Version":"7.2.0(tasmota)","BuildDateTime":"2019-12-22T11:41:04","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"286/1151"}}
06:54:01 MQT: stat/sonoff_4ch/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["GreatBramels_ac",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005AFF000000000000","00000000","00000000"]}}
06:54:01 MQT: stat/sonoff_4ch/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
06:54:01 MQT: stat/sonoff_4ch/STATUS5 = {"StatusNET":{"Hostname":"sonoff_4ch-5671","IPAddress":"192.168.69.50","Gateway":"192.168.69.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.69.1","Mac":"DC:4F:22:FA:16:27","Webserver":2,"WifiConfig":4}}
06:54:01 MQT: stat/sonoff_4ch/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.69.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_FA1627","MqttUser":"b0wtie","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
06:54:01 MQT: stat/sonoff_4ch/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 30 05:54:01 2019","Local":"Mon Dec 30 06:54:01 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:01"}}
06:54:01 MQT: stat/sonoff_4ch/STATUS10 = {"StatusSNS":{"Time":"2019-12-30T06:54:01","Switch2":"ON","Switch5":"ON"}}
06:54:01 MQT: stat/sonoff_4ch/STATUS11 = {"StatusSTS":{"Time":"2019-12-30T06:54:01","Uptime":"0T12:52:25","UptimeSec":46345,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"GreatBramels_ac","BSSId":"B0:7F:B9:3E:50:3A","Channel":5,"RSSI":38,"Signal":-81,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
should I upgrade to 8.x and post the log output of the above again?
```

### TO REPRODUCE
It does not matter if I upgrade between 7.2.0 without resetting or if I use "Reset 6", then upgrade and then restoring my configuration. The effect is the same: With 8.x my configuration does not react to the switch / sensor wired to tx/rx 

### EXPECTED BEHAVIOUR
I'd expect the same thing to happen as with 7.2.0: When I press the switch / when the magnetic sensor get's closed, the device should react to it.

### SCREENSHOTS
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-30:
Numbering of switches has to begin with 1 and second switch has to be 2.

digiblur, 2019-12-30:
Try the latest dev build as I also had this issue with GPIO1 and 3 sensors/devices and Theo fixed it a couple days ago.

b0wtie, 2019-12-30:
> Numbering of switches has to begin with 1 and second switch has to be 2.
> The configuration used before was not correct and it was just luck it worked.

It is possible, even likely that I misconfigured the device. However, what I need to achieve is, that when I press the button wired to GPIO1 it should have the same effect as the "Button 2". The second relay should connect, the device should do a "POWER 2".
I set GPIO1 to Switch 2 because I thought this would connect my external button with the second relay. Was that not correct? Should I set it to "Button 2" then?

I'll also try the latest dev build and report. Thanks for the advice.

meingraham, 2020-01-01:
You should be able to configure `Switch2` so that it corresponds to `Relay2`. I do this with a `Switch4` on my Sonoff 4Ch Pro to control the light on `Relay4`.

After you try the latest build, please address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-01 by meingraham