arendst/Tasmota #7374: Feature: depend timers by sunrise / sunset

mobifu1, 2019-12-30:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
Example: switch an outdoor lamp for lighting a property area.
When i set a timer1 to switch on this lamp at 06:00 and set timer2 to switch off this lamp at sunrise, this setting will only work in the winter. In the sommer the sunrise begins befor 06:00 o'clock and works **not** fine for this case.

**Describe the solution you'd like**  
May be i need a feature to combine a second timer.

Example:
Timer1: switch on at 06:00 o'clock
Timer2: switch off at sunrise
Timer3: switch on at sunset
Timer4: switch off at 21:00 o'clock

if (sunrise > timer1.value) execute.function.timer1;
if (sunset < timer4.value) execute.function.timer3;

**Describe alternatives you've considered**  
May be on the timer card, were i use the sunrise / sunset function, i can imagine a link to a second depend timer.

**Additional context**  
Thank you for that nice Project!!!

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-12-30:
There are several time & sun based rules in the [rules cookbook](https://tasmota.github.io/docs/#/Rule-Cookbook?id=make-sure-light-is-on-at-night) which should solve your problem.

If you have any further questions, please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-12-30 by meingraham

mobifu1, 2019-12-30:
Hallo Michael, 
YES, that works fine, thank you for the solution.
Regards Andreas

von unterwegs gesendet
mfG A.Jahnke