arendst/Tasmota #7377: Hold for reset not working on my switches with setoption1 and setoption13 set to 1

SylvainGa, 2019-12-30:
### PROBLEM DESCRIPTION
With Wificonfig 5, setoption1 1 and setoption13 1, when I hold the single switch (KS-602S) button down for 40 seconds, it sets Option13 to 0 (as shown on the console screen) but because setoption1 is set to 1, I cannot do triple, quadruple, etc button press. I believe that setoption1 should also be reset to 0 at the same time as option13 when the button is held for over 10 x setoption32. If that's the case, I think adding this at line 199 of support_button.ino would fix it

              snprintf_P(scmnd, sizeof(scmnd), PSTR(D_CMND_SETOPTION "1 0"));  // Disable limiting to single, double press and hold
              ExecuteCommand(scmnd, SRC_BUTTON);

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [X] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Generic (18) - KS-602S
- [X] Tasmota binary firmware version number used: 8.1.0 Release
  - [X] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [X] Flashing tools used: Tuya-Convert
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`: ```
  Configuration output here:

Not pertinent

```
- [X] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

Not pertinent
```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
16:36:43 MQT: LumiereExterieur/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["LumiereExterieur"],"Topic":"LumiereExterieur","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
16:36:43 MQT: LumiereExterieur/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:18:43","StartupUTC":"2019-12-30T15:18:00","Sleep":50,"CfgHolder":4617,"BootCount":16,"SaveCount":54,"SaveAddress":"FA000"}}
16:36:43 MQT: LumiereExterieur/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"336/699"}}
16:36:43 MQT: LumiereExterieur/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BELL776","Test"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000AA009","2805C8000100060000005A00000000000000","00000200","00000000"]}}
16:36:43 MQT: LumiereExterieur/stat/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
16:36:43 MQT: LumiereExterieur/stat/STATUS5 = {"StatusNET":{"Hostname":"LumiereExterieur-0996","IPAddress":"192.168.2.100","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"68:C6:3A:EE:83:E4","Webserver":2,"WifiConfig":2}}
16:36:43 MQT: LumiereExterieur/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.84","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_EE83E4","MqttUser":"mqtt","MqttCount":6,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
16:36:43 MQT: LumiereExterieur/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 30 15:36:43 2019","Local":"Mon Dec 30 16:36:43 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:01"}}
16:36:43 MQT: LumiereExterieur/stat/STATUS10 = {"StatusSNS":{"Time":"2019-12-30T16:36:43"}}
16:36:43 MQT: LumiereExterieur/stat/STATUS11 = {"StatusSTS":{"Time":"2019-12-30T16:36:43","Uptime":"0T00:18:43","UptimeSec":1123,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":6,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BELL776","BSSId":"A8:9A:93:FD:CC:2E","Channel":6,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [X] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

Not pertinent
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Set the options as follow:
Wificonfig 5
setoption1 1
setoption13 1
Switch to console view
Hold button for 40 seconds and all it does is setoption13 to 0.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

setoption1 should also be set to 0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-30:
SetOption1 1 is for prevent reseting with long press
See wiki https://tasmota.github.io/docs/#/Commands?id=setoptions

SylvainGa, 2019-12-30:
It actually to prevent anything other than a single press, double press or
hold press, not just long press.

SylvainGa, 2019-12-30:
In my case, holding the button do NOT reset to default but only set option13 to 0.

Jason2866, 2019-12-30:
`SetOption13 1`  disables multipress functions! 
[image]

SylvainGa, 2019-12-30:
Yes, and holding the button for 10 x SetOption32 resets SetOption13 to 0
but because SetOption1 is at 1, holding the button for 10 x SetOption32
does NOTHING else.

Jason2866, 2019-12-30:
As designed and explained in documentation under `setoption1`

SylvainGa, 2019-12-30:
Ok, first hold for 40 seconds did a reset of SetOption13 to 0 *(even with
SetOption1 at 1)* and a subsequent hold for 40 seconds did reset to factory
default WHEN the SetOption1 is a 0. So this is by design. I guess I'll have
to toggle the power to reset devices if needed be.

meingraham, 2019-12-30:
Answered.

closed 2019-12-30 by meingraham