arendst/Tasmota #7379: Dimmer keeps adding on 6% to all dimming values and no dimming function below 23%

Supermillhouse, 2019-12-30:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
Dimmer value overshoots consistently when set using either MQTT or the slider in the web interfaces. The dimmer value increases if the lamp is toggled off and then on again (observed to happen at least twice in two cycles of off and on again).

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): [RJWF-02A Dimmer](https://templates.blakadder.com/rjwf-02a.html) 
- [x] Tasmota binary firmware version number used: 8.1.0.2
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: tuya convert followed by OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
20:38:43 CMD: Backlog Template; Module; GPIO 255
20:38:43 SRC: WebConsole from 192.168.2.81
20:38:43 CMD: Group 0, Index 1, Command "BACKLOG", Data "Template; Module; GPIO 255"
20:38:44 SRC: Backlog
20:38:44 CMD: Group 0, Index 1, Command "TEMPLATE", Data ""
20:38:44 MQT: Dimmer_2/stat/RESULT = {"NAME":"RJWF-02A","GPIO":[17,107,0,108,0,0,0,0,0,0,52,0,0],"FLAG":0,"BASE":54}
20:38:44 SRC: Backlog
20:38:44 CMD: Group 0, Index 1, Command "MODULE", Data ""
20:38:44 MQT: Dimmer_2/stat/RESULT = {"Module":{"0":"RJWF-02A"}}
20:38:44 SRC: Backlog
20:38:44 CMD: Group 0, Index 1, Command "GPIO", Data "255"
20:38:44 MQT: Dimmer_2/stat/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"107":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"108":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"52":"Led1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
20:40:30 CMD: Status 0
20:40:30 SRC: WebConsole from 192.168.2.81
20:40:30 CMD: Group 0, Index 1, Command "STATUS", Data "0"
20:40:30 MQT: Dimmer_2/stat/STATUS = {"Status":{"Module":0,"FriendlyName":["Dimmer 2"],"Topic":"Dimmer_2","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:40:30 MQT: Dimmer_2/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:21:12","StartupUTC":"2019-12-30T19:19:18","Sleep":50,"CfgHolder":4617,"BootCount":22,"SaveCount":369,"SaveAddress":"F4000"}}
20:40:30 MQT: Dimmer_2/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0.2(1e06976-tasmota)","BuildDateTime":"2019-12-30T19:00:11","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"321/699"}}
20:40:30 MQT: Dimmer_2/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SuperIoT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","2805C8000100060000005A00000000000000","08080200","00000000"]}}
20:40:30 MQT: Dimmer_2/stat/STATUS4 = {"StatusMEM":{"ProgramSize":567,"Free":436,"Heap":23,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":0,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
20:40:30 MQT: Dimmer_2/stat/STATUS5 = {"StatusNET":{"Hostname":"Dimmer_2","IPAddress":"192.168.2.98","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.10","Mac":"2C:F4:32:19:18:94","Webserver":2,"WifiConfig":2,"WifiPower":17.0}}
20:40:30 MQT: Dimmer_2/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.2","MqttPort":1883,"MqttClientMask":"Dimmer 2","MqttClient":"Dimmer 2","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:40:30 MQT: Dimmer_2/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 30 19:40:30 2019","Local":"Mon Dec 30 20:40:30 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:01"}}
20:40:30 MQT: Dimmer_2/stat/STATUS10 = {"StatusSNS":{"Time":"2019-12-30T20:40:30"}}
20:40:30 MQT: Dimmer_2/stat/STATUS11 = {"StatusSTS":{"Time":"2019-12-30T20:40:30","Uptime":"0T00:21:12","UptimeSec":1272,"Heap":23,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":68,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SuperIoT","BSSId":"18:E8:29:FE:4D:16","Channel":11,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
00:00:17 MQT: Received Topic "Dimmer_2/cmnd/Dimmer", Data Size 2, Data "50"
00:00:17 SRC: MQTT
00:00:17 CMD: Group 0, Index 1, Command "DIMMER", Data "50"
00:00:17 MQT: Dimmer_2/tele/STATE = {"Time":"1970-01-01T00:00:17","Uptime":"0T00:00:19","UptimeSec":19,"Heap":27,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":1,"POWER":"ON","Dimmer":50,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"SuperIoT","BSSId":"18:E8:29:FE:4D:16","Channel":11,"RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:07"}}
00:00:17 MQT: Dimmer_2/stat/RESULT = {"POWER":"ON","Dimmer":50}
00:00:17 LGT: Channels 02020000000000000000
00:00:17 TYA: Send dim value=128 (id=2)
00:00:17 TYA: Send "55aa00060008020200040000008095"
[... 142 lines cut ...]
If I then toggle the lamp on and off with hardware button it sets the dimmer to 62
If I then toggle the lamp on and off with hardware button it sets the dimmer to 68
as shown in log

I think the toggling problem will be solved if the lamp actually set itself properly to 50 and didn't add 6% on to the value.
FYI if I set the dimmer to 44% the log will report 50% so it isn't like the tuya dimmer controller cant be set to 50%

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

I expected with min and max of 58 and 255 to be set that 0% would send 58 and 100% would send 255 and 50% to send roughly 98 but it sends 128 as per log

There are also some other weird behaviors where anything below 23% it just sends 58 but the percentage slider goes up to 23%

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

if max is purely the 100% value and min is just cropping off of the bottom this wasn't particularly obvious in the documentation, I just assumed it worked differently because now I have a dead bottom half of a slider just because tuya doesn't like values below 58.

however disregarding this i still don't know why every percentage value i send comes back with 6% added on to it, and subsequently keeps getting compounded every time the switch is toggled.

**(Please, remember to close the issue when the problem has been addressed)**

dalovering, 2020-01-13:
Hello! I started to do some digging on this today. 
I have a tuya dimmer (Treatlife DS01) that does something similar - though mine dims to 15% minimum.
Not sure about the 6% adder, but in the case of the minimum - this is a setting stored in the tuya MCU, not the esp8266. 
In the original firmware, there is a function to change this lower limit, but now that I have tasmota flashed I can’t determine the proper serial command to send to the mcu to set this range.

I wonder if anyone here has done this work yet, but my plan of attack is below. 

1. (DONE) Open the dimmer, solder onto serial pins, snoop on the esp8266 serial comms.
2. Flash back to stock firmware
3. Play with all settings possible while snooping to try to catalog all serial functions.
4. Flash back to tasmota 
5. Use console, TuyaSend, and newly learned functions to manually change settings
6. (If there is demand) Add functionality to the tamosta tuya-mcu module.

Unfortunately I was out of time for the weekend and my kitchen needs a light switch, so I’ll have to return to it another day.

KmanOz, 2020-01-15:
Yes I purchased a couple of the same dimmers as the OP and experienced the same issues. The original firmware was much better at dimming smoothly across the range however still not great at the lower limits.

KmanOz, 2020-01-21:
Let me add something that might help. I use HASS but the same issue can be created using the built in web interface. If you set the dimmer to 15% and then click toggle to turn off the light, then continuously click toggle to turn the light on then off, each time you do so the light increases by 6% so eventually after a few toggles the light becomes fully on. The slider however in the interface does not move.

ValliantStorme, 2020-01-21:
Have you tried disabling Gamma Correction (`LedTable`)? In my experience, a lot of Tuya-related things have hardware curves, and it causes issues like this, at least in my lightbulbs. How it applies to your dimmer, I'm not entirely sure, but it might be a good thing to check out.

[Relevant wiki page](https://tasmota.github.io/docs/#/Lights?id=gamma-correction)

KmanOz, 2020-01-21:
Set LedTable 0 made no difference.

dalovering, 2020-01-21:
I was able to fix the lower limit issue! 
In my case the TuyaMCU defined dpid 3 as dimmer low limit, and dpid 4 as dimmer mode (enum LED, incandescent, halogen). 

This is consistent with the default tuya “generic dimmer” template, so I suspect most will be similar. 

Once I found them I’m able to set the dimmer lower limit to “x” in MCU scale with: 
`TuyaSend2 3,x`

You will have to do some digging to ensure your dpids are the same. A good start is to send this:
`SerialSend5 55aa0001000000 `
Which will request the status of all dpids. You should see one of data type “int” with the value of the low limit you see (150 in my case)

On increasing by 6% each time, can you test this after setting:
`Dimmerrange 0,255`
I think it may be a scaling issue. 
Also ensure that `Fade OFF` As I was having strange issues with fade on mine. 

Hope this helps!!

ascillato2, 2020-01-22:
Thanks for sharing your solution. We should add this information to the docs. Thanks

closed 2020-01-22 by ascillato2

KmanOz, 2020-01-27:
LOL how is the answer a solution and it is CLOSED? The original question has not been answered. What a waste of space a forum like this is.

ascillato, 2020-01-27:
Hi, @KmanOz 

The steps explained by @dalovering didn't solve your issue? Please share what you have done and why it is not solved. Thanks.

ovidiugm, 2020-03-20:
The instructions above attempt to fix the lower dimming level, not level increase on toggling on/off.

I observe the same issue with my dimmer freshly flashed with Tasmota 8.1.0.2, where on each light toggle the dimmer level increases. Any thing I could try to fix this?
The issue is not present when cycling on/off by short pressing the hardware button on the module itself.

 I am attaching the relevant files. Thanks!
[Tasmota_Backlog_Template_Module_GPIO_255.txt](https://github.com/arendst/Tasmota/files/4360211/Tasmota_Backlog_Template_Module_GPIO_255.txt)
[Tasmota_Status_0.txt](https://github.com/arendst/Tasmota/files/4360212/Tasmota_Status_0.txt)
[Tasmota_Weblog_4.txt](https://github.com/arendst/Tasmota/files/4360213/Tasmota_Weblog_4.txt)
[Tasmota_Backlog_Rule1_Rule2_Rule3.txt](https://github.com/arendst/Tasmota/files/4360214/Tasmota_Backlog_Rule1_Rule2_Rule3.txt)

[image]

On power on / off dimmer level is increased, I am attaching weblog 4 output.

KmanOz, 2020-03-21:
@ovidiugm You're far more patient than myself. This is still closed with no real resolution.

martin072, 2020-09-05:
> I was able to fix the lower limit issue!
> In my case the TuyaMCU defined:
> `dpid 3 = Dimmer low limit, (int range 0-1000)`
> and
> `dpid 4 = dimmer mode, (enum LED, incandescent, halogen)`
> 
> This is consistent with the default tuya “generic dimmer” template, so I suspect most will be similar.
> 
> Once I found them I’m able to set the dimmer lower limit to “x” in MCU scale with:
> `TuyaSend2 3,x`
> 
> You will have to do some digging to ensure your dpids are the same. A good start is to send this:
> `SerialSend5 55aa0001000000 `
> Which will request the status of all dpids. You should see one of data type “int” with the value of the low limit you see (150 in my case)
> 
> On increasing by 6% each time, can you test this after setting:
> `Dimmerrange 0,255`
> I think it may be a scaling issue.
> Also ensure that `Fade OFF` As I was having strange issues with fade on mine.
> 
> Hope this helps!!

@dalovering 
May I ask how you find out the dpid's for the minimum dimmer range? 
I just successfully flashed a tuya dimmer with the latest tasmota 8.4.0 and noticed when the dimmer range is all the way down to 1 it still shows a relative bright light. I understand this is a mcu limit and not tasmota, and just wondering if I could change it to dim just a little more. 
All seems to work fine with module 54 selected and added the TuyaMCU 21,2 command.  

This is my first tuyamcu unit I have, so not that experienced with it. 

thanks!

dalovering, 2020-09-06:
@martin072 I kind of got lucky with mine - I made a tuya developer account and began "creating a new device" using the dimmer template, and my dimmer was using the default dpids for everything. 

Can you send `SerialSend5 55aa0001000000` in the console and paste the result here? Will be easier to help troubleshoot if we see what your dpid's are all set at.

In my case I have: 

```
04:37:24 CMD: Group 0, Index 5, Command "SERIALSEND", Data "55aa0001000000"
04:37:24 MQT: stat/kitchenLights/RESULT = {"SerialSend":"Done"}
04:37:24 {"TuyaReceived":{"Data":"55AA0301002A7B2270223A2272336667317974796434736436376761222C2276223A22342E312E32222C226D223A307DA2","Cmnd":1,"CmndData":"7B2270223A2272336667317974796434736436376761222C2276223A22342E312E32222C226D223A307D"}}
04:37:24 TYA: MCU Product ID: {"p":"r3fg1ytyd4sd67ga","v":"4.1.2","m":0}
04:37:24 TYA: Send "55aa0002000001"
04:37:24 {"TuyaReceived":{"Data":"55AA0302000004","Cmnd":2}}
04:37:24 TYA: RX MCU configuration Mode=0
04:37:24 TYA: Read MCU state
04:37:24 TYA: Send "55aa0008000007"
04:37:24 {"TuyaReceived":{"Data":"55AA03070005010100010112","Cmnd":7,"CmndData":"0101000101","1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
04:37:24 TYA: fnId=11 is set for dpId=1
04:37:24 TYA: RX Relay-1 --> MCU State: On Current State:On
04:37:24 {"TuyaReceived":{"Data":"55AA030700080202000400000096AF","Cmnd":7,"CmndData":"0202000400000096","2":{"DpId":2,"DpIdType":2,"DpIdData":"00000096"}}}
04:37:24 TYA: fnId=21 is set for dpId=2
04:37:24 TYA: RX Dim State=150
04:37:24 {"TuyaReceived":{"Data":"55AA030700080302000400000094AE","Cmnd":7,"CmndData":"0302000400000094","3":{"DpId":3,"DpIdType":2,"DpIdData":"00000094"}}}
04:37:24 TYA: fnId=0 is set for dpId=3
04:37:24 {"TuyaReceived":{"Data":"55AA03070005040400010017","Cmnd":7,"CmndData":"0404000100","4":{"DpId":4,"DpIdType":4,"DpIdData":"00"}}}
04:37:24 TYA: fnId=0 is set for dpId=4
04:37:30 WIF: Checking connection...
04:37:30 WIF: Connected
```

The lower limit (for me) is stored in `dpid 3` and is shown by:
```
04:37:24 {"TuyaReceived":{"Data":"55AA030700080302000400000094AE","Cmnd":7,"CmndData":"0302000400000094","3":{"DpId":3,"DpIdType":2,"DpIdData":"00000094"}}}
```
to be `00000094 (hex) = 148 (dec) = 14.8%`

And my bulb mode is shown in `{"DpId":4,"DpIdType":4,"DpIdData":"00"}}}` to be `00` which corresponds to LED.

These two settings affect how low the brightness can go (and how much noise the bulbs makes when its dim).

martin072, 2020-09-06:
Thanks for your quick response, I got a totally different output: 🧐

`
11:47:23 TYA: Heartbeat
11:47:34 TYA: Send "55aa00000000ff"
11:47:34 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:47:34 TYA: Heartbeat
11:47:36 WIF: Checking connection...
11:47:41 CMD: SerialSend5 55aa0001000000
11:47:41 SRC: WebConsole from 192.168.73.108
11:47:41 CMD: Group 0, Index 5, Command "SERIALSEND", Data "55aa0001000000"
11:47:41 RSL: stat/tasmota_60B20C/RESULT = {"SerialSend":"Done"}
11:47:41 {"TuyaReceived":{"Data":"55AA000100156253585353464172564B746334447943312E302E305A","Cmnd":1,"CmndData":"6253585353464172564B746334447943312E302E30"}}
11:47:41 TYA: MCU Product ID: bSXSSFArVKtc4DyC1.0.0
11:47:41 TYA: Send "55aa0002000001"
11:47:41 {"TuyaReceived":{"Data":"55AA000200020E0011","Cmnd":2,"CmndData":"0E00"}}
11:47:41 TYA: RX MCU configuration Mode=2
11:47:41 TYA: Read MCU state
11:47:41 TYA: Send "55aa0008000007"
11:47:41 {"TuyaReceived":{"Data":"55AA0007000501010001010F","Cmnd":7,"CmndData":"0101000101","DpType1Id1":1,"1":{"DpId":1,"DpIdType":1,"DpIdData":"01"}}}
11:47:41 TYA: fnId=11 is set for dpId=1
11:47:41 TYA: RX Relay-1 --> MCU State: On Current State:On
11:47:41 {"TuyaReceived":{"Data":"55AA00070008020200040000000016","Cmnd":7,"CmndData":"0202000400000000","DpType2Id2":0,"2":{"DpId":2,"DpIdType":2,"DpIdData":"00000000"}}}
11:47:41 TYA: fnId=21 is set for dpId=2
11:47:41 TYA: RX Dim State=0
11:47:45 TYA: Send "55aa00000000ff"
11:47:45 {"TuyaReceived":{"Data":"55AA000000010101","Cmnd":0,"CmndData":"01"}}
11:47:45 TYA: Heartbeat
`

Not really sure where to look here. any ideas? otherwise it is as it is.

dalovering, 2020-09-06:
Hmmm... it appears as though only dpid 1 and 2 are enabled in your MCU firmware. Just to be sure though, try a few things:
Ensure you set `weblog 4` and try `SerialSend5 55aa0001000000 ` again. See if the result is the same (I think Tasmota may not  display unmapped dpids in other weblog modes). 

Also, in your original firmware (if you used it at all), did you have any options for different bulb types or the lower limit?

You COULD try setting dpid 3 to maybe 100 (10%) by running `TuyaSend2 3,100` and see if it sets and changes anything. There is some (small) risk here that dpid does something different in your case, and we currently don’t know whether it exists or what it is set to, so we couldn’t just “set it back” if there is an issue. 

PS be sure to use code blocks for long logs 😊 https://docs.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks

martin072, 2020-09-07:
Thanks, thought i did the code block thing.

Weblog 4 was enabled, and the output is the same. I did not run the original firmware, flashed it straight out of the box. 
I won't risk it, so will leave it as it is for now. 

Thank you for the quick reply's

dalovering, 2020-09-07:
Darn. In that case I think your MCU just doesn’t use those features. You may consider trying a different bulb, some have different dimming responses.

martin072, 2020-10-04:
Been a while, but I had some time today to play with it again. I will settle with the minimum dim range, however, something else I noticed; The mcu always seems to fade in/out when turning on/off... When on max dim range, turning the light off seems to take forever, as it fade's it out. Do you know if this is an mcu thing or tasmota? I have turned Fade OFF and used LedTable 0 as well, still the light fades in and out? I am running 8.5.1 on the unit (8.4.0 had the same behaviour)
One difference; when removing power from the device, the first time On will not do the Fade thing. Any on/off's after the first ones will..