arendst/Tasmota #7382: platformio compile fails: Exclude "ArduinoSTL" lib via platformio.ini by default to solve it

systemofapwne, 2019-12-31:
**Have you looked for this feature in other issues and in the docs?**  
Yes

**Is your feature request related to a problem? Please describe.**  
For one of my projects, I need the "ArduinoSTL" library to be present in platformio (e.g. when there is no full STL support due to resource-limits, but one yet needs compatible STL syntax support - like on an Arduino). Whenever this package is installed, it interferes with *any* project, which builds against a target with native STL support but yet uses the Arduino libs.

**Describe the solution you'd like**  
In the past, the solution for me was to add

`lib_ignore = ArduinoSTL`

to my platformio.ini

**Describe alternatives you've considered**  
None

**Additional context**  
Not blockint ArduinoSTL will introduce a clash between the native STL of the compilation target and the "ArduinoSTL", thus compilation completely fails.

I therefore request a forced exclusion of this lib via platformio.ini by default, so nobody with the "ArduinoSTL" lib installed wonders, why compiling tasmota fails.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-12-31:
Not a issue nor a bug in Tasmota.

systemofapwne, 2019-12-31:
I know, it is no bug with Tasmota. I know it is on "my side" or - lets rephrase it - on **anyones** side having ArduinoSTL installed.

The thing is: The cause for is not immediately clear (hundreds of compile errors). It took me several hours to find out, that the cause for not compiling tasmota (with default config!) was for **having a third party library installed, which is not even linked against with tasmota**.

Jason2866, 2019-12-31:
Yes, but adding settings that may be never needed for someone else litters.
Thx, for sharing your finding. If someone else have this problem will find this here.
You can easily add your needed setting(s) in platformio_override.ini
This file will not be overwritten.

systemofapwne, 2019-12-31:
Im very sad to hear about this decision.

- Sure, it is not required for everyone
- But it does not hurt anyone
- And it helps people with this specific situation
--> All together a win-win situation

The thing is, that I expect nobody with that specific problem to suspect ArduinoSTL, so they will probably not associate my report here with their problems. But when they do, they probably already wasted hours just to find out, that ArduinoSTL is the root cause.

arendst, 2019-12-31:
Agree with @Jason2866. 

Perhaps you can use your platform_verride.ini file for your local changes.

systemofapwne, 2019-12-31:
Yes, this is, what I already did in order to fix the problems for me. I have no problems anymore and I just wanted to elaborate my findings here.

But everyone else, who has **this totally non-tasmota related library, that tasmota should not use** installed will fail to compile tasmota **with no sign, that this tasmota-unrelated package is the culprit of the error**.

That is, why I asked for this addition but I can understand your standpoint.

arendst, 2019-12-31:
Thx for your information.

closed 2019-12-31 by arendst