arendst/Tasmota #7385: Flash Mi-Light WL5 with Tasmota?

fkempter, 2019-12-31:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._

**Describe the solution you'd like**  
**### Someone has already flashed the MI-Light WL5 LED strip controller?
[image]
**

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-01:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Documentation](http://tasmota.com/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-01 by meingraham

WMP, 2020-06-19:
This device has WR3 module, based on RTL8710BN. This is impossible to flash it with Tasmota or ESPurna . 

[image]

trockenasche, 2020-06-19:
I can confirm that there was also a WR3 module on my WL5.

WMP, 2020-06-19:
I use tuyapi to control this device without internet network.

douginoz, 2020-06-30:
@WMP Have you got your WL5 working in homebridge yet?  I saw your comment in another github about being unable to make it work due to TLS issues.
I have 3 of these devices and would like to get it working too, so I would love to get your homebridge configuration if you got it working.  you can email me at inthecentreofitall at gmail.com

hoshy42, 2020-09-30:
Hi guys,

I unsoldered the unholy WR3 module and put an ESP-12e with Tasmota in its place. 

[image]

It's working, but now I am stuck fiddling around with the DpId's.

[image]

This is what it looks like with the help of the Tasmota Tuya Bookmarklet 
(found here: https://github.com/sillyfrog/Tasmota-Tuya-Helper)

I managed to figure out that DpId 22 is the main dimmer channel (10-1000) and 20 is the classic boolean ON/OFF.
23 and 26 do not seem to change anything (although i dont have anything connetcted to the CW and W channels @ the moment, only the RGB ones )

My guess is that i have to send some kind of coded 5 channel color string like "FF00000000" for "channel 1 at full"

Any pointers?

Thanks in advance
hoshy42

Aasikki, 2020-11-07:
@hoshy42 Have you had any progress with this? I was thinking about doing this and happy to see it works at least that far. Might habe to order some esp-12:s to try this and figure out how to make it work but I don't really know much about these things.

grasshide, 2021-09-16:
> Hi guys,
>
> I unsoldered the unholy WR3 module and put an ESP-12e with Tasmota in its place.
>
> [image]
>
> It's working, but now I am stuck fiddling around with the DpId's.
>
> [image]
>
> This is what it looks [...]

Sorry to bring this old thread up. But did document what you did anywhere? I recently bought a Tuya WB5 controller and I'm very interested in how to switch to the esp-12!

lucboudreau, 2021-12-11:
Same here. If anyone has more details on this device, please share.

kquinsland, 2021-12-12:
I am currently working on similar.

I only recently got an ESP module to boot while soldered into the MiBoxer module (the Mi5 has the `EN` pin pulled up... but i totally forgot that `RST` also needs to be pulled up)

Before bothering with the ESP, I use a logic analyzer to capture quite a few things. I will try to clean up my notes enough so the `pulseview` captures will make sense.

I have managed to figure out that `DpID: 101` has something to do with the button / mode... it's almost like a complete dump of the state for the onboard MCU. 

Here are two tasmota commands that can be used to set an LED strip to a golden color or a blue color:

```
BLUE-ish: 
SerialSend5 55 AA 00 06 00 19 1C 03 00 15 31 30 31 30 66 30 33 65 38 30 33 65 38 30 30 30 30 30 30 30 30 FA

GOLD-ish:
SerialSend5 55 AA 00 06 00 19 1C 03 00 15 31 30 30 35 32 30 33 65 38 30 33 65 38 30 30 30 30 30 30 30 30 CA
```

Those commands don't work _all the time_ the 'mode' button must be pushed some number of times to get the proper mode.

In at least one of the mode(s), DPiD 21 is light mode.

[image]

Aasikki, 2021-12-12:
Would be awesome if this could be made to work! I don't have any unused esp's at the moment and don't really think I know enough to help but I'd like to help if I can so I guess I'll have to get me some of those chips.

Jason2866, 2021-12-12:
Why not remove the controlling second MCU completely? Find the traces to the MosFets
cut the traces (or remove the MCU) and solder it directly to ESP Gpios.

kquinsland, 2021-12-13:
> Would be awesome if this could be made to work! I don't have any unused esp's at the moment and don't really think I know enough to help but I'd like to help if I can so I guess I'll have to get me some of those chips.

I more or less have it!  A **massive** _thank you_ to both @hoshy42 and @grasshide for their initial work and the _wonderful_ tool from @sillyfrog that made the RE effort 10x less painful than it could have been.

I cleaned up my notes and uploaded the various captures that I took with a logic analyzer here: 

https://github.com/kquinsland/Mi-Light-WL5-TuYa-MCU-ReverseEng

There is a TON more to figure out with this thing but the basics are there... even if they're not perfect.

-------

> Why not remove the controlling second MCU completely? Find the traces to the MosFets cut the traces (or remove the MCU) and solder it directly to ESP Gpios.

This is exactly what I was considering doing if there wasn't already a decent start on reverse engineering things. The tasmota docs are pretty helpful as are the contributions made my others in this thread. If I didn't make serious progress on this thing by the end of the weekend (aka... now!) then I was going to cut my losses and move on down that path.

I have not checked, but I am just assuming that the MCU and mossfets are all 3.3 capable. I really wanted to avoid too much destructive modifications to the PCB though just to keep the mod accessible to more people / lessen the amount of work that I had to do to convert the 4 others I have in my possession.  If level shifting electronics were going to be required then i'd probably just cut my losses and harvest the components and use them to build my own LED controller.  Ironically, this is _precisely_ what i was trying to avoid to begin with :P.

lucboudreau, 2021-12-23:
I've put together everyone's notes and tested them. I have a couple fully functioning WL5s using the template linked below. It needs a brain transplant with an ESP12-S. Other modules may require jumper wires.

https://github.com/blakadder/templates/pull/1289

kquinsland, 2021-12-23:
@lucboudreau Thanks for putting that together!

Where/How did you figure out the "red" light / strip mode?

```
Make sure to press the switch and select the mode corresponding to the strip you are using. ie. To use an RGB strip, you press the switch until it flashes red. The setting will stick between restarts.
```

This is the little status LED and not the light strip... right?

lucboudreau, 2021-12-23:
It was explained in the manual. By pressing the switch, it cycles between different wiring modes.

White Light Flashing: Single Color Mode
Yellow Light Flashing: Dual White Mode
Red Light Flashing: RGB Mode
Green Light Flashing: RGBW Mode
Blue Light Flashing: RGB+CCT Mode

kquinsland, 2021-12-23:
> It was explained in the manual. By pressing the switch, it cycles between different wiring modes.

Huh.  I didn't have any text to that effect in the manual(s) that I found.... But very good to know!  You may wish to add that to the template page you have open as I am sure that other people might have the controller but only wish to use a subset of the 5 channels.

lucboudreau, 2021-12-23:
[image]

kquinsland, 2021-12-23:
> [image]

Thanks!  Of course that details is missing from the manual that came with mine:
https://github.com/kquinsland/Mi-Light-WL5-TuYa-MCU-ReverseEng/blob/main/docs/YL5_EN.pdf

techsolo12, 2021-12-31:
@lucboudreau 
Did you got the cct control working. I mean that you can change the color temperature of cw and ww?

techsolo12, 2022-01-01:
> Why not remove the controlling second MCU completely? Find the traces to the MosFets cut the traces (or remove the MCU) and solder it directly to ESP Gpios. <

The problem is the MosFets will get only 3,3V from the esp. For full power you need 5V on the MosFet.

Saxin, 2022-04-21:
@lucboudreau Thanks for your efforts.
I've connected a RGB + CCT (5 channels) strip to WL5, the template doesn't give option to control the 5th channel (warm white),
Per my understanding it should look like shown here - [https://tasmota.github.io/docs/Lights/#5-channels-rgbcct-lights](https://tasmota.github.io/docs/Lights/#5-channels-rgbcct-lights)

Any chance we are missing something in the template ?
Or maybe it's some configuration.

douginoz, 2022-04-21:
I have 3 of these WL5 horrors that are still effectively brand new. I replaced them with esp-compatible modules as soon as I found that they were going to be a problem. 

If someone wants one or more let me know. You can  pay the shipping from Victoria Canada (v9c3n5) and I'll send them out.  I'm at inthecentreofitall@gmail.com.

Aasikki, 2022-08-14:
> @lucboudreau Thanks for your efforts. I've connected a RGB + CCT (5 channels) strip to WL5, the template doesn't give option to control the 5th channel (warm white), Per my understanding it should [...]

Same here, except that I only have rgb, no white channel at all. 

Also another issue, I made the mistake of trying the effects from home assistant and now I'm stuck in the rainbow effect and can't get out of it at all. Not even unplugging the psu for a while resets it.

Ed-Savill, 2022-12-22:
Sorry to reopen this but it's related to this thread.

I have got an ESP12-S transplanted in the WL5 and got Tasmota on an working (sort of).

I need help getting RGB+CCT working with this.

I have tried this code from another template (linked below) and it sort of works. I can control RGB & CCT separately.
When I am on an RGB value, and select an CCT value, the mode changes fine. But when I am on a CCT value and select an RGB value. The mode is stuck on the CCT mode.
 
Backlog DimmerRange 0,1000; TuyaMCU 11,20; TuyaMCU 26,21; TuyaMCU 21,22; TuyaMCU 23,23; TuyaMCU 24,24;

Source: https://templates.blakadder.com/miboxer_FUT039W.html

Can anyone help please?

Aasikki, 2023-01-22:
> Sorry to reopen this but it's related to this thread.
> 
> I have got an ESP12-S transplanted in the WL5 and got Tasmota on it working (sort of).
> 
> I need help getting RGB+CCT working with this.
> 
> I have tried this code from another template (linked below) and it sort of works. I can control RGB & CCT separately. But when I am on an RGB value, and select an CCT value, the mode changes fine. But when I am on a CCT value and select an RGB value. The mode is stuck on the CCT mode.
> 
> Backlog DimmerRange 0,1000; TuyaMCU 11,20; TuyaMCU 26,21; TuyaMCU 21,22; TuyaMCU 23,23; TuyaMCU 24,24;
> 
> Source: https://templates.blakadder.com/miboxer_FUT039W.html
> 
> Can anyone help please?

I'm having the exact same problem. Rgb works just fine but cct gives trouble. I'm already considering buying other controllers as I have no idea how to fix it. Any help would really be appreciated so my cheap ass can save that money haha!

Aasikki, 2023-01-23:
> Sorry to reopen this but it's related to this thread.
>
> I have got an ESP12-S transplanted in the WL5 and got Tasmota on it working (sort of).
>
> I need help getting RGB+CCT working with this.
>
> I have [...]

I'm having the exact same problem. Rgb works just fine but cct gives trouble. I'm already considering buying other controllers as I have no idea how to fix it. Any help would really be appreciated so my cheap ass can save that money haha!

Jason2866, 2023-01-23:
Writing in an old close issue will not be visible for many.

Please open a new discussion.