arendst/Tasmota #7387: LCD-Commands are buggy LCD2004

BigWumpus, 2019-12-31:
### PROBLEM DESCRIPTION
I started again a project to use an 4x20 LCD-alphanumeric-display (LCD2004).
I try to understand the commands in the wiki here (https://github.com/arendst/Tasmota/wiki/Displays) the parameters are not explained.
Then i used google and my old script to work with...
DisplayModel 1
DisplayAddress 39
DisplayMode 0
DisplayRows 4
DisplayCols 20
works for me.
But I can't address any position in line 0 (DisplayText [l0c2]ABC) will not work.
And I can't address any position at col 0 (DisplayText [l1c0]XYZ) will address line 0 col 0.
This is not fine and it is an older problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [ ] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [X] Tasmota binary firmware version number used: 8.1.0.2
  - [ ] Pre-compiled
  - [X] Self-compiled
    - [ ] IDE / Compiler used: _____
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
  Configuration output here:
23:00:00 CMD: Backlog Template; Module; GPIO 255
23:00:01 MQT: stat/Home-Display/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
23:00:01 MQT: stat/Home-Display/RESULT = {"Module":{"18":"Generic"}}
23:00:01 MQT: stat/Home-Display/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"6":"I2C SDA"},"GPIO5":{"5":"I2C SCL"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:

23:01:46 CMD: status 0
23:01:46 MQT: stat/Home-Display/STATUS = {"Status":{"Module":18,"FriendlyName":["Home-Display"],"Topic":"Home-Display","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
23:01:46 MQT: stat/Home-Display/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"External System","Uptime":"0T03:53:01","StartupUTC":"2019-12-31T18:08:45","Sleep":50,"CfgHolder":4617,"BootCount":14,"SaveCount":44,"SaveAddress":"3F7000"}}
23:01:46 MQT: stat/Home-Display/STATUS2 = {"StatusFWR":{"Version":"8.1.0.2(tasmota)","BuildDateTime":"2019.12.31 15:56:56","Boot":31,"Core":"2_6_2","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"366/699"}}
23:01:46 MQT: stat/Home-Display/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["@home",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C8000100060000005A00000000000000","00000000","00000000"]}}
23:01:46 MQT: stat/Home-Display/STATUS4 = {"StatusMEM":{"ProgramSize":568,"Free":2500,"Heap":23,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000407","8FF8E38F","043683A1","20B617CF","01001BC0","00007881"],"Drivers":"1,2,3,4,6,7,8,9,10,12,13,16,18,19,20,21,22,24,26,29","Sensors":"1,3,4,5,6,7,8,9,10,14,15,17,18,20,26,34"}}
23:01:46 MQT: stat/Home-Display/STATUS5 = {"StatusNET":{"Hostname":"Home-Display","IPAddress":"192.168.178.183","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.178.1","Mac":"CC:50:E3:0F:1F:24","Webserver":2,"WifiConfig":4,"WifiPower":17.0}}
23:01:46 MQT: stat/Home-Display/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.11","MqttPort":8883,"MqttClientMask":"Home-Display","MqttClient":"Home-Display","MqttUser":"HA","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
23:01:46 MQT: stat/Home-Display/STATUS7 = {"StatusTIM":{"UTC":"Tue Dec 31 22:01:46 2019","Local":"Tue Dec 31 23:01:46 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:02"}}
23:01:46 MQT: stat/Home-Display/STATUS10 = {"StatusSNS":{"Time":"2019-12-31T23:01:46"}}
23:01:46 MQT: stat/Home-Display/STATUS11 = {"StatusSTS":{"Time":"2019-12-31T23:01:46","Uptime":"0T03:53:01","UptimeSec":13981,"Vcc":3.002,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"@home","BSSId":"5C:49:79:38:A1:02","Channel":13,"RSSI":100,"Signal":-46,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2020-01-01:
First some workarounds:

1) Use x and y instead of c and l like ``DisplayText [y0x2]ABC``. This will be solved in a fix where line and column numbers are offset by 1. So ``DisplayText [l1c3]ABC`` will then work.
2) Use p to pad the correct amount of characters like ``DisplayText [y3x4p3]A~2CB``. This will be solved where the extra spaces will be removed.

I have 1 fixed. Working on 2.

BigWumpus, 2020-01-01:
Thanks!
y/x is working, p also...

closed 2020-01-01 by BigWumpus

Race666, 2021-04-14:
@BigWumpus I'm also want to connect a (HD44780) Display. Have you got the display running right now? Which display do you use? Is there a detailed description for that display (i2c expander, how the pins are connected i2c <-> HD44780) ?

BigWumpus, 2021-04-18:
Am 14.04.2021 um 08:49 schrieb Race666:
>
> @BigWumpus <https://github.com/BigWumpus> I'm also want to connect a 
> (HD44780) Display. Have you got the display running right now? Which 
> display do you use? Is there a detailed description for that display 
> (i2c expander, how the pins are connected i2c <-> HD44780) ?
>
Hello,

I soldered the I2C-Adapter to a 2004A-type Display (20 chars, 4 lines), 
set the jumper and connect the pins to a WEMOS D1-clone:
5V, Gnd, D2=SDA, D1=SCL

Race666, 2021-04-19:
@BigWumpus  Thx for reply :-) In the meanwhile I [patched ](https://michlstechblog.info/blog/download/electronic/hd44780-i2c-LCDd-diff-v9.3.1.patch )xdsp_01_lcd.in, LiquidCrystal_I2C.c and LiquidCrystal_I2C.h and got my old LCDd/lcdproc compatible HD44780 i2c display (wiring between PCF8574 <-> HD44780 is different) running :-)