arendst/Tasmota #7389: Sonoff S20 lost connection when the relay is off

Korte68, 2020-01-01:
### PROBLEM DESCRIPTION
The S20 device working well until the relay is on state. after switch it off, the connection is lost, the LED is blinking. If  I switch it on with the button, the connection working again.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff S20
- [x] Tasmota binary firmware version number used: 8.1.0
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: OTA
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
12:02:33 CMD: Backlog Template; Module; GPIO 255
12:02:33 MQT: stat/sonoff_kon1/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
12:02:33 MQT: stat/sonoff_kon1/RESULT = {"Module":{"8":"Sonoff S2X"}}
12:02:33 MQT: stat/sonoff_kon1/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  12:03:52 CMD: Backlog Rule1; Rule2; Rule3
12:03:52 MQT: stat/sonoff_kon1/RESULT = {"Rule1":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:03:52 MQT: stat/sonoff_kon1/RESULT = {"Rule2":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
12:03:52 MQT: stat/sonoff_kon1/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}
```
- [ ] Provide the output of this command: `Status 0`:
```
 12:04:29 MQT: stat/sonoff_kon1/STATUS = {"Status":{"Module":8,"FriendlyName":["Sonoff"],"Topic":"sonoff_kon1","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
12:04:29 MQT: stat/sonoff_kon1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"https://github.com/arendst/Tasmota/releases/download/v8.1.0/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:40","StartupUTC":"2020-01-01T11:01:49","Sleep":50,"CfgHolder":4617,"BootCount":3096,"SaveCount":6919,"SaveAddress":"F7000"}}
12:04:29 MQT: stat/sonoff_kon1/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":4,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"363/699"}}
12:04:29 MQT: stat/sonoff_kon1/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["kortechnika","kortelak"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000029","2805C8000100060000005AFF000000000000","00000001","00000000"]}}
12:04:29 MQT: stat/sonoff_kon1/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
12:04:29 MQT: stat/sonoff_kon1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_kon1-2621","IPAddress":"192.168.1.238","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"18:FE:34:CC:CA:3D","Webserver":2,"WifiConfig":2}}
12:04:29 MQT: stat/sonoff_kon1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.100","MqttPort":1883,"MqttClientMask":"sonoff_kon1","MqttClient":"sonoff_kon1","MqttUser":"homeassistant","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
12:04:29 MQT: stat/sonoff_kon1/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 01 11:04:29 2020","Local":"Wed Jan 01 12:04:29 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:03"}}
12:04:29 MQT: stat/sonoff_kon1/STATUS10 = {"StatusSNS":{"Time":"2020-01-01T12:04:29"}}
12:04:29 MQT: stat/sonoff_kon1/STATUS11 = {"StatusSTS":{"Time":"2020-01-01T12:04:29","Uptime":"0T00:02:40","UptimeSec":160,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"kortechnika","BSSId":"CA:41:FC:9A:B5:FE","Channel":2,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:08"}}}
```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
00:00:00 CFG: Loaded from flash at FA, Count 6924
00:00:00 QPC: Flag 0E
00:00:00 CFG: CR 363/699
00:00:00 SRC: Restart
00:00:00 Project tasmota Sonoff Version 8.1.0(tasmota)-2_6_1
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP2 kortelak in mode 11N as sonoff_kon1-2621...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 APP: Button1 multi-press 1
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:03 SRC: Button
00:00:03 RSL: stat/sonoff_kon1/RESULT = {"POWER":"ON"}
00:00:03 RSL: stat/sonoff_kon1/POWER = ON
00:00:04 CFG: Saved to flash at F9, Count 6925, Bytes 4096
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Attempting connection...
[... 65 lines cut ...]
00:00:32 MQT: stat/sonoff_kon1/POWER = ON (retained)
00:00:38 MQT: tele/sonoff_kon1/STATE = {"Time":"1970-01-01T00:00:38","Uptime":"0T00:00:38","UptimeSec":38,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":49,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"kortechnika","BSSId":"CA:41:FC:9A:B5:FE","Channel":2,"RSSI":100,"Signal":-47,"LinkCount":1,"Downtime":"0T00:00:32"}}
12:17:10 NTP: Drift 0, (UTC) Wed Jan 01 11:17:10 2020, (DST) Sun Mar 29 02:00:00 2020, (STD) Sun Oct 25 03:00:00 2020
12:17:20 WIF: Checking connection...
12:17:20 WIF: Connected
12:17:40 WIF: Checking connection...
12:17:40 WIF: Connected
```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Switch the relay off.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Should stay connected.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-01:
Hardware is defect.

meingraham, 2020-01-01:
Please address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2020-01-01 by meingraham

tty74it, 2020-03-08:
I've the same problem with s20 and 8.1.0.  Now i've tryed 7.2.0 seem to work

Korte68, 2020-03-08:
sonoff minimal is too big...
Upload Failed

Not enough space

tty74it, 2020-05-13:
The problem reappeared. I've tried with esphome and espurna, and the problem is the same.  After a full functional system when relay is on, the wifi disconnect after 60 sec or more when relay is off, ( extensive data loos on mikrotik ap). I push the button , relay turn on and all works.

fpdragon, 2023-07-21:
Hi all,

I think I had a similar problem on my Sonoff S20 with the most current firmware 13.0.0 and my new Wifi Mesh Tenda 12X.

Before, all was working with tasmota ~12 (not sure anymore) and my old wifi.
Then I received the Tenda 12X mesh wifi and I had the connection issues you described.
Then I upgraded to tasmota 13.0.0 in case it may be already fixed. Still the same problem.
I contacted the Discord group and they told me that it may be an issue with the router/mesh device (a.k. as Asus problem?!). If so, then I am only able to play with the wifi settings of the router, disable some settings and if this does not help then go back to the old router, but I won't prefere that. It is also said that it is a bug but it is related to the ESP preshipped firmware and can't be fixed by the tasmota team.
First I gave up but then I took another try and found this post, that is especially for the Sonoff S20 and sounded promissing.

First I was not able to downgrade from 13.0.0 to 7.2.0 but then I successfully downgraded to 8.5.1 over the web gui. Afterwards the downgrade to 7.2.0 was also possible in a second step.
So I am now on 7.2.0, as proposed.

First tests are showing that the bug with the complete disconnect seems to be "solved". The device keeps available somehow.

However, I noticed with permanent pinging that about 40% of all ping requests are lost so the connection is very unstable.
This is also very noticable in the web interface and the device is not usable in that way.

Recap:
So there seems to be a known bug in the ESP chips/firmware of an older generation that leads to incompatibilitys with some modern routers and lead to connection drops.
Additionally there is this sonoff S20 issue related with the button/relais that leads to a complete disconnect until the button is pressed again. This may be also related to the first problem but seems to be special case for this device. On tasmota 7.2.0 the complete disconnect seems to be solved but still the connection is very unstable in case of an older ESP FW and an incompatible router.
Are my conclusions correct?

If there is still no method to upgrade the ESP internal firmware I guess I have to drop the older devices, as of the Sonoff S20, and replace it with newer one.

Helicon007, 2025-01-02:
My Sonoff S20 has worked for years without any update, still 6.5.0.2. Starting a few days ago I faced the same issue. Wifi only with relais on. This is not an Software issue. It has something to do with the very noisy voltage supply build in. If I connect external 3.3V the Wifi is working without any issues. Only internal supply via 230V to 3.3V is disturbing the connection. With the relais on the voltage is slightly more stable what makes the difference. 

Measuring with an Oscilloscope the voltage has a triangular shape like a saw blade between 2.6V and 2.9V and 100 microseconds what is not ideal for the ESP. Capacitors looked fine and there was no visible defect. So i decided to replace the S20 with a more modern switch and let it be. 

I hope this information find the people looking for it.

Regards, Helicon

joba-1, 2025-01-02:
good move. If capacitors allow 10% ripple they are not doing their job.

csanz91, 2025-01-06:
> My Sonoff S20 has worked for years without any update, still 6.5.0.2. Starting a few days ago I faced the same issue. Wifi only with relais on. This is not an Software issue. It has something to do [...]

I just fixed my Sonoff S20, which had the same problem. The issue was the 330uF 16V capacitor located to the right of the transformer. I replaced it with a similar one, and it started working fine again.

michalicekmilan, 2025-10-26:
@csanz91 indeed replacing 330uF 16V capacitor fixed the issue, thanks for this hint!