arendst/Tasmota #7391: Autoset MQTT topic/hostname based of friendly name

imduffy15, 2020-01-01:
**Have you looked for this feature in other issues and in the docs?**  

Yes

**Is your feature request related to a problem? Please describe.**  

I have to manually set a friendly name and mqtt name, it would be nice if I just had to set one unique identifier.

**Describe the solution you'd like**  

I think we could copy what home assistant does for generating entity IDs. They take the friendly names, lower case them and remove any spaces.

Sitting room light -> sitting_room_light
Bedroom lamp -> bedroom_lamp

**Describe alternatives you've considered**  

Unsure of alternatives, maybe some sort of device provisioning script that uses backlog commands to set all this up without touch Tasmota code.

**Additional context**  

None

Jason2866, 2020-01-01:
There are users which are not using Home Assistant -;)
For example: Using for topic/hostname the same name as for friendly name
would make use of Alexa impossible.
Since you can set everything like you need/want, this change is not needed
and breaks compatibility with other use cases

imduffy15, 2020-01-01:
> Since you can set everything like you need/want, this change is not needed

I'm not suggesting we remove the ability to change, just suggesting we improve the default value which is static as "Tasmota"

Jason2866, 2020-01-01:
Will probably not happen since this is a setting that needs to be done only for setup (one time).
Implementing this option needs additional code which is missing for other features
which are not just for setup.

meingraham, 2020-01-01:
As explained.

closed 2020-01-01 by meingraham