arendst/Tasmota #7393: Sonoff POW R2 reports wrong values

HTPCanwender, 2020-01-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

I want to use the Sonoff POW R2 to detect if my TV and my PC are switched on or off. The TV is connected to the output of the relay. I use the measured power consumption to determine if the TV is switched on or in standby mode and report it with the rule

ON Energy#Power < 19 DO Publish stat/Tasmota-48/PWR-THRESHOLD OFF ENDON ON Energy#Power > 19 DO Publish stat/Tasmota-48/PWR-THRESHOLD ON ENDON

Relay1 keeps switched on all the time in normal operation. I connected the 5V voltage of the PC via a voltage divider to GPIO5. The status of the corresponding Relay2i is reported over MQTT as normal.

The configuration is:

Module type (Sonoff Basic)
Generic (18)

D3 GPIO0 Button1	  Button1 (17)
TX GPIO1 Serial Out  CSE7766 Tx (145)
D4 GPIO2 	          None (0)
RX GPIO3 Serial In     CSE7766 RX (146)
D2 GPIO4 	          None (0)
D1 GPIO5 	          Switch2n (83)
D6 GPIO12 Relay1	  Relay1 (21)
D7 GPIO13 Led1i	  Relay2i (30)
D5 GPIO14 Sensor	  None (0)
D8 GPIO15                None (0)
D0 GPIO16                None (0)
A0 ADC0	                  None (0)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POW R2
- [x] Tasmota binary firmware version number used: 8.1.0(tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
    - [ ] IDE / Compiler used: _____
- [x] Flashing tools used: Tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
17:10:53 MQT: stat/Tasmota-48/RESULT = {"NAME":"Generic","GPIO":[17,145,0,146,0,0,0,0,21,56,0,0,0],"FLAG":0,"BASE":43}
17:10:53 MQT: stat/Tasmota-48/RESULT = {"Module":{"18":"Generic"}}
17:10:54 MQT: stat/Tasmota-48/RESULT = {"GPIO0":{"17":"Button1"},"GPIO1":{"145":"CSE7766 Tx"},"GPIO2":{"0":"None"},"GPIO3":{"146":"CSE7766 Rx"},"GPIO4":{"0":"None"},"GPIO5":{"83":"Switch2n"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"21":"Relay1"},"GPIO13":{"30":"Relay2i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
17:13:20 MQT: stat/Tasmota-48/RESULT = {"Rule1":"ON","Once":"ON","StopOnError":"OFF","Free":369,"Rules":"ON Energy#Power < 19 DO Publish stat/Tasmota-48/PWR-THRESHOLD OFF ENDON ON Energy#Power > 19 DO Publish stat/Tasmota-48/PWR-THRESHOLD ON ENDON"}
17:13:20 MQT: stat/Tasmota-48/RESULT = {"Rule2":"OFF","Once":"ON","StopOnError":"OFF","Free":511,"Rules":""}
17:13:20 MQT: stat/Tasmota-48/RESULT = {"Rule3":"OFF","Once":"OFF","StopOnError":"OFF","Free":511,"Rules":""}

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
17:13:46 MQT: stat/Tasmota-48/STATUS = {"Status":{"Module":18,"FriendlyName":["Tasmota-48","Tasmota2"],"Topic":"Tasmota-48","ButtonTopic":"0","Power":3,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:13:46 MQT: stat/Tasmota-48/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T01:01:23","StartupUTC":"2020-01-01T15:12:23","Sleep":50,"CfgHolder":4617,"BootCount":27,"SaveCount":356,"SaveAddress":"F7000"}}
17:13:46 MQT: stat/Tasmota-48/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8266EX","CR":"373/699"}}
17:13:46 MQT: stat/Tasmota-48/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BoekiNet",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["04008009","2805C8000100068000005A00000000000000","00000000","00000000"]}}
17:13:46 MQT: stat/Tasmota-48/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":22,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
17:13:46 MQT: stat/Tasmota-48/STATUS5 = {"StatusNET":{"Hostname":"Tasmota-48-7934","IPAddress":"192.168.3.48","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.1","Mac":"CC:50:E3:1A:FE:FE","Webserver":2,"WifiConfig":4}}
17:13:46 MQT: stat/Tasmota-48/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.3.50","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1AFEFE","MqttUser":"BoekiDevice","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:13:46 MQT: stat/Tasmota-48/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 01 16:13:46 2020","Local":"Wed Jan 01 17:13:46 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:03"}}
17:13:46 MQT: stat/Tasmota-48/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
17:13:46 MQT: stat/Tasmota-48/STATUS10 = {"StatusSNS":{"Time":"2020-01-01T17:13:46","Switch2":"ON","ENERGY":{"TotalStartTime":"2019-12-29T19:59:07","Total":0.153,"Yesterday":0.279,"Today":0.153,"Power":26,"ApparentPower":31,"ReactivePower":16,"Factor":0.84,"Voltage":237,"Current":0.129}}}
17:13:46 MQT: stat/Tasmota-48/STATUS11 = {"StatusSTS":{"Time":"2020-01-01T17:13:46","Uptime":"0T01:01:23","UptimeSec":3683,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"ON","POWER2":"ON","Wifi":{"AP":1,"SSId":"BoekiNet","BSSId":"5C:49:79:FF:A7:6F","Channel":11,"RSSI":90,"Signal":-55,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [ ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

If the relay1 is switched off, the power consumption displayed should be zero independent from the state of the virtual relay2. But only if I switch off relay2 too, the right value (zero) is displayed. Please have a look at the screenshots. The displayed values seem to be random, independet of the status of relay1.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
[image]
[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-01:
No support since you connected additional stuff to a POW.
This is dangerous with power monitoring devices.
Please read wiki why.

HTPCanwender, 2020-01-01:
Thank you for your advice. Don't be afraid, I am an electrical engineer and know what I am doing.
After some hours of testing I think it's a hardware issue and not a software issue. If I connect a hardware switch to GPIO5 and GND everything works fine. Relay2 is set correctly and the power metering works well too. I will use an additional relay to uncouple the circuits of the pow and my PC. Then it will be safe.

ascillato, 2020-01-02:
> Don't be afraid, I am an electrical engineer and know what I am doing.

Sorry but safety comes first. It isn't a good idea to encourage this types of mods that are really dangerous. See old issues for proof. That is why we post warnings in the wiki and in the docs.

Closing issue.

closed 2020-01-02 by ascillato2