arendst/Tasmota #7399: Counter error during shutter position

Coroglio, 2020-01-01:
### PROBLEM DESCRIPTION
Moving shutter in single direction, during the run, the position counter reset the value in lower/higher than previous.
Same expeience on two differents devices
Please, I noted other open issues similar to this one, but nobody speak about shutterposition value explicitely (one has been closed for incomplete template informations)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/#/help/FAQ)
- [x] Searched the problem in the [forum](https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Dual R2
- [x] Tasmota binary firmware version number used: 8.1.0 Core/SDK Version  2_6_1/2.2.2
  - [ ] Pre-compiled
  - [x] Self-compiled
    - [x] IDE / Compiler used: GitPod
- [x] Flashing tools used: NodeMCU firmware programmer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```
  Configuration output here:
CMD: Backlog Template; Module; GPIO 255
MQT: cover/letto/stat/RESULT = {"NAME":"Generic","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":15,"BASE":18}
MQT: cover/letto/stat/RESULT = {"Module":{"39":"Sonoff Dual R2"}}
MQT: cover/letto/stat/RESULT = {"GPIO0":{"9":"Switch1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"22":"Relay2"},"GPIO9":{"10":"Switch2"},"GPIO10":{"17":"Button1"},"GPIO12":{"21":"Relay1"},"GPIO13":{"56":"Led1i"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```
  Rules output here:
NO RULES

```
- [x] Provide the output of this command: `Status 0`:
```
  STATUS 0 output here:
CMD: STATUS 0
MQT: cover/letto/stat/STATUS = {"Status":{"Module":39,"FriendlyName":["Tapparella1","Blinds1"],"Topic":"cover/letto","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[7,7,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
MQT: cover/letto/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:58:58","StartupUTC":"2020-01-01T21:28:44","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":42,"SaveAddress":"FA000"}}
MQT: cover/letto/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-30T13:25:24","Boot":31,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"392/699"}}
MQT: cover/letto/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["MYSSID1","MYSSID2"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C8000100060000005A00000000000000","40000008","00000000"]}}
MQT: cover/letto/stat/STATUS4 = {"StatusMEM":{"ProgramSize":561,"Free":440,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":255,"Features":["00000809","8FCAE397","04368321","22B217CD","01001BC0","00005981"],"Drivers":"1,2,3,4,5,8,9,10,12,16,18,20,21,22,24,26,27,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,17,18,20,22,26,34"}}
MQT: cover/letto/stat/STATUS5 = {"StatusNET":{"Hostname":"Tapparella-CameraLetto","IPAddress":"192.168.1.151","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.254","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":4}}
MQT: cover/letto/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.251","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AD0C35","MqttUser":"mqttuser","MqttCount":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
MQT: cover/letto/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Jan 01 22:27:42 2020","Local":"Wed Jan 01 23:27:42 2020","StartDST":"Sun Mar 29 02:00:00 2020","EndDST":"Sun Oct 25 03:00:00 2020","Timezone":99,"Sunrise":"07:26","Sunset":"16:44"}}
MQT: cover/letto/stat/STATUS10 = {"StatusSNS":{"Time":"2020-01-01T23:27:42","Switch1":"ON","Switch2":"ON","Shutter1":{"Position":0,"direction":0}}}
MQT: cover/letto/stat/STATUS11 = {"StatusSTS":{"Time":"2020-01-01T23:27:42","Uptime":"0T00:58:58","UptimeSec":3538,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":54,"MqttCount":2,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MYSSID1","BSSId":"XX:XX:XX:XX:XX:XX","Channel":11,"RSSI":76,"Signal":-62,"LinkCount":1,"Downtime":"0T00:00:06"}}}

```
- [x ] Provide the output of the Console log output when you experience your issue; if applicable:
  _(Please use_ `weblog 4` _for more debug information)_
```
  Console output here:
MQT: cover/letto/stat/SHUTTER1 = 0
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":0,"direction":0}}
CFG: Saved to flash at FB, Count 49, Bytes 4096
WIF: Checking connection...
WIF: Connected
CMD: shutterposition 50
SRC: WebConsole from 192.168.1.18
CMD: Group 0, Index 1, Command "SHUTTERPOSITION", Data "50"
SHT: Pos. in: payload 50 (2), payload 50, idx 1, src 9
SHT: lastsource 9:, real 6, target 17900, payload 50
SHT: Start in dir 0
SRC: Shutter
SHT: Switched relay: 1 by Shutter
MQT: cover/letto/stat/RESULT = {"POWER1":"ON"}
MQT: cover/letto/stat/POWER1 = ON
MQT: cover/letto/stat/RESULT = {"ShutterPosition1":50}
CFG: Saved to flash at FA, Count 50, Bytes 4096
SHT: Shutter 1: Real. 906, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 0.45 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":2,"direction":1}}
SHT: Shutter 1: Real. 2906, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 1.45 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":8,"direction":1}}
SHT: Shutter 1: Real. 5006, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 2.50 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":13,"direction":1}}
SHT: Shutter 1: Real. 7006, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 3.50 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":19,"direction":1}}
SHT: Shutter 1: Real. 9006, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 4.50 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":25,"direction":1}}
SHT: Shutter 1: Real. 10906, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 5.45 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":0,"direction":1}}
SHT: Shutter 1: Real. 12906, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 6.50 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":6,"direction":1}}
SHT: Shutter 1: Real. 15006, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 7.50 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":11,"direction":1}}
SHT: Shutter 1: Real. 16906, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 8.45 [s], freq 1000
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":17,"direction":1}}
SRC: Shutter
SHT: Switched relay: 1 by Shutter
MQT: cover/letto/stat/RESULT = {"POWER1":"OFF"}
MQT: cover/letto/stat/POWER1 = OFF
SHT: Shutter 1: Real. 17906, Start: 6, Stop: 17900, dir 1, motordelay 0, rtc: 8.9 [s], freq 1000
MQT: cover/letto/stat/SHUTTER1 = 0
MQT: cover/letto/tele/RESULT = {"Shutter1":{"Position":0,"direction":0}}
CFG: Saved to flash at F9, Count 51, Bytes 4096

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Moving shutter by GUI (slider or console), external keys (switch or button), MQTT command

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
shutteropen and shutterclose are execute witheout problem (motor is stopped by automatic power overload protection ?) The slider position reflect the shutterposition value that isn't the real value of cover position (ie slider and shutterposition 0 while the blind is 30% open)

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-02:
Please use latest development version. The issue is fixed.

Coroglio, 2020-01-02:
Oh thanks ! Work fine !

closed 2020-01-02 by Coroglio