arendst/Tasmota #7401: Suggestion - Dedicated Smarthome WiFi with Tasmota Mesh

lokidaibel, 2020-01-02:
**Have you looked for this feature in other issues and in the docs?**  

**Is your feature request related to a problem? Please describe.**  
Not realy but a general Problem of Smarthome Environments

**Describe the solution you'd like**  
Implement a Mesh feature like i describe here.

**Describe alternatives you've considered**  
If it cant be done everyone has to keep fighting the Wifi Problem ;D

**Additional context**  
one of the bigger chalanges in the smarthome environment is to provide a good Wifi network structure so that all your devices can be reached.

But actuall we dont need this. Most Tasmota systems has a managment Backend like Openhab2 that runs on a Raspberry .

Why not implement a optinal fuction in addition to the classic Wifi AP konfiguration to use the Raspberry to setup a entrypoint for all Tasmota devices which then will create a Meshed network infrastructure only for your smarthome ?

[image]

So every tasmota device can reach another x devices and route the traffic. In this case we will have a dedicated wifi thats extends with new Tasmota device.

Pros:
+  you can extend your Home without be afraid about wifi quality.
+     all Tasmota devices will get better wifi strength
+ more security: you can manage the access to the tasmota devices more easly from the Raspberry

Contra (that i can assume):
- more power usage on the device
- more load
- KO: maybe the device has not enougth CPU power for this ?
- Not enougth size on the image to implement this on the most devices ?

Please let us discuss this.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2020-01-02:
Thanks for sharing your ideas. A PR is welcome to discuss the possibilitys.

ascillato, 2020-01-02:
I have already tested that approach. The ESP8266 device can manage at the same time a connection as a station and as an access point. So It can connect to a router or other device AND allow other devices to connect to it. This uses very little CPU power but a lot more of energy.

This approach is doable, but requires some huge effort to adapt Tasmota to it due to a mesh network will require its own lightweight protocol for TASMOTA to TASMOTA communications and retransmissions.

Anyway, as the reliability is good, it will never be as good as a dedicated wifi router.

Also, if you do a wifi scan, you will see a bunch of SSIDs on your home (one per device). Yes, you can hide the SSID, but you will see a lot of hidden networks.

From the security point, you can have that mesh encrypted and with its own protocol, and as I tested, you can make that the esp8266 kick any device that could connect to it but is not a Tasmota device.

So, I think is doable, but the effort is quite huge and I don't know if it is better than a well designed wifi network. 

Let's see also, how is going to be the new standard that Google, Apple and Zigbee are studying now based on IP. May be is something like mesh?

ascillato, 2020-01-02:
From the energy side, there will be some devices like some sonoff basics with bad power regulators, that won't be able to manage the extra energy consumption as they were designed to have the esp8266 with sleep enabled.

While the esp8266 is on AP mode, sleep cannot be enabled and the consumption rises a lot compared to station mode with dynamic sleep.

arendst, 2020-01-02:
Considering the amount of effort to make this happen I also think it won't be released soon.

As I also had issues with routers not allowing more connections than 32 I redesigned my wifi environment using 2 Ubiquity AP's which can handle 128 devices each and I'm very pleased with the result. All use now one SSID used by both AP's and RSSI is mostly very good. The ones with bad reception in the old situation also have trouble keeping the connection in the new situation so I probably have to add a third AP ;-)

All this without adding mesh functionality to Tasmota.

lokidaibel, 2020-01-02:
@arendst:
Thanks for the tipp with the Ubiquity AP. Will also definitiv look into this.
But from the security point of view there is a valid concern to send your smarthome devices into a seperate WiFi.

Therfore i think this PR here would be nice addition to existing infrastructure.

i will definitly take a look at these APs. Cause in some edges my WiFi infrastratcture is not final
(finished building 2 months ago and as everyone that builds a house in germany my liquidity is now very restricted ;D)

@ascillato
Cool. So the Hardware of my eg. Sonoff Touch T1 should not be a problem for this ?

ascillato, 2020-01-02:
Should work, but be aware that adding mesh is a huge work and the outcome might be of lower reliability than a good designed wifi network.

About your smartphones, you can make separated LANs in the same wifi network to enhance security if you want. That is why, a more powerful solution can be achieved with good wifi routers like Mikrotik, Unifi, etc.

ascillato, 2020-01-02:
So, let's see if there is someone interested on working in your feature request, and to make a PR (Pull Request - code to add that feature to tasmota).

stefanbode, 2020-01-05:
The idea is nice and I know from another implementation @NI National instruments that you should not underestimate the devil inside. You must prevent loops, support alternatives and have a plan how to reach a device. All this can be code heavy and ac killer for the es8266. Is there any GitHub project already evaluating a mesh network with esp8266?

stefanbode, 2020-01-05:
Maybe a point to start. Anyhow overhead looks quite large

altelch, 2020-01-05:
https://github.com/esp8266/Arduino/pull/6280 Mesh is planed for 2.7 Milestone in Arduino Core.

ascillato2, 2020-01-11:
Closing this request as it is being developed on the core itself. Thanks for sharing your ideas.

closed 2020-01-11 by ascillato2

kugelkopf123, 2020-06-17:
BTW, it has been added: 
https://github.com/esp8266/Arduino/pull/6280

sibero80, 2020-11-02:
I live in South America, where most people live in apartment buildings made in concrete, RF contamination is high and good quality networking (routers, AP's) equipment is very expensive to get.

Most of the essential devices such as computers or TVs are connected via cable, but I still find some blind spots in my tasmota network even though it's a very small apartment (less than 70sqm but lots of concrete).

I hope this feature gets implement in order to make the most of existing hardware.